Bodmer/TFT_eSPI #3357: ESP32 WROOM 32D with an ILI9341 not displaying the full screen/visual artifacts in a 3rd % of the screen.

Gagz9k, 2024-06-07:
Hi!

1. I'm having an issue where using an **ESP32 WROOM 32D** with an **ILI9341** part of the screen (the side or buttom depending in how you look at it) gets distorted/repeated. I can change the orientation in the sketch, but it only change the position of the failed area.

The resulting visual is as follow:
[image]
[image]
[image]

Notice that changing the orientation change the problematic area, not only in location, but also in content, from white noise to a "copy" of a part of the display, but while the main part is animated, the duplicated/error is still. No matter the test, part of the screen is simple not displaying data correctly.

I'm using the TFT_Meters for demonstration, but the same error occur with virtually any test, including the generic ones and custom code done using the library. 

2. I'm using ArduinoIDE
3. I'm using TFT_eSPI 2.5.43
4. I've tried ESP32 by Espressif Systems 3.0.1 and Arduino ESP32 Boards by Arduino, version 2.0.13
5. I'm using the ESP32 Dev Module, but I've test all other modules compatible with my ESP32 WROOM 32D, with the same result.
6. I'm using a ILI9341 2.8''
7. SPI 
8. For connections I'm following this tutorials; https://www.youtube.com/watch?v=rq5yPJbX_uk&t=358s&ab_channel=XTronical & https://www.youtube.com/watch?v=pYp1Nxmfdeg&ab_channel=ShotokuTech and this is the basic layout 

[image]

I've already redone the setting of connection a few times, changed the logic board and the screen and tested with shorter cables, different USB and different breadboard. 

9. My setup files for reference are: 
[User_Setup and User_Setup_Select.zip](https://github.com/user-attachments/files/15687704/User_Setup.and.User_Setup_Select.zip)

10. The sketch if the basic TFT_Meter test
[TFT_Meters.zip](https://github.com/user-attachments/files/15687718/TFT_Meters.zip)

I'm new to code and at absolute beginner in electronics and I'm at my wits end with this! But I do want to get this project going. Any help is appreciated.

dart1280, 2024-06-11:
I had the same problem a while ago, I think I fixed it by using the ILI9341_2 driver....

```
#define ILI9341_2_DRIVER     
//#define ILI9341_DRIVER  

//#define TFT_WIDTH  240
//#define TFT_HEIGHT 320         
#define TFT_BACKLIGHT_ON HIGH  
#define TFT_MISO 12
#define TFT_MOSI 13 // In some display driver board, it might be written as "SDA" and so on.
#define TFT_SCLK 14
#define TFT_CS   15  // Chip select control pin
#define TFT_DC   2  // Data Command control pin
#define TFT_RST  -1  // Reset pin (could connect to Arduino RESET pin)
#define TFT_BL   21  // LED back-light
//#define TFT_RGB_ORDER TFT_RGB  // Colour order Red-Green-Blue     ---------------------
#define TFT_RGB_ORDER TFT_BGR  // Colour order Blue-Green-Red

#define TFT_INVERSION_ON   //if the colours are wrong try the other setting
// #define TFT_INVERSION_OFF   ///-------------------------------------------------------

#define TOUCH_CS 33     // Chip select pin (T_CS) of touch screen-------------this esp32 display is non-touch screen!!

#define LOAD_GLCD   // Font 1. Original Adafruit 8 pixel font needs ~1820 bytes in FLASH
#define LOAD_FONT2  // Font 2. Small 16 pixel high font, needs ~3534 bytes in FLASH, 96 characters
#define LOAD_FONT4  // Font 4. Medium 26 pixel high font, needs ~5848 bytes in FLASH, 96 characters
#define LOAD_FONT6  // Font 6. Large 48 pixel font, needs ~2666 bytes in FLASH, only characters 1234567890:-.apm
#define LOAD_FONT7  // Font 7. 7 segment 48 pixel font, needs ~2438 bytes in FLASH, only characters 1234567890:-.
#define LOAD_FONT8  // Font 8. Large 75 pixel font needs ~3256 bytes in FLASH, only characters 1234567890:-.
//#define LOAD_FONT8N // Font 8. Alternative to Font 8 above, slightly narrower, so 3 digits fit a 160 pixel TFT
#define LOAD_GFXFF  // FreeFonts. Include access to the 48 Adafruit_GFX free fonts FF1 to FF48 and custom fonts

// Comment out the #define below to stop the SPIFFS filing system and smooth font code being loaded
// this will save ~20kbytes of FLASH
#define SMOOTH_FONT

#define SPI_FREQUENCY  55000000 
#define SPI_READ_FREQUENCY  20000000
```

Gagz9k, 2024-06-12:
> I had the same problem a while ago, I think I fixed it by using the ILI9341_2 driver....
> 

Hi @dart1280 thank you for the idea, I haven't tried with the driver 2, but unfortunately didn't resolved the issue in my case ): but reading a bit from the driver 2 issue page here I think the problem might be the ILI9341 itself, as I got them from Aliexpress and, as I'm reading, this are probable clones with an issue in the ILI9341_Init.h file. 

For reference this are the ones I got: https://es.aliexpress.com/item/4001281753944.html?spm=a2g0o.order_list.order_list_main.86.30ca194ddfiHeq&gatewayAdapt=glo2esp

I will order from Adafruit a couple screens to confirm if the problem is the provider. 

In any case, thank you for the reply! It did help a lot.