Bodmer/TFT_eSPI #3358: Error compiling for board ESP32S3 Dev Module

summerfind, 2024-06-07:
I used Arduino IDE 1.8.19, compiling for ESP32S3 of the TFT_eTouch.h 's example: Conways_Life got error.

used updated TFT_eSPI.h and selected: Setup70d_ILI9488_S3_Parallel

how to fix?
Thanks
Adam

`Arduino: 1.8.19 (Windows 7), Board: "ESP32S3 Dev Module, Disabled, Disabled, QIO 80MHz, 4MB (32Mb), Core 1, Core 1, Hardware CDC and JTAG, Disabled, Disabled, Disabled, UART0 / Hardware CDC, Default 4MB with spiffs (1.2MB APP/1.5MB SPIFFS), 240MHz (WiFi), 921600, None, Disabled"

In file included from C:\Users\HUA.DELLV-PC\Documents\Arduino\libraries\TFT_eTouch-master\examples\TFT_eSPI\Conways_Life\Conways_Life.ino:17:

C:\Users\HUA.DELLV-PC\Documents\Arduino\libraries\TFT_eSPI-master_S3/TFT_eSPI.h:966:8: error: #error >>>>------>> Touch functions not supported in 8/16-bit parallel mode or with RP2040 PIO.

       #error >>>>------>> Touch functions not supported in 8/16-bit parallel mode or with RP2040 PIO.

        ^~~~~

exit status 1

Error compiling for board ESP32S3 Dev Module.

This report would have more information with
"Show verbose output during compilation"
option enabled in File -> Preferences.`

elgerg, 2024-06-07:
See answer here:
https://github.com/Bodmer/TFT_eSPI/discussions/3356#discussioncomment-9699237

summerfind, 2024-06-07:
> See answer here: [#3356 (comment)](https://github.com/Bodmer/TFT_eSPI/discussions/3356#discussioncomment-9699237)

Thanks.
I searched them, and doing the second to last solution to take in TFT_eTouch-master library, this post error right from it's example.
actually the last suggestion by Bodmer is to run: Read_User_Setup to verify the touch pin is listed, even the Read_User_Setup got the same error.

elgerg, 2024-06-07:
Oh ok,

Can you post a link to where you got Conways_Life from? It doesnt seem to be in the Bodmer repo.

Thanks

summerfind, 2024-06-07:
Sure.
https://github.com/Bodmer/TFT_eSPI/issues/3169
https://github.com/achillhasler/TFT_eTouch

elgerg, 2024-06-07:
That's a really old repo that hasnt been updated in years.

You should raise your issue with them not here.

summerfind, 2024-06-07:
Thanks.
I'll raise issue there too.

For here the question is as long as added: #define TOUCH_CS  3  // to TFT_eSPI's setup file, none sketch can compiling,  even the Read_User_Setup got compiling error: Error compiling for board ESP32S3 Dev Module.
why?

`In file included from C:\Users\HUA.DELLV-PC\Documents\Arduino\libraries\TFT_eSPI-master_S3\examples\Test and diagnostics\Read_User_Setup\Read_User_Setup.ino:18:
C:\Users\HUA.DELLV-PC\Documents\Arduino\libraries\TFT_eSPI-master_S3/TFT_eSPI.h:966:8: error: #error >>>>------>> Touch functions not supported in 8/16-bit parallel mode or with RP2040 PIO.
  966 |       #error >>>>------>> Touch functions not supported in 8/16-bit parallel mode or with RP2040 PIO.
      |        ^~~~~
exit status 1
Error compiling for board ESP32S3 Dev Module.
`

elgerg, 2024-06-08:
Please see original answer. This lib only supports the touch controller when it shares the spi bus with the screen. You are using a parallel display..