Bodmer/TFT_eSPI #3364: Please add support for the drive controller ST7262

appleshaman, 2024-06-11:
I have one 480*800 screen with an ST7262 drive, but I could not find this drive type in the custom config file; please add them; thanks a lot

eagl1, 2024-06-16:
Me too, I'm trying to develop the code for this. But want to know some stuff about the specs because it's not like the other displays. It uses 16-bit RGB parallel data lines.

andreoliveirapguerra, 2024-08-23:
@eagl1 @appleshaman  For those please check https://github.com/rzeldent/esp32-smartdisplay.

eagl1, 2024-08-29:
Yeah I know that library and there is another arduino library:

[https://www.arduino.cc/reference/en/libraries/gfx-library-for-arduino/](url)

FXVT, 2024-12-13:
Bonjour, I also have a board based on the ST72762 (https://www.waveshare.com/wiki/ESP32-S3-Touch-LCD-5), but I can't get it to work except with the pre-made LVGL examples. However, I would prefer to use TFT_eSPI. Do you plan to update this excellent library to make it compatible with the ST72762? I tried to figure out how to modify the library, but it's too complicated for me. I get lost in the many hardware and software parameters. Thank you.

jakubkar1, 2024-12-27:
First of all, I want to thank you for the TFT_eSPI library. It's great. I've been using it for many years and I have a lot of code in it. I just bought a Waveshare 7" ST7262 display ( https://www.waveshare.com/wiki/ESP32-S3-Touch-LCD-7#Environment_Setting ) and I wanted to get it working with this library. I only managed to do it for LVGL, but I don't want to reprogram everything again. So I'm also joining the request for implementation in TFT_eSPI.  :)

eagl1, 2024-12-27:
Hi,

This is the `Arduino_GFX` library I've deleted most of the author source code and kept only required code for the ST7262. I'm preparing this source code to merge it with `TFT_eSPI` library but I haven't completed the work.

This is the original library:
[https://github.com/moononournation/Arduino_GFX/tree/master](url)

[Arduino_GFX_TFT_eSPI.zip](https://github.com/user-attachments/files/18264333/Arduino_GFX_TFT_eSPI.zip)

jakubkar1, 2024-12-28:
First of all, I want to thank you for your quick response. I installed the Arduino_GFX-1.5.0 library into my project,
copied your files to Arduino_GFX-1.5.0\src and ran Arduino_GFX-1.5.0\examples\PDQgraphicstest.
The code obviously ran (Serial.prints were OK), but the screen was black. 

`paltformio.ini setting:
[env:esp32-s3-devkitc-1]
platform = espressif32
board = esp32-s3-devkitc-1
framework = arduino
monitor_speed = 115200`

I searched a bit in the code and came across Arduino_GFX_Library.h and the valid setup code:

`#elif defined(ESP32) && (CONFIG_IDF_TARGET_ESP32S3)
#define DF_GFX_SCK 36
#define DF_GFX_MOSI 35
#define DF_GFX_MISO GFX_NOT_DEFINED
#define DF_GFX_CS 40
#define DF_GFX_DC 41
#define DF_GFX_RST 42
#define DF_GFX_BL 48
`
and it seems to me that it doesn't match my 7" display, see.  https://www.waveshare.com/wiki/ESP32-S3-Touch-LCD-7#Interface_Description

Displays are not devices I would like to deal with, I now use programming as a brain exercise and to fulfill my Arduino HW projects,
even though I am a programmer and I wrote my first program sometime in 1972, yes 1972 and the computer was still a vacuum tube...
Displays are already complex devices for me :) and I don't understand at all how a large open source project can be maintained.
So if my communication is not in accordance with the rules, I apologize.
Don't delay because of me and devote yourself to systematic work on your project, although your helpfulness is nice.
I hope I can still manage to reprogram into lvgl :)

eagl1, 2024-12-28:
Send me your arduino project folder. And let me check, your have to comment/uncommeent the display constructor function. And set the required parameters that fit your display specs.

eagl1, 2024-12-28:
Another thing to add, you must install ESP32 2.0.17 board manager version, otherwise the code won't work.

[image]

jakubkar1, 2024-12-28:
[Displej_ESP32_S3_GFX.zip](https://github.com/user-attachments/files/18266615/Displej_ESP32_S3_GFX.zip)
here is the project, but as I see, the problem will probably be that I am using VSC and platformio. Arduino ide is friendlier in some things.

eagl1, 2024-12-28:
Yes, I want you to do the following:

1. Run the project in Arduino IDE.
2. From your running code on VS or in Espriff IDE, copy the display setting, I prefer the one from espriff lvgl; like this:

[image]

First compile the project and upload it to the display, if it worked, then:

From this file `rgb_lcd_example_main`, copy the rgb settings from line 25 to 49, and edit the matching parameters to the one in the arduino code in the `Arduino_GFX_dev_device.h` file, like this:

Here, I removed all the other displays constructors and kept the working ones for my displays. I have 2 ST7262 rgb displays and 1 st7796s.

```
/*  Library settings:
 *  1. CH340 2019 works
 *  2. this one worked with esp32 2.0.17 board package
*/
//#define ESP32_3248S035    
//#define ESP32_8048S043 
#define ESP32_EA7013

///////////////////////////////////////////////////////////////////////////////////////////////////////////////
#if defined(ESP32_8048S043)
#define GFX_DEV_DEVICE ESP32_8048S043
#define GFX_BL 2
Arduino_GFX_ESP32RGBPanel *rgbpanel = new Arduino_GFX_ESP32RGBPanel(
    40 /* DE */, 41 /* VSYNC */, 39 /* HSYNC */, 42 /* PCLK */,
    45 /* R0 */, 48 /* R1 */, 47 /* R2 */, 21 /* R3 */, 14 /* R4 */,
    5 /* G0 */, 6 /* G1 */, 7 /* G2 */, 15 /* G3 */, 16 /* G4 */, 4 /* G5 */,
    8 /* B0 */, 3 /* B1 */, 46 /* B2 */, 9 /* B3 */, 1 /* B4 */,
    0 /* hsync_polarity */, 8 /* hsync_front_porch */, 4 /* hsync_pulse_width */, 8 /* hsync_back_porch */,
    0 /* vsync_polarity */, 8 /* vsync_front_porch */, 4 /* vsync_pulse_width */, 8 /* vsync_back_porch */,
    1 /* pclk_active_neg */, 16000000 /* prefer_speed */);
Arduino_GFX_RGB_Display *gfx = new Arduino_GFX_RGB_Display(
    800 /* width */, 480 /* height */, rgbpanel, 0 /* rotation */, true /* auto_flush */);

///////////////////////////////////////////////////////////////////////////////////////////////////////////////
#elif defined(ESP32_3248S035)
#define GFX_DEV_DEVICE ESP32_3248S035
#define GFX_BL 27
Arduino_GFX_DataBus *bus = new Arduino_ESP32SPI(2 /* DC */, 15 /* CS */, 14 /* SCK */, 13 /* MOSI */, 12 /* MISO */, HSPI /* spi_num */);
Arduino_GFX_Core_Library *gfx = new Arduino_ST7796(bus, GFX_NOT_DEFINED /* RST */, 0 /* rotation */);

///////////////////////////////////////////////////////////////////////////////////////////////////////////////
#elif defined(ESP32_EA7013)
#define GFX_DEV_DEVICE ESP32_EA7013
#define GFX_BL 1
Arduino_GFX_ESP32RGBPanel *rgbpanel = new Arduino_GFX_ESP32RGBPanel(
    5 /* DE */, 2 /* VSYNC */, 4 /* HSYNC */, 6 /* PCLK */,
    7 /* R0 */, 15 /* R1 */, 16 /* R2 */, 8 /* R3 */, 3 /* R4 */,
    46 /* G0 */, 9 /* G1 */, 10 /* G2 */, 11 /* G3 */, 12 /* G4 */, 13 /* G5 */,
    14 /* B0 */, 21 /* B1 */, 47 /* B2 */, 48 /* B3 */, 45 /* B4 */,    

    0 /* hsync_polarity */, 10 /* hsync_front_porch */, 16 /* hsync_pulse_width */, 32 /* hsync_back_porch */,
    0 /* vsync_polarity */, 15 /* vsync_front_porch */, 3 /* vsync_pulse_width */, 12 /* vsync_back_porch */,
    1 /* pclk_active_neg */, 15000000 /* prefer_speed */);
    
Arduino_GFX_RGB_Display *gfx = new Arduino_GFX_RGB_Display(
    800 /* width */, 480 /* height */, rgbpanel, 0 /* rotation */, true /* auto_flush */);

#endif

```

Check the parameters and edit them accordingly. If didn't work for you, copy your working project and send it.

jakubkar1, 2024-12-29:
Unfortunately, after reinstalling esp32 espressive to a lower version in Arduino IDE (be careful, the display has 16MB FLASH and OPI PSRAM, so you need to adjust the settings in Tools), my other projects crashed again and
the result of the translations is:

c:\Users\..\Documents\Arduino\libraries\ESP32_Display_Panel\src\lcd\ESP_PanelLcd.cpp:13:10: fatal error: esp_memory_utils.h: No such file or directory
#include "esp_memory_utils.h".

So the effort was worth it.

I decided to substitute the code I wrote using TFT_eSPI commands using the lvgl library.
It should be done quickly, I mostly use text statements of various measurements. But now I have other work on HW, so I'll see how it goes when I get to it.

I used the library:
https://github.com/rzeldent/esp32-smartdisplay/tree/main
where the various boards are precisely described by their production code and you can easily enter them in platformio.ini, which is also included with the library.
The library is made mainly for Chinese Sunton Smart display boards, aka CYD (Cheap Yellow Display). They are compact displays
with ESP32 for which you can easily buy a simple acrylic case and are therefore ideal as simple display stations for me.
And my 7" board from Waveshare is also included there. The code of my board is esp32-s3touchlcd7.json.

I don't understand displays, they seem very complicated to me and I believe that if you look at the relevant settings of esp32-s3touchlcd7 then you will find everything you need there.

eagl1, 2024-12-29:
Yes, I have couple of those CYD, they worked for me with platformio projects with no problems, but it's difficult for me to design the gui with those VS code platform.

What I meant by installed earlier version of the arduino esp32 board set, is that it's the only way currently to run a code using Arduino_GFX library.

But if you have other projects requiring newer version of esp32 board package, then that's going to be an issue for you.

Anyway, I was working on meging the Arduino_GFX functions into TFT_eSPI library but I left the project for now as I have other projects to work on.

jakubkar1, 2024-12-30:
I wish you that your plans come true.

giri-v, 2025-01-05:
I too have the waveshare ESP32 4.3 lcd board and I've tried several solutions, but since I'm also using platformio, I'm not having much success.

I've used this amazing library for all my other projects. I'd love to use it on this one too.

Please add support for the ST7262. Here is the wiki for the HW: https://www.waveshare.com/wiki/ESP32-S3-Touch-LCD-4.3

jakubkar1, 2025-01-05:
yes, I also have a struggle with the board https://www.waveshare.com/wiki/ESP32-S3-Touch-LCD-7. For me - I will never buy Waveshare for platformio again. I don't have time to research what version of individual system parts I have installed and reach into the system to put the board into operation and then stare at a black screen again during the next system update. I need the display for work and don't have it as a separate hobby. It disappointed me greatly.