Bodmer/TFT_eSPI #3366: About using the ST7796 on the ESP32 using an 8Bit interface

CJT1111, 2024-06-12:
Now I use ESP32 and ST7796 （320*480 spi or 8Bit hardware interface screen）for display control, and the interface is an 8bit interface. I see the 8bit interface definition described in User_Setup.h file (indicating that ILI9481 and ILI9341 have been tested, but ST7796 is not available). And then my User_Setup file looks like this:
`
// 驱动定义
#define ST7796_DRIVER

// 颜色定义
#define TFT_RGB_ORDER TFT_BGR // Colour order Blue-Green-Red
#define TFT_INVERSION_ON

// 尺寸定义
#define TFT_WIDTH 320
#define TFT_HEIGHT 480

// 管脚定义(ESP32)
// #define USE_HSPI_PORT // 默认使用VSPI
// #define TFT_MISO -1
// #define TFT_MOSI 15
// #define TFT_SCLK 16
// #define TFT_CS 17
// #define TFT_DC 18
// #define TFT_RST -1
// 8Bit接口定义
#define TFT_PARALLEL_8_BIT //定义为8Bit接口(默认为SPI接口)
#define TFT_CS 33 
#define TFT_DC 15 
#define TFT_RST 32
#define TFT_WR 4 
#define TFT_RD 2 
#define TFT_D0 12 
#define TFT_D1 13
#define TFT_D2 26
#define TFT_D3 25
#define TFT_D4 17
#define TFT_D5 16
#define TFT_D6 27
#define TFT_D7 14

// 字体定义
#define LOAD_GLCD  // 字高8 Font 1. Original Adafruit 8 pixel font needs ~1820 bytes in FLASH
#define LOAD_FONT2 // 字高16 Font 2. Small 16 pixel high font, needs ~3534 bytes in FLASH, 96 characters
#define LOAD_FONT4 // 字高26 Font 4. Medium 26 pixel high font, needs ~5848 bytes in FLASH, 96 characters
// #define LOAD_FONT6 //字高48(仅数字)Font 6. Large 48 pixel font, needs ~2666 bytes in FLASH, only characters 1234567890:-.apm
// #define LOAD_FONT7 //字高48.数码管(仅数字) Font 7. 7 segment 48 pixel font, needs ~2438 bytes in FLASH, only characters 1234567890:-.
// #define LOAD_FONT8 //字高75(仅数字) Font 8. Large 75 pixel font needs ~3256 bytes in FLASH, only characters 1234567890:-.
// #define LOAD_FONT8N // Font 8. Alternative to Font 8 above, slightly narrower, so 3 digits fit a 160 pixel TFT
//  #define LOAD_GFXFF // FreeFonts. Include access to the 48 Adafruit_GFX free fonts FF1 to FF48 and custom fonts

// SPI定义
#define SPI_FREQUENCY 80000000
`
And then the Arduino program is written like this

`
#include <SPI.h>
#include <TFT_eSPI.h>
#include <Adafruit_GFX.h>
TFT_eSPI tft = TFT_eSPI();                 // 引脚请自行配置tft_espi库中的 User_Setup.h文件
TFT_eSprite tft_part = TFT_eSprite(&tft);  // 创建局部区域缓冲区

unsigned long TimeA;
unsigned long TimeB;

void setup() {
  Serial.begin(9600);
  Serial.println("测试开始");
  tft.begin();         // TFT初始化
  tft.setRotation(0);  // 旋转角度设置
  delay(1000);         // 等待启动
}

void loop() {

  TimeA = micros();  // 记录开始时间
  // 局部刷新快速显示技术
  tft_part.createSprite(200, 200);  // 创建一个局部空间(x,y),注意不可是全局
  for (int i = 1; i <= 100; i++) {
    tft_part.fillScreen(TFT_GREEN);  //全屏由上而下刷新为绿色
    tft_part.pushSprite(0, 0);       // Sprite中内容一次推向屏幕(位置X,位置Y)
  }
  tft_part.deleteSprite();  // 删除Sprite
  TimeB = micros();         // 记录结束时间

  unsigned long elapsedTime = TimeB - TimeA;               // 计算经过的时间
  double fps = (double)(100 * 1000 * 1000) / elapsedTime;  // 计算帧率

  Serial.print("测试时间: ");
  Serial.print(elapsedTime);
  Serial.print(" 微秒, 测试帧数: ");
  Serial.println(fps);

  delay(500);
}
`
But the display result is black screen, I don't know whether ESPI supports 8Bit hardware interface of ST7796, or there is a problem with my code. Please take a look.

This screen in M0, M1, M2 are high power level SPI interface, using ESPI can drive, I have tried. However, when M0 and M1 are set to high level, M2 is set to low level, and the screen works in 8bit hardware interface mode, there is no result driver.

In addition, the screen hardware information is as follows：
[
[ST7796S_SPEC_V1.0.pdf](https://github.com/user-attachments/files/15807382/ST7796S_SPEC_V1.0.pdf)
](url)

eagl1, 2024-06-19:
I recommend you to use `GFX_Library_for_Arduino-1.4.7`, there's both 8-bit/16-bit interface for different microcontrollers.

CJT1111, 2024-06-26:
> I recommend you to use `GFX_Library_for_Arduino-1.4.7`, there's both 8-bit/16-bit interface for different microcontrollers.

But there is a problem that LVGL only supports ESPI libraries, so I have to use ESPI if I want to use LVGL

eagl1, 2024-06-26:
OK, how about this:

1. Copy one of these 8-bit parallel config files for your esp32:
`Setup31_ST7796_Parallel_STM32` or `Setup104_RP2040_ST7796_parallel` and rename it as you like.
2. Change whatever necessary to fit your setup, mainly the pin arrangement.
3. Don't forget to use the new config file in the main `User_Setup_Select.h`.