Bodmer/TFT_eSPI #3384: ESP32 C3 SuperMini ST7735 1.8" screen. All examples fail. Serial Monitor Shows an Error

tgirard, 2024-06-30:
When I try to use the ESP32 C3 Super Mini with the St7735 Driver, I am never able to get anything on the screen and there seems to be an error that comes up on Example sketches that use the Serial Monitor. My Setup (Pins and Driver) Work fine with the Adafuit driver 
( I really REALLY Don't want to use the Adafruit setup)

I've included:
User_Setup.h (I only changed the driver and pins)
Output from Upload.txt  (This was from 'Output' from the IDE when uploading to the processor)
SerialMonitor Error Information.txt (This is what the Serial Monitor was continuously repeating on Examples that used it)

Let me know if there's anything else I can capture for you :)

IDE: Arduino 2.3.2 Windows 11 
TFT_ESPI: 2.5.43 (I did a clean Install 6/30/2024)
Boards: ESP32 by Espresif 3.0.2
Processor: ESP32 C3 Super Mini From AliExpres
https://www.aliexpress.us/item/3256806413352402.html?spm=a2g0o.order_list.order_list_main.55.5d651802WPL50h&gatewayAdapt=glo2usa
TFT driver: ST7735
Interface type: SPI

I'm more than happy to give you one of these boards if you want one. Just shoot me an address. I'm in the US.

[TFT_ESPI_ERROR_DOCS.zip](https://github.com/user-attachments/files/16045685/TFT_ESPI_ERROR_DOCS.zip)

IAmOrion, 2024-07-01:
+1 - came here with same problem! 

Additional info...

Serial output:

```
ESP-ROM:esp32c3-api1-20210207
Build:Feb  7 2021
rst:0x3 (RTC_SW_SYS_RST),boot:0xd (SPI_FAST_FLASH_BOOT)
Saved PC:0x40048b82
SPIWP:0xee
mode:DIO, clock div:1
load:0x3fcd5820,len:0x458
load:0x403cc710,len:0x814
load:0x403ce710,len:0x2880
entry 0x403cc710
Guru Meditation Error: Core  0 panic'ed (Store access fault). Exception was unhandled.

Core  0 register dump:
MEPC    : 0x420060a4  RA      : 0x42006094  SP      : 0x3fca2d50  GP      : 0x3fc8d400  
TP      : 0x3fc8e8dc  T0      : 0x00000000  T1      : 0xffc3ffff  T2      : 0x00000000  
S0/FP   : 0x60004000  S1      : 0x3fc8e014  A0      : 0x00000000  A1      : 0x00001001  
A2      : 0x00000000  A3      : 0x00000001  A4      : 0x08000000  A5      : 0x00000010  
A6      : 0x00000003  A7      : 0x00000001  S2      : 0x00000000  S3      : 0x00000000  
S4      : 0x00000000  S5      : 0x00000000  S6      : 0x00000000  S7      : 0x00000000  
S8      : 0x00000000  S9      : 0x00000000  S10     : 0x00000000  S11     : 0x00000000  
T3      : 0x00000000  T4      : 0x00000000  T5      : 0x00000041  T6      : 0x0000000f  
MSTATUS : 0x00001881  MTVEC   : 0x40380001  MCAUSE  : 0x00000007  MTVAL   : 0x00000010  
MHARTID : 0x00000000  

Stack memory:
3fca2d50: 0x00000003 0x00000000 0x02625a00 0x00000001 0x00000001 0x3fc9a000 0x3fc8e014 0x42006522
3fca2d70: 0x00000000 0x3fc8cc88 0x00000001 0x4200b514 0x3fc8e000 0x00000000 0x00000000 0x4200010e
3fca2d90: 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000
3fca2db0: 0x00000000 0x00000000 0x00000000 0x4200ab0c 0x00000000 0x00000000 0x00000000 0x403885b2
3fca2dd0: 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000
3fca2df0: 0x00000000 0xa5a5a5a5 0xa5a5a5a5 0xa5a5a5a5 0xa5a5a5a5 0xa5a5a5a5 0xa5a5a5a5 0xbaad5678
3fca2e10: 0x00000160 0xabba1234 0x00000154 0x3fca2d60 0xd58f349e 0x3fc9f4bc 0x3fc9a36c 0x3fca2e1c
3fca2e30: 0x3fc9a364 0x00000018 0xd58b2538 0x69c7ec2d 0x3fca2e1c 0x00000000 0x00000001 0x3fca0e0c
3fca2e50: 0x706f6f6c 0x6b736154 0x208faf00 0x00bfe61a 0x00000000 0x3fca2e00 0x00000001 0x00000001
3fca2e70: 0x00000000 0x00000000 0x00000000 0x3fc9c160 0x3fc9c1c8 0x3fc9c230 0x00000000 0x00000000
3fca2e90: 0x00000001 0x00000000 0x00000000 0x00000000 0x42025e86 0x00000000 0x00000000 0x00000000
3fca2eb0: 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000
3fca2ed0: 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000
3fca2ef0: 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000
3fca2f10: 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000
3fca2f30: 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000
3fca2f50: 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x08000000
3fca2f70: 0xbaad5678 0x0001d085 0x3fc9ba34 0x3fc9ba34 0x3c791415 0x2c46b6a7 0xf62a7d6f 0x1ac53459
3fca2f90: 0xc68a38d4 0x310daea7 0x586e6a08 0xf69878b7 0x0889c5c9 0xa346da95 0xb1354af8 0x9e71271e
3fca2fb0: 0x10884aa6 0x6a57d999 0x9fb77668 0x4d1605b1 0x4f1aa3b1 0x69fce350 0xdf447be0 0x6c36c65d
3fca2fd0: 0x7e72fd47 0x1ce7199b 0x8f88bbce 0x2fe1b3dc 0xcfafe14f 0x229eb9e9 0x8f10af1c 0x1b1592e2
3fca2ff0: 0x6a272455 0xf0eea32d 0xb3958680 0xbf8c93ee 0x059662e8 0x696f0b8e 0x03e83a70 0x24917b4c
3fca3010: 0xd2740bfe 0x5ca7de36 0x060935fe 0x8f73b77b 0x2ce615d3 0xa8a966d3 0x4fcd27e1 0xe465cfcd
3fca3030: 0xb4c0b19b 0x6d264e90 0x35a4dbbf 0x67937682 0x4d98989d 0x4b267516 0xfbd94d99 0x605ddbc6
3fca3050: 0x8c66dc6f 0xae79d6a4 0xce4495eb 0x7fc3b819 0x35130d8e 0x2a42d470 0x3431d6de 0x849c758a
3fca3070: 0x12f675cf 0xa8537c9e 0x12dc21e3 0xd990e846 0x7b541c3f 0x16ce7dda 0x994d2adc 0x580454ed
3fca3090: 0x2453622b 0x3d3fef9d 0x0b1ee1c2 0x41ec437b 0x60f84265 0x9648ad89 0x929aa738 0xc9c28eba
3fca30b0: 0x42e111df 0xe9fb9c80 0x8282ab7d 0x73c7d17b 0xb8b56771 0xf20d7fa8 0xea54e41b 0x62045afe
3fca30d0: 0x2ecd102a 0x77e34984 0x834301cb 0x66423dab 0xcb12e0f0 0xca5f0551 0x7c362135 0xc6cf55e2
3fca30f0: 0x5e54d87d 0x59152a99 0x492d71ca 0xb0b21ec6 0x38b921b6 0xd6cb6ef1 0xa25ebedd 0x4268164c
3fca3110: 0x3cf39d94 0x5fd8f208 0x14ba4ff4 0xa19f434c 0xdb8af573 0x7c776103 0x3e2ecafc 0x733fb924
3fca3130: 0xd2a429b0 0xb37f5fa9 0x63df536c 0x27bfff47 0x4951df17 0x20da37ed 0x2f7b196a 0xb1661259

ELF file SHA256: 435455b14a6afb6a
```

Stack Trace decode gives me:

```
0x403885b2: vPortTaskWrapper at /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/freertos/FreeRTOS-Kernel/portable/riscv/port.c:205
```

[ESP32C3_SUPER_MINI_CAN_BUS_GEAR_KNOB_TFT_eSPI.ino.elf.zip](https://github.com/user-attachments/files/16055530/ESP32C3_SUPER_MINI_CAN_BUS_GEAR_KNOB_TFT_eSPI.ino.elf.zip)

JiriBilek, 2024-07-01:
There is a problem with SPI host numbering. Maybe something was changed in Arduino_ESP 3.x, I don't know, I didn't compare the v2 and v3 defines.
Try to change the line
https://github.com/Bodmer/TFT_eSPI/blob/fae22f785f8b4f7970195487826a348d08d082ae/Processors/TFT_eSPI_ESP32_C3.h#L71
to

```
#if ESP_ARDUINO_VERSION_MAJOR < 3
  #define SPI_PORT SPI2_HOST
#else
  #define SPI_PORT 2
#endif
```

IAmOrion, 2024-07-01:
> ```
> #if ESP_ARDUINO_VERSION_MAJOR < 3
>   #define SPI_PORT SPI2_HOST
> #else
>   #define SPI_PORT 2
> #endif
> ```

Well, that fixes it crashing, however it breaks stuff and doesn't work ha.

LittleFS stopped working and whilst the ESP32C3 doesn't crash, nothing get's displayed on the lcd.

LittleFS works fine when not using TFT_eSPI and I can display fine to the LCD using Adafruit ST7789 library so the pins used in code and the TFT_eSPI setup files are correct

JiriBilek, 2024-07-01:
Today, I connected ILI9163 128x128 TFT and with this fix it works (i.e. displays correctly).

JiriBilek, 2024-07-01:
Just a thought (I fell into this "trap" in the past): isn't it a problem with backlight?

IAmOrion, 2024-07-01:
No I don't believe so, it's probably something funky going on with the lcd.

[http://www.lcdwiki.com/0.96inch_IPS_ST7789_Module](http://www.lcdwiki.com/0.96inch_IPS_ST7789_Module)

They provide some files that the manual says you need to replace, it's possible something isn't quite right in it, or there's been "breaking changes" since ver 2.5.23 that they include with their download

JimDrewGH, 2024-07-01:
What version of the Arduino core files are you using?  Anything beyond v2.0.14 does not work with any of the ESP32's except the original.. so ESP32-C3, ESP32-S2, ESP32-S3, ESP32-Hx, etc. all fail when doing the init().  We are hoping that there is a fix for this, but in the mean time you can just roll back your core files and everything works fine.

JiriBilek, 2024-07-01:
esp32 core 3.0.2.

IAmOrion, 2024-07-01:
Arduino Core? Or did you mean ESP32 Core? I'm confused! 

ESP32 Boards is 3.0.2

IAmOrion, 2024-07-01:
> esp32 core 3.0.2.

Ohh, so you're suggesting I roll back the ESP32 core, ok, I'll report back shortly

JiriBilek, 2024-07-01:
No, ESP core is the most recent. Arduino IDE 2.3.2, also the most recent

IAmOrion, 2024-07-01:
ooh, I use Arduino IDE 1.8.19 because I still hate the UI of 2.x.x

tgirard, 2024-07-02:
So it seems to be the mini and not the screen or the driver. I used the same settings I used for the ST7735 but for the ILI9341 and still, the same error when I upload:

Guru Meditation Error: Core  0 panic'ed (Store access fault). Exception was unhandled.

Does Bodmer still work on this? I sure hope so. :)

JiriBilek, 2024-07-02:
The fix I posted is necessary at least for esp32 library v3+ (didn't test it for older cores). 
Here `SPI2_HOST == 1`, thus `SPI_PORT == 1` and this is used in expression
`volatile uint32_t* _spi_cmd       = (volatile uint32_t*)(SPI_CMD_REG(SPI_PORT));`
giving `_spi_cmd == nullptr` that obviously causes exception on dereferencing.

I didn't meet any other problems.
My setup is: custom ESP-C3 board and ILI9163 128x128 TFT 1.44" display.

tgirard, 2024-07-03:
Hey JiriBilek, What file are you modifying and where? could you post a snippet and show me (us) what file you changed ? I am a semi-noob but can usually figure it out with a little help. 
FYI, I tried to use the M5StampS3 and got the same error with the same setup I posted above.

JimDrewGH, 2024-07-03:
> Arduino Core? Or did you mean ESP32 Core? I'm confused!
> 
> ESP32 Boards is 3.0.2

Yes, you need to change the board file back to v2.0.14.  This problem occurs with v2.0.15 and later, so it's be going on since perhaps last year?  I am not sure when v2.0.15 was released, but that is the version that first breaks any ESP32 that is not the original version.

JimDrewGH, 2024-07-03:
> The fix I posted is necessary at least for esp32 library v3+ (didn't test it for older cores). Here `SPI2_HOST == 1`, thus `SPI_PORT == 1` and this is used in expression `volatile uint32_t* _spi_cmd = (volatile uint32_t*)(SPI_CMD_REG(SPI_PORT));` giving `_spi_cmd == nullptr` that obviously causes exception on dereferencing.
> 
> I didn't meet any other problems. My setup is: custom ESP-C3 board and ILI9163 128x128 TFT 1.44" display.
> 
> Edit: Just for curiosity: from idf folder in the core: `#define SPI_CMD_REG(i) (REG_SPI_BASE(i) + 0x0)` `#define REG_SPI_BASE(i) (((i)==2) ? (DR_REG_SPI2_BASE) : (0)) // only one GPSPI`

What exact file are you changing?  You didn't code format your message, so it is difficult to understand.

JiriBilek, 2024-07-03:
I am changing only the file `TFT_eSPI_ESP32_C3.h`, look at my first post here.

The other code snippets show the source of the problem deep in the IDF folder. I posted it here only as a clue for somebody who'd want to know more about the problem.

tgirard, 2024-07-04:
I made the change recommended by JiriBilek and had the same result as others. The fix stops the crash but, for me anyways (ST7735 1.8 Display) the graphics never show up. I'm gonna have to use Adafruit for now which sucks... Still has a lot of the features but it's so bloated and cumbersome... And NO presets... ACK

JiriBilek, 2024-07-04:
I'm sorry I couldn't help.
Today, I found a 128x160 display with ST7735 and it's running smoothly both with ST7735 and ILI9163 drivers (the chips are very similar). I tried the Pong_v3 demo.
So I suspect the problem is somewhere else.

mboehmerm, 2024-07-05:
For my ESP32-C3 i use :
Flash Mode: "QIO" -> Flash Mode: "DIO"

tgirard, 2024-07-06:
Thanks Jiri for trying to help. I really do appreciate it and I learned a little bit about TFT_ESPI :) I can only hope that a fix may come in from someone or Bodmer if he is still working on this at all ( I hope so, it's a awesome tool when it's working)

JimDrewGH, 2024-07-06:
> I made the change recommended by JiriBilek and had the same result as others. The fix stops the crash but, for me anyways (ST7735 1.8 Display) the graphics never show up.

Yes, same for me as well.  The ONLY working solution at this point is to roll back your core library to v2.0.14 (v6.60 for PlatformIO).  All later versions do not work with any of the ESP32 family *except* the original ESP32.

mboehmerm, 2024-07-15:
I have updated my post. I hope the workaround works for you too.

JiriBilek, 2024-07-15:
> First i edited in v3.0.2 the file
> 
>     C:\Users\username\AppData\Local\Arduino15\packages\esp32\tools\esp32-arduino-libs\idf-release_v5.1-bd2b9390ef\esp32c3\include\soc\esp32c3\include\soc\soc.h
> 
> and changed
> 
> #define REG_SPI_BASE(i) (((i)==2) ? (DR_REG_SPI2_BASE) : (0))   // only one GPSPI 
> 
> (back) to
> 
> #define REG_SPI_BASE(i) (((i)==2) ? (DR_REG_SPI2_BASE) : (DR_REG_SPI0_BASE - ((i) * 0x1000)))
> 
> as it was before in version 2.0.14 and exists in the soc.h of S2, S3, H2, C3 and C6 but not ESP32.
> 
>

Just a note:
Speaking about C3 this change is not necessary, for `i==2` gives the same result and according to the ESP32-C3 Technical Reference Manual (chapter 25.1): _SPI0 and SPI1 controllers are primarily reserved for internal use_.

mboehmerm, 2024-07-18:
But without this change it doesn't work for me. For whatever reason.

By the way, this also works for me with a (Nologo) ESP32-C3 Super Mini and now also with the ESP32-C6. However, further changes are necessary to make the code running on C3, C6 and probably H2.

Dag0d, 2024-07-30:
> For the **ESP32-S3** and v3.0.2 i added to my User_Setup.h :
> 
> ```
> #define USE_HSPI_PORT // HSPI pins 10-13
> //#define USE_FSPI_PORT // FSPI pins (34)-37 // USE_FSPI_PORT must be defined for ESP32-S3
> ```

This fixed it for me with the Waveshare ESP32-S3-Touch-LCD-1.28.
Thanks for finding that solution.

Any chance we will see the fix in the library anytime soon?

hitecSmartHome, 2024-08-12:
Nothing works for me.
I have applied the changes, my esp does not crash but can't display anything.

- Latest Arduino
- ESP-WROVER-E ( 8mb psram 16mb flash )

Basitadam, 2024-09-25:
> I have updated my post. I hope the workaround works for you too.

Worked for me, too.
Thanks

tonbor, 2024-10-10:
> Downgrading esp32 board file to v2.0.14 works for me too.
> 
> For the **ESP32-S2** and v3.0.2 i added to my User_Setup.h :
> 
> ```
> #define USE_HSPI_PORT` // HSPI pins 10-13 // Comment this line for FSPI ( Pins 34-37 )
> ```
> 
> For the **ESP32-S3** and v3.0.2 i added to my User_Setup.h :
> 
> ```
> #define USE_HSPI_PORT // HSPI pins 10-13
> //#define USE_FSPI_PORT // FSPI pins (34)-37 // USE_FSPI_PORT must be defined for ESP32-S3
> ```
> 
> Solution for **ESP32-C3** and v3.0.2 found here : [espressif/arduino-esp32#9618](https://github.com/espressif/arduino-esp32/issues/9618)
> 
> ```
> The change done in IDF to 
> #define REG_SPI_BASE(i) (((i)>=2) ? (DR_REG_SPI2_BASE + (i-2) * 0x1000) : (0)) // GPSPI2 and GPSPI3
> is wrong. Correct is
> #define REG_SPI_BASE(i) (((i)==2) ? (DR_REG_SPI2_BASE) : (DR_REG_SPI0_BASE - ((i) * 0x1000))) // GPSPI2 and GPSPI3
> Changed this in our IDF fork for Tasmota, which is the base for compiling the Tasmota Arduino libs.
> Edit: The changes done for the other SOCs are imho wrong too. Only ESP32 looks correct.
> ```
> 
> ### ~First i edited in v3.0.2 the file~
> > C:\Users\username\AppData\Local\Arduino15\packages\esp32\tools\esp32-arduino-libs\idf-release_v5.1-bd2b9390ef\esp32c3\include\soc\esp32c3\include\soc\soc.h
> 
> ~and changed~
> 
> ```
> #define REG_SPI_BASE(i) (((i)==2) ? (DR_REG_SPI2_BASE) : (0))   // only one GPSPI 
> ```
> 
> ~(back) to~
> 
> ```
> #define REG_SPI_BASE(i) (((i)==2) ? (DR_REG_SPI2_BASE) : (DR_REG_SPI0_BASE - ((i) * 0x1000)))
> ```
> 
> ~as it was before in version 2.0.14 and exists similar in the soc.h of S2, S3, H2, C3 and C6 but not ESP32.~
> 
> ### ~Second i did the changes mentioned above in~
> ### I modified only this file
> > C:\Users\username\Documents\Arduino\libraries\TFT_eSPI\Processors\TFT_eSPI_ESP32_C3.h
> 
> and changed
> 
> ```
>  #define SPI_PORT SPI2_HOST 
> ```
> 
> to
> 
> ```
> #if ESP_ARDUINO_VERSION_MAJOR < 3
>   #define SPI_PORT SPI2_HOST
> #else
>   #define SPI_PORT 2
> #endif
> ```
> 
> Then i replaced
> 
> ```
>   #ifndef REG_SPI_BASE
>     #define REG_SPI_BASE(i) DR_REG_SPI2_BASE
>   #endif
> ```
> 
> with
> 
> ```
>   #ifdef REG_SPI_BASE
>     #undef REG_SPI_BASE
>   #endif
> 
>   #define REG_SPI_BASE(i) (((i)==2) ? (DR_REG_SPI2_BASE) : (DR_REG_SPI0_BASE - ((i) * 0x1000))) // GPSPI2 and GPSPI3 
> ```
> 
> Now the ST7789 display works with my Tasmota ESP32-C3, ESP32-S2 mini and ESP32-S3.
> 
> **TFT_eSPI is a fantastic and fast library.**
> 
> I hope there is soon a better solution/update for this problem and support for ESP32-H2 and C6.

No idea what to change for S3, TFT_eSPI_ESP32_C3.h is not TFT_eSPI_ESP32_S3.h

mboehmerm, 2024-10-10:
For the **ESP32-S3** and v3.0.2 i added to my User_Setup.h :
```
 #define USE_HSPI_PORT // HSPI pins 10-13
```

tonbor, 2024-10-15:
> For the **ESP32-S3** and v3.0.2 i added to my User_Setup.h :
> 
> ```
>  #define USE_HSPI_PORT // HSPI pins 10-13
> ```

Not working, anything else to adapt?

Basitadam, 2024-10-16:
Are you using ESP32_C3?
Which lines are uncommented in your User_Setup_Select.h file?

tonbor, 2024-10-16:
> Are you using ESP32_C3? Which lines are uncommented in your User_Setup_Select.h file?

Esp32 S3

Basitadam, 2024-10-16:
Which lines are uncommented in your User_Setup_Select.h file?

tonbor, 2024-10-16:
Made my own, and selected in TFT_eSPI.h No restart anymore.
// Setup for the ESP32 S3 with ILI9341 display
// Note SPI DMA with ESP32 S3 is not currently supported
#define USER_SETUP_ID 111
// See SetupX_Template.h for all options available
#define ILI9488_DRIVER
 #define USE_HSPI_PORT // HSPI pins 10-13
                    // Typical board default pins - change to match your board
#define TFT_CS   15 //     10 or 34 (FSPI CS0) 
#define TFT_MOSI 11 //     11 or 35 (FSPI D)
#define TFT_SCLK 12 //     12 or 36 (FSPI CLK)
#define TFT_MISO 13 //     13 or 37 (FSPI Q)

// Use pins in range 0-31
#define TFT_DC    14
#define TFT_RST   -1

Basitadam, 2024-10-16:
So is the problem been solved?

Just a tip: You can almost assign any pin to any function on S3. You don't have to stick to TFT_MOSI 11 for example.
Here are some setups I use for S3:

#define TFT_MOSI    35  // aka SDA
#define TFT_SCLK    36  // aka SCL
#define TFT_CS      26  // Chip select control pin
#define TFT_DC      47  // Data Command control pin
#define TFT_RST     4   // Reset pin (could connect to Arduino RESET pin)
#define TFT_MISO    19  

#define TFT_MISO 42
#define TFT_MOSI 2
#define TFT_SCLK 1
#define TFT_CS   39  // Chip select control pin
#define TFT_DC   41  // Data Command control pin
#define TFT_RST  40
#define TFT_BL	6
#define TFT_BACKLIGHT_ON HIGH

#define TFT_MISO  -1 //12
#define TFT_MOSI  11
#define TFT_SCLK  10
#define TFT_CS    9  // Chip select control pin
#define TFT_DC    8  // Data Command control pin
#define TFT_RST   14  // Reset pin (could connect to RST pin)
#define TFT_BL    2
#define TFT_BACKLIGHT_ON HIGH

Makes it easier to connect especially via dupont jumper cables

tonbor, 2024-10-17:
Now it is,  #define TFT_RST -1 was not working. Had to connect RST to get it working

> So is the problem been solved?
>
> Just a tip: You can almost assign any pin to any function on S3. You don't have to stick to TFT_MOSI 11 for example. Here are some setups I use for S3:
>
> #define [...]

Now it is, #define TFT_RST -1 was not working. Had to connect RST to get it working

piotrcurious, 2025-02-08:
bumped on this problem with esp32-c3 and st7735 160x80 display . 
the fix on espi define allows some examples to run , but anything using dma fails.

Basitadam, 2025-02-09:
> bumped on this problem with esp32-c3 and st7735 160x80 display . the fix on espi define allows some examples to run , but anything using dma fails.

DMA only works with S3 (with a small modification on the processor file)

piotrcurious, 2025-02-09:
Thanks for information. i tried to google reason why it doesn't work on c3 but I could not find much about it. It is some hardware bug/limitation or just implementation issue?
C3 is already quite chopped down so dma would be handy.

-------- Original Message --------

Basitadam, 2025-02-10:
just an implementation issue

blackCmd, 2025-02-25:
The Adafruit ILI9341 LCD output examples work on the EPS32-C3 Super Mini board, but all examples from TFT_eSPI result in infinite rebooting and display a white screen.

Basitadam, 2025-02-25:
> The Adafruit ILI9341 LCD output examples work on the EPS32-C3 Super Mini board, but all examples from TFT_eSPI result in infinite rebooting and display a white screen.

Can you use this file and try again:

[TFT_eSPI_ESP32_C3.zip](https://github.com/user-attachments/files/18958888/TFT_eSPI_ESP32_C3.zip)

Copy it to 
...\Arduino\libraries\TFT_eSPI\Processors
folder. Make sure you get a copy of  your file first

blackCmd, 2025-02-25:
> Can you use this file and try again:
>
> [TFT_eSPI_ESP32_C3.zip](https://github.com/user-attachments/files/18958888/TFT_eSPI_ESP32_C3.zip)
>
> Copy it to ...\Arduino\libraries\TFT_eSPI\Processors folder. [...]

Thank you for your quick response😊. I'm currently outside, but I'll test it as soon as I get home. Thank you. By the way, is this file one that you modified yourself?

Basitadam, 2025-02-25:
I don't remember who but a someone else suggested this mod, I just put it in the file

blackCmd, 2025-02-25:
> I don't remember who but a someone else suggested this mod, I just put it in the file

Oh, Is it works?

Basitadam, 2025-02-25:
Yes. Worked for me. I think @JiriBilek did it

RavWin, 2025-02-27:
After replacing TFT_eSPI_ESP32_C3.h with above archive or editing it manually rebooting stops, but PDQ3 or any other example shows nothing on supermini ESP32C3 with ST7735 1.8" 128x160 display and also no serial output in console when device starts.

RavWin, 2025-02-27:
> After replacing TFT_eSPI_ESP32_C3.h with above archive or editing it manually rebooting stops, but PDQ3 or any other example shows nothing on supermini ESP32C3 with ST7735 1.8" 128x160 display and also no serial output in console when device starts.

It helped to turn on
#define TFT_SDA_READ      // This option is for ESP32 ONLY, tested with ST7789 and GC9A01 display only
Works great

Edupotamo, 2025-03-24:
replacing TFT_eSPI_ESP32_C3.h works fine with round display GC9A0. Thank you very much. tested with arduinoIDE as well with platformIO+VSCode.
shall we espect a library update with the new TFT_eSPI_ESP32_C3.h file?

Neels-v-Wyk, 2025-06-15:
Currently also running into this with a [ESP32-C3 with a GC9D01 display](https://www.waveshare.com/wiki/ESP32-C3-LCD-0.71) (note that it's different from the GC9A01 display, which is mentioned in the above two comments)

The above changes didn't work for me unfortunately, although I'm facing the error described in https://github.com/Bodmer/TFT_eSPI/issues/3346 which lead me here.

Rolling back to version 2.0.14 seems to be the only fix for this for now :/

Neels-v-Wyk, 2025-06-15:
I guess even more interesting is that it only works when I roll back AND use the file from @Basitadam, might be useful for someone in the future. If I don't do that, I only get a black screen that doesn't update.

lucasadipietro, 2025-12-02:
The Problem: When compiling the demo code with the latest Arduino ESP32 Board Manager version (3.0.0 or higher), the code uploads but the screen remains completely white/blank. This happens because the libraries provided are not compatible with the new ESP32 Core 3.x architecture yet.

The Solution: The user MUST downgrade the ESP32 Board version in the Arduino IDE.

Go to Tools > Board > Boards Manager.

Search for "esp32".

Select version 2.0.14 (or any 2.0.x version) instead of the latest version.

Install and re-upload the code.

After doing this downgrade to v2.0.14, the screen works perfectly with your original factory code and pin configuration.