Bodmer/TFT_eSPI #3397: PowerPoint making an INO file not compling

flixmyswitch, 2024-07-08:
I have used the MACRO in the supplied powerpoint slides to make the INO file.

When i compile the sketch i get this error.  it does not like this on  line 31

TFT_eSPI_ext tft = TFT_eSPI_ext();

I have tested that my CYD display works by using the 3d cube exapmle.

C:\Users\andys\Downloads\PowerPoint_to_sketch-main\PowerPoint_Screens\PowerPoint_Screens.ino:31:33: error: no matching function for call to 'TFT_eSPI_ext::TFT_eSPI_ext()'
 TFT_eSPI_ext tft = TFT_eSPI_ext();
                                 ^
In file included from c:\Users\andys\Documents\Arduino\libraries\TFT_eSPI_ext/ILI9341_t3.h:2,
                 from c:\Users\andys\Documents\Arduino\libraries\TFT_eSPI_ext/font_Arial.h:4,
                 from C:\Users\andys\Downloads\PowerPoint_to_sketch-main\PowerPoint_Screens\PowerPoint_Screens.ino:24:
c:\Users\andys\Documents\Arduino\libraries\TFT_eSPI_ext/TFT_eSPI_ext.h:36:3: note: candidate: 'TFT_eSPI_ext::TFT_eSPI_ext(TFT_eSPI*)'
 

  TFT_eSPI_ext(TFT_eSPI *tft)
   ^~~~~~~~~~~~
c:\Users\andys\Documents\Arduino\libraries\TFT_eSPI_ext/TFT_eSPI_ext.h:36:3: note:   candidate expects 1 argument, 0 provided
c:\Users\andys\Documents\Arduino\libraries\TFT_eSPI_ext/TFT_eSPI_ext.h:29:7: note: candidate: 'TFT_eSPI_ext::TFT_eSPI_ext(const TFT_eSPI_ext&)'
 class TFT_eSPI_ext : public TFT_eSPI
       ^~~~~~~~~~~~
c:\Users\andys\Documents\Arduino\libraries\TFT_eSPI_ext/TFT_eSPI_ext.h:29:7: note:   candidate expects 1 argument, 0 provided

exit status 1

Compilation error: no matching function for call to 'TFT_eSPI_ext::TFT_eSPI_ext()'

I have looked at the other exapmle codes within the other required libraries, and been trying different thing but still get complie error

Bodmer, 2024-07-12:
Try changing the error line in PowerPoint_Screens.ino to:
TFT_eSPI_ext tft = TFT_eSPI_ext(&tft);

flixmyswitch, 2024-07-19:
Thank you that worked

closed 2024-07-19 by flixmyswitch