Bodmer/TFT_eSPI #3402: ST7789 blank display after update to 2.5.43

MasaoC, 2024-07-11:
I have an ST7789 module that works fine until version 2.5.34, but after 2.5.43, the display is blank.
Hardware configuration including pin configuration is the same, I modified User_Setup_select.h and Setup18_ST7789 after installing the library in either case.

I'm Using Raspberry Pi Pico with Arduino IDE.  Updated library on IDE.
I reverted to 2.5.34 and the display is now working.
Board package version 3.9.3 of Rasberry Pi Pico/RP2040.  
Board: Rasberry Pi Pico
SPI interface.

I modified the setup file as below on both versions.

I'm just going to use the old version, but I'm just informing this issue.

```
// See SetupX_Template.h for all options available
#define USER_SETUP_ID 18

#define ST7789_DRIVER     // Configure all registers

#define TFT_WIDTH  240
#define TFT_HEIGHT 320

#define CGRAM_OFFSET      // Library will add offsets required

// For Pico (PR2040)
#define TFT_CS 17   // Chip Select pin
#define TFT_DC 16   // Data Command control pin 
#define TFT_RST -1  // No Reset pin
#define TFT_MOSI 19
#define TFT_SCLK 18

#define LOAD_GLCD   // Font 1. Original Adafruit 8 pixel font needs ~1820 bytes in FLASH
#define LOAD_FONT2  // Font 2. Small 16 pixel high font, needs ~3534 bytes in FLASH, 96 characters
#define LOAD_FONT4  // Font 4. Medium 26 pixel high font, needs ~5848 bytes in FLASH, 96 characters
#define LOAD_FONT6  // Font 6. Large 48 pixel font, needs ~2666 bytes in FLASH, only characters 1234567890:-.apm
#define LOAD_FONT7  // Font 7. 7 segment 48 pixel font, needs ~2438 bytes in FLASH, only characters 1234567890:.
#define LOAD_FONT8  // Font 8. Large 75 pixel font needs ~3256 bytes in FLASH, only characters 1234567890:-.
// #define LOAD_FONT8N // Font 8. Alternative to Font 8 above, slightly narrower, so 3 digits fit a 160 pixel TFT
#define LOAD_GFXFF  // FreeFonts. Include access to the 48 Adafruit_GFX free fonts FF1 to FF48 and custom fonts

#define SMOOTH_FONT

#define SPI_FREQUENCY       20000000
#define SPI_READ_FREQUENCY  20000000
#define SPI_TOUCH_FREQUENCY  2500000

```

Bodmer, 2024-07-12:
Thanks for reporting his, I have not used the library for some time due to other time pressures but will investigate as the RP2040 is my favorite processor!

orhanyor, 2024-07-27:
if you add this line to your user setup it should work. #define RP2040_PIO_SPI
I had the same problem here 
https://github.com/Bodmer/TFT_eSPI/issues/3252

TobiasVanDyk, 2024-07-29:
When I add #define RP2040_PIO_SPI it does not play nicely with the TouchScreen:

`In file included from C:\Users\Tobias\Documents\Arduino\libraries\TFT_eSPI/TFT_eSPI.h:107,
                 from C:\Users\Tobias\Documents\Arduino\EarleArduino\TouchLCDst7789-Pico\VolumeMacroPad340\VolumeMacroPad340.ino:24:
C:\Users\Tobias\Documents\Arduino\libraries\TFT_eSPI/Processors/TFT_eSPI_RP2040.h:244:6: error: #error Touch screen not supported in parallel or SPI PIO mode, use a separate library.
  244 |     #error Touch screen not supported in parallel or SPI PIO mode, use a separate library.
      |      ^~~~~
C:\Users\Tobias\Documents\Arduino\libraries\TFT_eSPI/TFT_eSPI.h:966:8: error: #error >>>>------>> Touch functions not supported in 8/16-bit parallel mode or with RP2040 PIO.
  966 |       #error >>>>------>> Touch functions not supported in 8/16-bit parallel mode or with RP2040 PIO.
      |        ^~~~~`

umeiko, 2024-07-31:
mine works
```
// ST7789 240 x 280 display with no chip select line
#define USER_SETUP_ID 18

#define ST7789_DRIVER     // Configure all registers

#define TFT_WIDTH  240
#define TFT_HEIGHT 280

#define CGRAM_OFFSET      // Library will add offsets required
#define TFT_RGB_ORDER TFT_RGB  // Colour order Red-Green-Blue

#define TFT_CS 9   // CS
#define TFT_DC 8   // RX
#define TFT_RST -1  // No Reset pin
#define TFT_MOSI 7  // TX
#define TFT_SCLK 6  // SCK

#define LOAD_GLCD   // Font 1. Original Adafruit 8 pixel font needs ~1820 bytes in FLASH
#define LOAD_FONT2  // Font 2. Small 16 pixel high font, needs ~3534 bytes in FLASH, 96 characters
#define LOAD_FONT4  // Font 4. Medium 26 pixel high font, needs ~5848 bytes in FLASH, 96 characters
#define LOAD_FONT6  // Font 6. Large 48 pixel font, needs ~2666 bytes in FLASH, only characters 1234567890:-.apm
#define LOAD_FONT7  // Font 7. 7 segment 48 pixel font, needs ~2438 bytes in FLASH, only characters 1234567890:.
#define LOAD_FONT8  // Font 8. Large 75 pixel font needs ~3256 bytes in FLASH, only characters 1234567890:-.
//#define LOAD_FONT8N // Font 8. Alternative to Font 8 above, slightly narrower, so 3 digits fit a 160 pixel TFT
#define LOAD_GFXFF  // FreeFonts. Include access to the 48 Adafruit_GFX free fonts FF1 to FF48 and custom fonts

#define SMOOTH_FONT

#define RP2040_PIO_SPI
#define RP2040_DMA
// #define SPI_FREQUENCY  27000000
// #define SPI_FREQUENCY  80000000
#define SPI_FREQUENCY  40000000
// #define SPI_FREQUENCY  20000000

#define SUPPORT_TRANSACTIONS
``` 
But my issue is the color is wrong,  seems `tft.pushImage()` draw the (blue, red, green) But not (red, green, blue) as expect.

MasaoC, 2024-08-15:
Thanks for everyone's replay,
unfortunately, I'm using a 16bit-Parallel connection with TFT, so #define RP2040_PIO_SPI will not be the solution.

jdrowell, 2024-08-15:
I wish I'd found this issue before going through all the trouble of git bisecting the pico-sdk. Simply adding

#define RP2040_PIO_SPI
#define RP2040_DMA

like described above fixed the issues with my LilyGo T-Display 2040 on arduino-pico later than 3.2.2. I don't know enough about SPI to tell what the best fix is, but the breaking change in pico-sdk was here:

    0121007c853a36cebcf78e591437d6c79c964d4b Improve SPI set-up: Don't change the config whilst it is enabled (#1227)

It seems that disabling and re-enabling SPI does not work well with the TFT_eSPI driver, at least on this board. The only instance that really affects the sketches I tested was in src/rp2_common/hardware_spi/include/hardware/spi.h static inline void spi_set_format(). Just commenting lines 185, 186 and 197 got the TFT working again. And because pico-sdk is still at 1.5.1 on the latest tag of arduino-pico, this fix works for all versions of arduino-pico after 3.2.2.

Splarkszter, 2024-08-21:
I can't make mine work but I'm using an ESP32.

It's the same ST7789 240x320 but without touchscreen.

I have a pico but i need to solder the pins so I haven't been able to try it.