Bodmer/TFT_eSPI #3414: Unable to use DMA with ST7789 + ESP32-S3

TheWolf95, 2024-07-23:
Example output:

```
  Serial.begin(115200);
  tft.init();
```
IDE: PlatformIO
Library: bodmer/TFT_eSPI@^2.5.43
MCU: ESP32-S3-WROOM-1 N8R8
TFT Driver: ST7789 - SPI (SCK, SDA, RES, DC)

I am trying to use the `pushImageDMA()` method to achieve faster refreshrate on a ST7789 display, without MISO and CS pin ([link to aliexpress](https://it.aliexpress.com/item/1005005621788608.html)). DMA initialization is ok, but any update to the screen is not sent (no fillScreen, no pushImageDMA, not print, ecc), just the last frame is showed.

Wiring: (it is a custom board from another project, I connected spare pins exposed)

TFT_MOSI   18
TFT_SCLK    17
TFT_DC         2
TFT_RST        1

TFT config:
```
#define ST7789_2_DRIVER

#define TFT_RGB_ORDER TFT_RGB

 #define TFT_WIDTH  240
 #define TFT_HEIGHT  240

#define SPI_FREQUENCY 80*1000000
#define SPI_READ_FREQUENCY  20000000
#define SPI_TOUCH_FREQUENCY  2500000

#define USE_HSPI_PORT
```

Setup before DMA push:
```
if(tft.initDMA()) Serial.println("[DMA] Init ok.");
DMA_BUFFER = (uint16_t *) heap_caps_malloc(256 * sizeof(uint16_t), MALLOC_CAP_DMA);
SCREENBUFFER = (uint16_t *) heap_caps_malloc(256 * sizeof(uint16_t), MALLOC_CAP_DMA);
```

Push in loop function:
```
extern "C" int lcd_write_frame(const uint16_t x, const uint16_t y, const uint16_t width, const uint16_t height) {
...
  for (i = 0; i < height; i++) {
      for (X_ = 0; X_ < width; X_++) { 
          
          SCREENBUFFER[X_] = gbc_16bit[(0x3F & ((uint8_t *)SCREENMEMORY[i])[X_])];

      }

     tft.startWrite();  // I know this should be called to keep the SPI interface reserved through CS pin (not my case), but who knows
     tft.pushImageDMA(0, i, 240, 1, (uint16_t *) SCREENBUFFER, (uint16_t *) DMA_BUFFER);
     tft.endWrite();

  }
...
}
```

I get a refresh times of around 80ms (12 FPS) for 240x240 16bit images, I was aiming on improving them (the code is a fork of GBCCAT, ported to PlatformIO + TFT_eSPI library)

umeiko, 2024-08-12:
Mine DMA works on esp32s3 with `TFT_ESPI_VERSION 2.5.43`. 
```C
#define ST7789_DRIVER     // Configure all registers

#define TFT_WIDTH  240
#define TFT_HEIGHT 280

#define CGRAM_OFFSET      // Library will add offsets required

#define TFT_RGB_ORDER TFT_RGB  // Colour order Red-Green-Blue
#define ESP32_DMA
#define USE_HSPI_PORT

#define TFT_MOSI 4
#define TFT_SCLK 3
#define TFT_CS 2
#define TFT_DC 1
#define TFT_RST 5

#define TFT_BL 9  // LCD背光控制引脚

#define TFT_BACKLIGHT_ON HIGH  // HIGH or LOW are options

#define LOAD_GLCD   // Font 1. Original Adafruit 8 pixel font needs ~1820 bytes in FLASH
#define LOAD_FONT2  // Font 2. Small 16 pixel high font, needs ~3534 bytes in FLASH, 96 characters
#define LOAD_FONT4  // Font 4. Medium 26 pixel high font, needs ~5848 bytes in FLASH, 96 characters
#define LOAD_FONT6  // Font 6. Large 48 pixel font, needs ~2666 bytes in FLASH, only characters 1234567890:-.apm
#define LOAD_FONT7  // Font 7. 7 segment 48 pixel font, needs ~2438 bytes in FLASH, only characters 1234567890:.
#define LOAD_FONT8  // Font 8. Large 75 pixel font needs ~3256 bytes in FLASH, only characters 1234567890:-.
//#define LOAD_FONT8N // Font 8. Alternative to Font 8 above, slightly narrower, so 3 digits fit a 160 pixel TFT
#define LOAD_GFXFF  // FreeFonts. Include access to the 48 Adafruit_GFX free fonts FF1 to FF48 and custom fonts

#define SMOOTH_FONT

// #define SPI_FREQUENCY  27000000
// #define SPI_FREQUENCY  80000000
#define SPI_FREQUENCY  40000000

#define SPI_READ_FREQUENCY  20000000
#define SPI_TOUCH_FREQUENCY  2500000
```
You can test DMA working by running builtin example `Flash_Jpg_DMA.ino`.

TheWolf95, 2024-08-12:
@umeiko thank you for your help, although I already got it running I've faced a new issue. I've replied to my discussion [here](https://github.com/Bodmer/TFT_eSPI/discussions/3432#discussioncomment-10316770), do you have any idea why would this happen?

Thank you for sharing you settings again!

rsiemens77, 2024-08-22:
> DMA initialization is ok, but any update to the screen is not sent (no fillScreen, no pushImageDMA, not print, ecc), just the last frame is showed.

I seem to have the same issue you were having, only the first frame displays, and then the display is frozen.  I'm using 2.5.43.  If I switch to the non-DMA update, everything works fine.  I followed your discussion, but it's unclear what exactly fixed your issue?  I've got HSPI and MISO defined as others suggested.

closed 2024-08-22 by TheWolf95