Bodmer/TFT_eSPI #3418: Crash with Lolin s3 + ili9341 + platformio + espressif32@6.7.0

max22-, 2024-07-26:
I have tried a minimal program with the arduino framework and platformio. 
The board used is a [lolin s3](https://www.wemos.cc/en/latest/s3/s3.html).
With espressif32@6.6.0 (and previous versions), it works, but since the 6.7.0 version the board crashes during TFT_eSPI.begin().
The display used is a [ILI9341](https://www.pjrc.com/store/display_ili9341_touch.jpg), connected via the SPI bus.

The code is really short so i paste it here:
```
#include <Arduino.h>
#include <TFT_eSPI.h>

TFT_eSPI tft;

void setup() {
  Serial.begin(115200);
  tft.begin();
  tft.setRotation(3);
  tft.fillScreen(TFT_BLACK);
  tft.println("Ready.");
}

void loop() {
  Serial.println("Hello, World!");
  delay(1000);
}
```

And the platformio.ini file :

```
[env:lolin_s3]
platform = espressif32@6.7.0
board = lolin_s3
framework = arduino
monitor_speed = 115200
monitor_filters = esp32_exception_decoder
lib_deps = 
	bodmer/TFT_eSPI@^2.5.43
build_flags = 
	-DUSER_SETUP_LOADED
	-DILI9341_DRIVER
	-DTFT_MISO=13
	-DTFT_MOSI=11
	-DTFT_SCLK=12
	-DTFT_CS=10
	-DTFT_DC=14
	-DTFT_RST=21
	-DLOAD_GLCD
	-DLOAD_FONT2
	-DLOAD_FONT4
	-DLOAD_FONT6
	-DLOAD_FONT7
	-DLOAD_FONT8
	-DLOAD_GFXFF
	-DSPI_FREQUENCY=70000000
	-DSPI_READ_FREQUENCY=20000000
	-DSPI_TOUCH_FREQUENCY=2500000
```

Here is the output of esp32_exception_decoder :

```
Guru Meditation Error: Core  1 panic'ed (StoreProhibited). Exception was unhandled.

Core  1 register dump:
PC      : 0x420026bf  PS      : 0x00060a30  A0      : 0x82002794  A1      : 0x3fcebbc0  
A2      : 0x00000010  A3      : 0x00000000  A4      : 0x60004000  A5      : 0x0000000b  
A6      : 0x000000ff  A7      : 0x00000008  A8      : 0x08000000  A9      : 0x3fcebb90  
A10     : 0x3fc97454  A11     : 0x00000001  A12     : 0xffffffff  A13     : 0x00200000  
A14     : 0x00000000  A15     : 0x3fc94498  SAR     : 0x00000010  EXCCAUSE: 0x0000001d  
EXCVADDR: 0x00000010  LBEG    : 0x420145c0  LEND    : 0x42014624  LCOUNT  : 0x00000000  

Backtrace: 0x420026bc:0x3fcebbc0 0x42002791:0x3fcebbf0 0x4200279c:0x3fcebc10 0x420017e3:0x3fcebc30 0x42003f4a:0x3fcebc50
  #0  0x420026bc in TFT_eSPI::begin_tft_write() at .pio/libdeps/lolin_s3/TFT_eSPI/TFT_eSPI.cpp:81
      (inlined by) TFT_eSPI::writecommand(unsigned char) at .pio/libdeps/lolin_s3/TFT_eSPI/TFT_eSPI.cpp:982
  #1  0x42002791 in TFT_eSPI::init(unsigned char) at .pio/libdeps/lolin_s3/TFT_eSPI/TFT_eSPI.cpp:692
  #2  0x4200279c in TFT_eSPI::begin(unsigned char) at .pio/libdeps/lolin_s3/TFT_eSPI/TFT_eSPI.cpp:603
  #3  0x420017e3 in setup() at src/main.cpp:8
  #4  0x42003f4a in loopTask(void*) at /home/maxime/.platformio/packages/framework-arduinoespressif32/cores/esp32/main.cpp:42
```

I don't know if it is a bug in your library or in the new version of the esp32 platform (or something i did wrong). By the way, thank you very much for your awesome work.

PhilG1300, 2024-07-27:
Hi,
For my ESP32S3 dev board using SPI to a GC9A01 display.
I put the following in platform.ini

build_flags= -DUSE_HSPI_PORT=1

Add it to the build_flags list you have
-DUSE_HSPI_PORT=1

This stops the crashing when calling the initialisation.

max22-, 2024-07-27:
Indeed, it works !!
thank you very much !

closed 2024-07-27 by max22-