Bodmer/TFT_eSPI #3426: Esp32S3 dev Board + St7789 display (320x240) - not compatible ???

cosminiq, 2024-07-31:
Hello everyone!

I have an ST7789 display with the following pins: GND, VCC, SCL, SDA, RST, DC, CS, and BL. I am trying to use it with an ESP32-S3 development board. While the display works great with a Raspberry Pi Pico, I can't get an image when using it with the ESP32-S3. Additionally, after uploading the example code, my ESP32-S3 goes into panic mode. I have attached a picture and the setup code. If you can help me, I would greatly appreciate it.

[image]

// Define the display driver and type
#define ST7789_DRIVER

// Define the display resolution
//#define TFT_WIDTH  240
//#define TFT_HEIGHT 240

/* Define the pins for **ESP32S3  + St7789** used for the display
#define TFT_MISO -1       // Not connected (MISO is not used in this setup)
#define TFT_MOSI 11       // (SDA) Data out to display
#define TFT_SCLK 12       // (SCL) Clock out to display
#define TFT_CS   10       // Chip select control pin
#define TFT_DC   9        // Data Command control pin
#define TFT_RST  8        // Reset pin
#define TFT_BL   7        // Backlight pin (optional, can be controlled via software)

// Power control settings (optional)
#define TFT_BACKLIGHT_ON HIGH

*/

//Definition of pins------Rasperry pi pico display ---------------------------------------
#define RPI_DISPLAY_TYPE
#define RP2040_PIO_SPI
#define TFT_SPI_PORT 0

#define TFT_INVERSION_ON 
#define TFT_BL 10
#define TFT_MISO -1
#define TFT_MOSI 19 
#define TFT_SCLK 18 
#define TFT_CS 17 
#define TFT_DC 21 
#define TFT_RST -1 
#define TFT_BACKLIGHT_ON HIGH
//

// #define SPI_FREQUENCY 1000000
// #define SPI_FREQUENCY 5000000
// #define SPI_FREQUENCY 10000000
// #define SPI_FREQUENCY 20000000
#define  SPI_FREQUENCY   27000000  // In realtà lo imposta a 26,67 MHz = 80/3
//#define SPI_FREQUENCY 40000000
//#define SPI_FREQUENCY 80000000

// Frequenza SPI ridotta opzionale per la lettura del TFT
//# define  SPI_READ_FREQUENCY   80000000

////////////////////////////////////////////////////////////////////////////////////////////
// Fonts to be available
////////////////////////////////////////////////////////////////////////////////////////////
#define LOAD_GLCD   // Font 1. Original Adafruit 8 pixel font needs ~1820 bytes in FLASH
#define LOAD_FONT2  // Font 2. Small 16 pixel high font, needs ~3534 bytes in FLASH, 96 characters
#define LOAD_FONT4  // Font 4. Medium 26 pixel high font, needs ~5848 bytes in FLASH, 96 characters
#define LOAD_FONT6  // Font 6. Large 48 pixel font, needs ~2666 bytes in FLASH, only characters 1234567890:-.apm
#define LOAD_FONT7  // Font 7. 7 segment 48 pixel font, needs ~2438 bytes in FLASH, only characters 1234567890:.
#define LOAD_FONT8  // Font 8. Large 75 pixel font needs ~3256 bytes in FLASH, only characters 1234567890:-.
#define LOAD_GFXFF  // FreeFonts. Include access to the 48 Adafruit_GFX free fonts FF1 to FF48 and custom fonts
#define SMOOTH_FONT

////////////////////////////////////////////////////////////////////////////////////////////

cosminiq, 2024-08-01:
After some research through closed topics, I identified the problem as this line: #define USE_HSPI_PORT. Now everything works correctly. I am attaching the code for User_Setup.h.
[image]
`// Setup file for ESP32S3 with ST7789 display

#define ST7789_DRIVER // Define the display driver for ST7789

#define TFT_WIDTH  240 // Define the width of the display in pixels
#define TFT_HEIGHT 320 // Define the height of the display in pixels
#define TFT_INVERSION_ON // Define display inversion setting (ST7789 often requires this)

#define TFT_CS    10   // Define the chip select pin for the display
#define TFT_DC    9    // Define the data/command pin for the display
#define TFT_SCLK  13   // Define the clock pin for the display
#define TFT_MOSI  11   // Define the MOSI (Master Out Slave In) pin for the display
#define TFT_RST   8    // Define the reset pin for the display
#define TFT_BL    3    // Define the backlight pin for the display

#define USE_HSPI_PORT   // Use HSPI port (HSPI is one of the SPI ports available on ESP32)
#define TFT_BACKLIGHT_ON HIGH  // Define the state to turn the backlight on (HIGH or LOW)

// #define SPI_FREQUENCY 1000000
// #define SPI_FREQUENCY 5000000
// #define SPI_FREQUENCY 10000000
// #define SPI_FREQUENCY 20000000
// #define SPI_FREQUENCY 27000000
// #define SPI_FREQUENCY 40000000
#define SPI_FREQUENCY  80000000 // Define the SPI communication frequency in Hertz

#define LOAD_GLCD   // Load GLCD font (Graphics LCD Font)
#define LOAD_FONT2  // Load Font 2 (Small 16 pixel high font)
#define LOAD_FONT4  // Load Font 4 (Medium 26 pixel high font)
#define LOAD_FONT6  // Load Font 6 (Large 48 pixel high font)
#define LOAD_FONT7  // Load Font 7 (7-segment 48 pixel high font)
#define LOAD_FONT8  // Load Font 8 (Large 75 pixel font)
#define LOAD_GFXFF  // Load GFXFF (Adafruit_GFX Free Fonts)
#define SMOOTH_FONT // Enable Smooth Fonts (anti-aliased fonts)`

closed 2024-08-01 by cosminiq

bdezso74, 2025-02-04:
Which ESP32 board package version did you use?