Bodmer/TFT_eSPI #3427: Unexpected Color Output with `tft.pushImage()`  On rp2040 with st7789 driver.

umeiko, 2024-08-02:
Hello,

I've encountered an issue with the` tft.pushImage()` function where the color output is not as expected. Despite setting the color to red using tft.color565(255, 0, 0), the resulting display is blue. Similarly, the color code (0, 255, 0) is displayed as green, while (0, 0, 255) shows up as red.

- Hardware: `RP2040` Board.
- `TFT_eSPI` Version: 2.5.43.
- `arduino-Pico` Version: 3.9.3 from [earlephilhower](https://github.com/earlephilhower/arduino-pico).

Example:

```C++
#include <Arduino.h>
#include <TFT_eSPI.h> 
#include <SPI.h>

#define _SCALE 9
static uint16_t buffer[32*_SCALE * 24*_SCALE] = {};
static int R_colour = 255;
static int G_colour = 0;
static int B_colour = 0;
TFT_eSPI tft = TFT_eSPI();  

void draw_block_on_buffer(int x, int y, int w, int max_x, int max_y, uint16_t color, uint16_t* data){
   for(int i =y; i < y + w; i++){
        for(int j = x; j < x + w; j++){
         data[i * max_x + j] = color;
      }
   }
}

void update_buffer(){
   for(int y=0; y<24; y++){ 
      for(int x=0; x<32; x++){
         int id = (23-y) * 32 + x;
         draw_block_on_buffer(x*_SCALE, y*_SCALE, _SCALE, 32*_SCALE, 24*_SCALE, tft.color565(R_colour, G_colour, B_colour), buffer);
      } 
   }
}

// works unexpected 
void draw_pushImage(){
   update_buffer(re_mapcolor);
   tft.pushImage(0, 0, 32*_SCALE, 24*_SCALE, buffer);
}

// works expected 
void draw_fillRect(){
   for (int i = 0 ; i < 24 ; i++){
     for (int j = 0; j < 32; j++){
       tft.fillRect(x*_SCALE,  y*_SCALE, _SCALE, _SCALE, tft.color565(R_colour, G_colour, B_colour));  
      }
   }
}

void setup(void){
   tft.init();
   tft.setRotation(0);
   draw_pushImage();
   // draw_fillRect();
}

void loop() 
{
   delay(3000);
}
```
My `User_Setup.h`:

```C++
// ST7789 240 x 280 display with no chip select line
#define USER_SETUP_ID 18

#define ST7789_DRIVER     // Configure all registers

#define TFT_WIDTH  240
#define TFT_HEIGHT 280

#define CGRAM_OFFSET      // Library will add offsets required
#define TFT_RGB_ORDER TFT_RGB  // Colour order Red-Green-Blue

#define TFT_CS 9   // CS
#define TFT_DC 8   // RX
#define TFT_RST -1  // No Reset pin
#define TFT_MOSI 7  // TX
#define TFT_SCLK 6  // SCK

#define LOAD_GLCD   // Font 1. Original Adafruit 8 pixel font needs ~1820 bytes in FLASH
#define LOAD_FONT2  // Font 2. Small 16 pixel high font, needs ~3534 bytes in FLASH, 96 characters
#define LOAD_FONT4  // Font 4. Medium 26 pixel high font, needs ~5848 bytes in FLASH, 96 characters
#define LOAD_FONT6  // Font 6. Large 48 pixel font, needs ~2666 bytes in FLASH, only characters 1234567890:-.apm
#define LOAD_FONT7  // Font 7. 7 segment 48 pixel font, needs ~2438 bytes in FLASH, only characters 1234567890:.
#define LOAD_FONT8  // Font 8. Large 75 pixel font needs ~3256 bytes in FLASH, only characters 1234567890:-.
//#define LOAD_FONT8N // Font 8. Alternative to Font 8 above, slightly narrower, so 3 digits fit a 160 pixel TFT
#define LOAD_GFXFF  // FreeFonts. Include access to the 48 Adafruit_GFX free fonts FF1 to FF48 and custom fonts

#define SMOOTH_FONT

#define RP2040_PIO_SPI
#define RP2040_DMA
// #define SPI_FREQUENCY  27000000
// #define SPI_FREQUENCY  80000000
#define SPI_FREQUENCY  40000000
// #define SPI_FREQUENCY  20000000

#define SUPPORT_TRANSACTIONS
```

umeiko, 2024-08-12:
solved by `tft.setSwapBytes(true);`

bi3qwq, 2024-09-04:
https://github.com/bi3qwq/ConvertImageToColor565

you can try it.