Bodmer/TFT_eSPI #3429: Guru Meditation Error: Core 1 panic'ed

Patsjemoe, 2024-08-04:
1. I get a Guru Meditation Error: Core 1 panic'ed,  after loading an LVGL file that is running well on ESP32-3248S035, using ESP32wroom , but due to insufficient memory to include BLE, I needed to go to ESP32-S3, with a separate TFT display , ILI9486. For this, I wrote a user_setup file, as I could not find any existing one to match this combination. I attached the user set up file.
[ILI9486.txt](https://github.com/user-attachments/files/16486968/ILI9486.txt)

2.Arduino IDE 2.3.2
3. TFT_eSPI library version  2.5.43
4. Board package version 3.0.4 esp32 by espressif systems
5. ESP32 S3 DevkitC-1
6. TFT driver ILI9486
7. Interface type SPI

Plus further information as appropriate to the problem:
1. TFT connections : ILI9486.txt , attached file, in fact user_setup.h file.
2. A zip file containing your setup file (just drag and drop in message window - do not paste in long files!)
3. ESP Exception Decoder
Sketch: ui FQBN: esp32:esp32:esp32s3

PC      : 0x420267fa  PS      : 0x00060430  A0      : 0x820269d8  A1      : 0x3fcebd20  
A2      : 0x3fc969a8  A3      : 0x00000000  A4      : 0x60004000  A5      : 0x00000010  
A6      : 0x000000ff  A7      : 0xff000000  A8      : 0x08000000  A9      : 0x3fcebcf0  
A10     : 0x3fcaa72c  A11     : 0x00000001  A12     : 0x02625a00  A13     : 0xffffffff  
A14     : 0x00000031  A15     : 0x00000000  SAR     : 0x00000006  EXCCAUSE: 0x0000001d  
EXCVADDR: 0x00000010  LBEG    : 0x4202ae24  LEND    : 0x4202ae6b  LCOUNT  : 0x00000003  

Backtrace: 0x420267f7:0x3fcebd20 0x420269d5:0x3fce

PC: 0x420267fa:  is in TFT_eSPI::writecommand(unsigned char) (i:\arduino\hardware\espressif\esp32\sketches\S035\SjotterS035\libraries\TFT_eSPI\TFT_eSPI.cpp:81).
EXCVADDR: 0x00000010

Decoding stack results
0x420267f7:  is in TFT_eSPI::writecommand(unsigned char) (i:\arduino\hardware\espressif\esp32\sketches\S035\SjotterS035\libraries\TFT_eSPI\TFT_eSPI.cpp:81).
0x420269d5:  is in TFT_eSPI::init(unsigned char) (i:\arduino\hardware\espressif\esp32\sketches\S035\SjotterS035\libraries\TFT_eSPI\TFT_eSPI.cpp:692).

Any suggestions  are welcome.
thanks in advance
best regards
Ludo

Patsjemoe, 2024-08-04:
When checking again, I had a few lines more in the exception decoder, here the complete : 
ESP Exception Decoder
Sketch: ui FQBN: esp32:esp32:esp32s3

PC      : 0x420267fa  PS      : 0x00060430  A0      : 0x820269d8  A1      : 0x3fcebd20  
A2      : 0x3fc969a8  A3      : 0x00000000  A4      : 0x60004000  A5      : 0x00000010  
A6      : 0x000000ff  A7      : 0xff000000  A8      : 0x08000000  A9      : 0x3fcebcf0  
A10     : 0x3fcaa72c  A11     : 0x00000001  A12     : 0x02625a00  A13     : 0xffffffff  
A14     : 0x00000031  A15     : 0x00000000  SAR     : 0x00000006  EXCCAUSE: 0x0000001d  
EXCVADDR: 0x00000010  LBEG    : 0x4202ae24  LEND    : 0x4202ae6b  LCOUNT  : 0x00000003  

Backtrace: 0x420267f7:0x3fcebd20 0x420269d5:0x3fcebd50 0x42026bf4:0x3fcebd80 0x420020bb:0x3fcebda0 0x4202be06:0x3fcebe10 0x4037ed22:0x3fcebe30

PC: 0x420267fa:  is in TFT_eSPI::writecommand(unsigned char) (i:\arduino\hardware\espressif\esp32\sketches\S035\SjotterS035\libraries\TFT_eSPI\TFT_eSPI.cpp:81).
EXCVADDR: 0x00000010

Decoding stack results
0x420267f7:  is in TFT_eSPI::writecommand(unsigned char) (i:\arduino\hardware\espressif\esp32\sketches\S035\SjotterS035\libraries\TFT_eSPI\TFT_eSPI.cpp:81).
0x420269d5:  is in TFT_eSPI::init(unsigned char) (i:\arduino\hardware\espressif\esp32\sketches\S035\SjotterS035\libraries\TFT_eSPI\TFT_eSPI.cpp:692).
0x42026bf4:  is in TFT_eSPI::begin(unsigned char) (i:\arduino\hardware\espressif\esp32\sketches\S035\SjotterS035\libraries\TFT_eSPI\TFT_eSPI.cpp:603).
0x420020bb: setup() at I:\arduino\hardware\espressif\esp32\sketches\S035\SjotterS035\ui\ui.ino:82
0x4202be06: loopTask(void*) at C:\Users\ludok\AppData\Local\Arduino15\packages\esp32\hardware\esp32\3.0.3\cores\esp32\main.cpp:59
0x4037ed22: vPortTaskWrapper at /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/freertos/FreeRTOS-Kernel/portable/xtensa\port.c:162

Patsjemoe, 2024-08-05:
Hi Bodmer,
I found a suggestion to change espressif board package to 2.0.14, and this works, luckaly confirming correct user setup.h.
But how can this be reported to Espressif, so they can correct the ESP32S3 board set up > 2.0.14 ?
regards
Ludo

Patsjemoe, 2024-08-05:
I have put the question to github espressif-esp32

Patsjemoe, 2024-08-05:
It would be nice to have a solution, as when I now compile other programs with 2.0.14, I get errors : 
compilation error: invalid fqbn: getting build properties for board esp32:esp32:esp32: invalid option 'zigbeemode', going back to 3.0.4 solves that problem....Would like to stay with 1 and the latest of course....

elgerg, 2024-08-05:
I think this issue is a duplicate of:
https://github.com/Bodmer/TFT_eSPI/issues/3384

Did you try the workaround?

Patsjemoe, 2024-08-05:
Hi Bodmer,
1)I tried some of llink #3304, no success. Also, rather confusing, as for some it took away the panic, but did not show display anymore. Switching to espressif 2.0.14 board package worked.
2) #3384 seems to be focussed on ESP32-C3, not esp32-S3, and I am not sure how to translate these. They talk about #define SPI_PORT SPI2_HOST, but this is not in ESP32-s3.

Can you confirm which workaround to test for the ESP32-S3 ?

Note : As I mentioned before, I also have put this issue in github espressif-esp32, and they state that the issue is that TFT_eSPI library is not compatible with Arduino v3. Can you talk to them to understand how to make it compatible ? 
best regards
Ludo

Patsjemoe, 2024-08-05:
Hi Bodmer,
In the datasheet I found that SPI_MOSI_DLEN_REG is replaced by SPI_MS_DLEN_REG... and changed this in my  esp32-s3.h  and esp32-S3.cpp, but do you know if this should be changed in more files ?

Patsjemoe, 2024-08-05:
oops, just found that you changed this already : 
  // Fix ESP32S3 IDF bug for name change
  #ifndef SPI_MOSI_DLEN_REG
    #define SPI_MOSI_DLEN_REG(x) SPI_MS_DLEN_REG(x)
  #endif

elgerg, 2024-08-05:
Hi,
Sorry I'm not bodmer but just another user. Hope you get on well.

Patsjemoe, 2024-08-05:
In your TFT_eSPI.cpp, at line 81, you have SET_BUS_WRITE_MODE;
Where/in which files can I find this ? I cannot find this in the ESP32-s3 datasheet, but the error is clearly pointing to this.

Patsjemoe, 2024-08-05:
OK, sorry I did not notice...

Patsjemoe, 2024-08-07:
Hi,
I believe to have found the issue with the SPI TFT_eSPI driver for ESP32-S3. I found it 
strange that this problem is going on since Espressif esp32 board package 2.0.15, where I am now
at 3.0.4  , and the suggestion to go back to 2.014 worked but I could not maintain this because then 
I had silly errors with other  programs like zigbee error in BLE client...
So going back was not a solution, and since there was none, I had to search myself.

I came to the solution by reading soo many "solutions" that did not solve anything, 
but 1 was solving the problem for  some people, including myself : #define USE_HSPI_PORT in user_setup.h
From here, I dived  into the ESP32 and ESP32_S3 technical reference manual to find their references to the SPI ports.

In ESP32 technical reference manual, you can find  FSPI, VSPI, HSPI, but these terms do not show up in the ESP32-S3
version, there it is SPI1, SPI2, SPI3, so this was the clue to go further....

What has changed in ESP32 board package 2.015....I do not know, I have no access to it, and cannot get 
an answer to my question what was changed.

So had to look into the library part, and came to TFT_eSPI_ESP32_S3.h, where my solution with #define USE_HSPI_PORT
was found...in #ifdef USE_HSPI_PORT : 

#ifdef USE_HSPI_PORT
  #ifdef CONFIG_IDF_TARGET_ESP32
    #define SPI_PORT HSPI  //HSPI is port 2 on ESP32
  #else
    #define SPI_PORT 3     //HSPI is port 3 on ESP32 S2
	//#define SPI_PORT FSPI     //(1*) changes as test-->>vspi?
  #endif
#elif defined(USE_FSPI_PORT)
    #define SPI_PORT 2 //FSPI(ESP32 S2)
#else
  #ifdef CONFIG_IDF_TARGET_ESP32
    #define SPI_PORT VSPI
  #elif CONFIG_IDF_TARGET_ESP32S2
    #define SPI_PORT 2 //FSPI(ESP32 S2)
  #elif CONFIG_IDF_TARGET_ESP32S3
    //#define SPI_PORT FSPI		//(2*) changed
	#define SPI_PORT 3		//(2*) changed
  #endif
#endif

Looking into this : I had selected ESP32-S3, so the first "if" is for ESP32 that it omits, then it defined
SPI_PORT 3 ...this worked.(but this is only saying...it is not esp32, but does not specify which processor 
with the else....so I think this needs to be reworked for the correct processor).

I then added some print instructions to see which SPI_PORT selection from this snippet was made and had 
following : 

With //#define USE_HSPI_PORT (taking out), with ESP32-S3, it selects VSPI, and gives the GURU panic error. 
(I was surprised it did not select FSPI  , as it should have gone to config_target_ESP32s3....)
So using #define USE_HSPI_PORT with esp32-s3 selects SPI_PORT 3....

With //#define USE_HSPI_PORT , and Changing #define SPI_PORT FSPI to #define SPI_PORT 3 (2*) solved the issue
and works with ESP32-S3 board V3.0.4

So my conclusion is : when selecting processor ESP32 -->> FSPI, VSPI, HSPI can be used
                      when selection processor ESP32-S3, SPI (1),2,3 must be used. (think 1 is internal...bit confusing)
		      I cannot judge on other processors, as I do not use them
                      Reworking above code in the library should solve the issues. 

Hope this cures a lot of headackes, and would appreciate your feedback to know if my ideas are confirmed.
regards
Ludo

HiGeek131, 2024-08-09:
Ture,dude!
I also encountered an infinite reboot situation this morning with the same error message as yours. After spending half a day debugging, I found that the root cause of the “SET_BUS_WRITE_MODE” error on line 81 in “TFT_eSPI.cpp” is out-of-bounds memory writing.

Looking closely at “SET_BUS_WRITE_MODE,” it actually executes the code “(volatile uint32_t)(((((0)>=2) ? (0x60024000 + (0-2) * 0x1000) : (0)) + 0x10)) = ((1UL << (27)))”. We can see that FSPI in the new version of the ESP32S3 firmware library refers to 0, and since 0>=2 is false, it returns the register address 0. Adding the offset 0x10 to register 0 and then performing a write operation causes a CPU panic and a reboot.

A temporary solution is to change “SPI_PORT” to “SPI_PORT+2” in the following lines within “TFT_eSPI\Processors\TFT_eSPI_ESP32_S3.h”:
`
#define _spi_cmd       (volatile uint32_t*)(SPI_CMD_REG(SPI_PORT))
#define _spi_user      (volatile uint32_t*)(SPI_USER_REG(SPI_PORT))
#define _spi_mosi_dlen (volatile uint32_t*)(SPI_MOSI_DLEN_REG(SPI_PORT))
#define _spi_w         (volatile uint32_t*)(SPI_W0_REG(SPI_PORT))
`
Alternatively, you can directly define SPI_PORT as 2 in:
`
#elif CONFIG_IDF_TARGET_ESP32S3
#define SPI_PORT FSPI
#endif
`

sjorsvr, 2024-12-19:
Hi, I have esp32s3 zero, st7789 display and use sd library, esp32 3.1.0 board, Arduino ide
the solution of Patsjemoe worked for me only with #define spi_port 2  (3 did not work for me)
earlier it did not work with the same settings and versions, but after clearing files in c:\Users\username\AppData\Local\arduino\cores and sketches it worked.
Arduino uses formerly compiled libraries and this caused it to not work earlier.