Bodmer/TFT_eSPI #3449: ST7701S and Adafruit Qualia RGB666

jfseaman, 2024-08-17:
Can TFT_eSPI support such a device combination? In testing, it seems very fast. 

https://www.adafruit.com/product/5800
https://www.adafruit.com/product/5793

I got them, "ported" some example code. Including "Benchmark" used for lots of libraries (MCU Friend, Adafruit GFX). 

I have been trying to come up with my own version of this combination for quite some time, not knowing that Adafruit make as exact of what I "need" from Januar 2024.

Sample init code is included in docs. GFX Library for Arduino by Moon On Our Nation can drive the combination with the information in the init code and signal lines. However, GFX does not have sprites.

I'm mostly an "applications" programmer. Lots of the stuff in TFTs I just don't get like "front_porch"?

brightproject, 2024-08-17:
> Can TFT_eSPI support such a device combination?

Hello @jfseaman 
You have raised a great topic, but it would have been better to do it in discussions.
I hope @Bodmer will move it there.
You have given little information on your question - what projects have you done, how did you connect the display, only through the `Qualia` board from `Adafruit`?
As far as I know, the `tfte_SPI` library does not support the `RGB` connection interface.
In general, your question is not completely clear - what do you want to get?

jfseaman, 2024-08-18:
I'm not a big Git user, did not know of "discussions". now I do. Because I worked on and wrote "targeted advertising" from the early 90's I have tendency to ignore the "banner area" where ads go and that is where "Discussions" is. ;)

Forgive me if it's not what you meant but I interpreted your question as rude so my reply may seem rude.  Before getting "offended" know that it's just how I interpreted it, it may not be how you intended it. I don't have a chip on my shoulder, I may have some ASC and ASD things going on and I do have a disconnect with how "young people" write and think. `However, I have been reading and writing with computers for a long long time. ;) Some enlightenment might be found at Dave's Garage and his books and videos on Autism such as https://www.youtube.com/watch?v=_kQqe9SDIC0

Remember, my long response could be an ASC or ASD thing as in "I have to"... Well, I know it is so...

Regarding your question, "what projects you have done". I'm not sure how this is important. Are you measuring me as a programmer? ;) I don't know hardware, sort of ;), I've been a programmer for 50+ years. Yes, I started learning this stuff at 8yo in 1966, binary, flowcharts, IBM 1401, Autocoder and punch cards (yuck). If you are old enough to remember BBSs then you may have found a MS-DOS program "bartman.com" in about 1990. I wrote that, no one cares, I'm good with that. I ported a driver for a very good but not popular 3COM ISA network card for Linux 0.93. There's that "sort of" again. I contributed it to the pile but don't knows if it was ever used by anyone other than me and the 2 guys I new form NN that had the same card, no one cares and I'm good with that. I spent 30 years as a professional including coding, design, project lead, project management, senior management, and "founder" level of 5 successful ".com"s. I started building "server farms" long before anyone used the word, mid 90's. More "sort of" on hardware. No one cares and I'm good with that. I made my money goal with those ".com"s and the stuff I made was real. All the companies are either still going or are part of something bigger now. No I won't list them for you because no one cares and I'm good with that. Back in the day there were "Play by mail" role playing games. This was my gaming of choice and ny choice of game was a form of "galactic exploration", a precursor to "SimEarth". We players joined, filled out paper sheets of "moves", sent it to the admins by USPS mail. The admins entered the "moves" and their judgments into the Macintosh, let the wheels turn then sent everyone the output. One weekend, I wrote a ~100,000+ line pseudo 3D graphics and database for IBM-PC, MS-DOS and CGA application to allow us gamers to enter, save and display our assets in the game as well as any other gamers assets we "found" per the game admins. Jolt Cola! The "user" gamer could "fly" around the "known universe" using the keypad and "see" the assets. Cutting edge? By 1988 standards? Well,  Windows, and X-Server were not public. And, no one cares and I'm good with that.  BTW: I tried to program on Mac. eff that chit. Well it was 1984 and it was a 128k Mac with one floppy so not fun for me. ;)

No one uses any of my "professional" code anymore. I'm good with that. Most of my current code is for motorcycle ECU and gauges. Speeduino is the ECU base code. I just "glue" outputs to the gauge device+code inputs and display. Take users inputs (yuck!) and use CAN or relay signals (yeah! old school) to tell the bike to do things.  Like I said, I am "applications" layer. I want to receive the data, process the data, output the data. None of my current code is suited to "community" exposure, or at least that's how I look at it. I could be wrong. Really, who cares about a replacement clock for a 1996 BMW R1100R that is Seeed Studio Xiao ESP32S3+their GC9A01 round display based, using Volos Projects clock face and other clock faces (user selectable), a DS3231 clock chip on the Xiao display, NTP to set time, WiFiProvisioner and  AsyncWebServer for user select WiFi Setup, Timezone and DST but is one code base for a bread board ESP32S3-Dev-Kit, generic GC9A01 and PCF8563 clock. It's only ever going to be fit to one bike. Maybe?

There, is my wick big enough for you. ;) If you were serious, take my reply as a serious return insult. If you are light hearted, take my reply light hearted and an ASC/ASD style response of much more than you were thinking of. ;)

BTW: There is "more" to my coding and computers as well as big wave surfing, high speed skateboarding and building surfboards and skateboards, motocross type dirtbike riding, etc.. Think the "hyperkinetic" Jordan character from the 1985 movie "Real Genius". I probably slept more than character was portrayed as not by much and you'll get the idea.

Now for the real issue. TFT_eSPI does not support RGB connection interface. I wanted ease of using "sprites" and "widgets" for the gauge UI. Not happening,  I'm good with that and that answers my question making an explanation of the physical connection useless. If you want to know, follow the links and study up on how hard it is to connect a ST7701S RGB display via breadboard to an ESP32S3, until I found about "io extenders" impossible. I found out by buying a LilyGo round display but it was still pushing my "hardware" skills and I could not get my choice of 720x720 ST7701S round display connected and working. So Adafruit did it for me, well not for ME but you know, now I can go back to applications level instead of hardware level. :) Maybe some of my development displays will work with the Qualia board :)

And if you are still reading and not boiling over this may send you. ;) The question is quite clear if you look at the hardware products and documentation. Not my issue of you don't see that. ;)

brightproject, 2024-08-18:
Thank you for the video, very informative.
You misunderstood the questions, in this community no one is rude to each other.
I just wanted to ask for a detailed answer to your problem.
I have also been working with `ESP32` and displays for a long time and once asked the author of this library a question about the `RGB interface`, but he replied that there was no support in sight.

> https://github.com/Bodmer/TFT_eSPI/discussions/2760#discussioncomment-6598314

I would recommend you to make your own PCB based on these schemes

[image]

[T-RGB.pdf](https://github.com/user-attachments/files/16651735/T-RGB.pdf)

 and use libraries🙂

> https://github.com/moononournation/Arduino_GFX/

jfseaman, 2024-08-18:
That is the board I am using. Distributed by Adafruit. Links were provided in OP.

The only missing component I wanted from TFT_eSPI was sprites.

I found a sprite library that uses Arduino GFX and perhaps Adafruit GFX. So I am now complete on the "libraries" front.