Bodmer/TFT_eSPI #3452: Not working with Waveshare ST7789V2 rounded display

step135, 2024-08-17:
I use ESP32 CAM.

> User_Setup.h

```
#define ST7789_2_DRIVER  // Use this definition for ST7789v2

// Define the display size
#define TFT_WIDTH  240     // ST7789v2 240 x 240 pixels
#define TFT_HEIGHT 280     // Change this if your display has a different resolution

// initialization sequence
#define TFT_INIT_DELAY 0x01

// Define the pins used for the display
#define TFT_MOSI  13       // Data out (MOSI)
#define TFT_SCLK  14       // Clock out (SCLK)
#define TFT_CS    15       // Chip select control pin (if used), set to -1 if not connected
#define TFT_DC    2       // Data/Command control pin (DC)
#define TFT_RST   12       // Reset pin (RST)

// Optional: Set the backlight pin if available
// #define TFT_BL   16      // Backlight control pin

// Set the SPI frequency
#define SPI_FREQUENCY  27000000 // 27 MHz (adjust as needed)
```

> text.ino

```
#include <TFT_eSPI.h>   // Include the graphics library (this includes the SPI library)

// Create an instance of the display driver
TFT_eSPI tft = TFT_eSPI();  // Invoke library, pins defined in User_Setup.h

void setup() {
  Serial.begin(115200);
  Serial.println("mx TFT2 started");

  pinMode(TFT_RST, OUTPUT);
  digitalWrite(TFT_RST, LOW);
  delay(50);  // Hold reset low for 50ms
  digitalWrite(TFT_RST, HIGH);
  delay(50);  // Wait for display to reset

  tft.init();
  tft.setRotation(1); // Set display rotation

  tft.fillScreen(TFT_BLUE); // Clear the screen
  tft.invertDisplay(true);
  tft.setTextColor(TFT_WHITE, TFT_BLACK);  // Set text color to white with black background
  tft.setTextSize(2); // Set the text size
  
  tft.setCursor(10, 10); // Set cursor position
  tft.print("Hello World! ST7789v2"); // Print "Hello World!" on the screen
 
  tft.setTextSize(6);
  tft.setTextColor(TFT_RED, TFT_ORANGE);
  tft.setCursor(10, 50);
  tft.print("end");

  Serial.println("end printed on TFT");
}

void loop() {
  if(true){
  tft.init();
  tft.setRotation(0); // Set display rotation

  //tft.fillScreen(TFT_BLUE); // Clear the screen
  tft.setTextColor(TFT_WHITE, TFT_BLACK);  // Set text color to white with black background
  tft.setTextSize(4); // Set the text size
  tft.setCursor(10, 150);
  tft.print("loop");
  Serial.println("loop on TFT");
  delay(5000);
  }
}
```

anishkarbhari, 2024-09-17:
I'm also facing a similar issue. If you have got any update on this please do share here

anishkarbhari, 2024-09-17:
[https://forum.lvgl.io/t/facing-st7789v2-240-280-display-issue-using-lvgl-display-look-black-not-showing-ui/14795](url)

You could try this once, although it did not work for me!

jcattone, 2024-09-25:
I'm in a similar boat with an integrated ESP32-S3+ST7789V2 display.  I'm trying to use Setup208_ST7789.h, as supplied from the Waveshare wiki's `[product page](https://www.waveshare.com/wiki/ESP32-S3-LCD-1.69#ST7789V2_Screen_Example)', where the '[sample demo](https://www.waveshare.com/wiki/ESP32-S3-LCD-1.69#Demo)' download provides the libraries.  I did update all of the included libraries to the latest releases.

I'd be happy to help test, or run diagnostics, if that would help.