Bodmer/TFT_eSPI #3454: ESP32 C3 Super Mini + GC9A01 = Blank Screen

TzOk83, 2024-08-21:
A description of the problem and the conditions that cause it to occur
I can't make TFT eSPI to work with ESP32 C3 Super Mini with GC9A01 display. The same HW setup works with Adafruit GC9A01 library.

IDE (e.g. Arduino or PlatformIO)
Arduino IDE

TFT_eSPI library version (try the latest, the problem may have been resolved!) from the Manage Libraries... menu
2.5.43

Board package version (e.g. 2.0.3) available from the Boards Manager... menu
3.0.3

Procesor, e.g RP2040, ESP32 S3 etc
ESP32 C3

TFT driver (e.g. ILI9341), a link to the vendors product web page is useful too.
GC9A01

Interface type (SPI or parallel)
SPI

Plus further information as appropriate to the problem:
1. TFT to processor connections used
SDA (MISO) - 6
SCLK (SCK) - 4
CS - 10
DC - 9
RST - 8

3. A zip file containing your setup file (just drag and drop in message window - do not paste in long files!)
`
#define USER_SETUP_INFO "User_Setup"

#define GC9A01_DRIVER

#define TFT_WIDTH  240 // ST7789 240 x 240 and 240 x 320
#define TFT_HEIGHT 240 // GC9A01 240 x 240

#define TFT_MOSI 6 
#define TFT_SCLK 4
#define TFT_CS   10
#define TFT_DC   9
#define TFT_RST  8

#define LOAD_GLCD
#define LOAD_FONT2 
#define LOAD_FONT4
#define LOAD_FONT6 
#define LOAD_FONT7
#define LOAD_FONT8

#define LOAD_GFXFF

#define SMOOTH_FONT

#define SPI_FREQUENCY 40000000
#define SPI_READ_FREQUENCY 6000000
`

TzOk83, 2024-08-21:
I have also tested another config
```
#define TFT_CS   7
#define TFT_MOSI 6
#define TFT_MISO 5
#define TFT_SCLK 4
#define TFT_DC   8
#define TFT_RST  10
```

and ESP32 v3.0.4 from Espressif GitHub.

No success so far :(

JimDrewGH, 2024-08-22:
Roll your board package back to v2.0.14 (or earlier) and it will likely work.  There is a thread here that has changes to the SPI port usage that will fix this with v2.0.15 and later. Until you fix it, you will need to roll back the core.

AleksNik2k, 2025-12-12:
Добавляю решение проблемы с дисплеем GC9A01 на контроллерах ESP32-S3, ESP32-C2, ESP32-C3 и т.д.
Добавляем в ваш проект обязательно SPI.h
#include <SPI.h>
И инициализируем SPIClass обязательно первым
SPIClass hspi(HSPI);
И только потом все остальное.
TFT_eSPI tft = TFT_eSPI();

Данное решение работает и с другими библиотеками типа LovyanGFX и LVGL. Проверял на  последних версиях библиотек.

cwgstreet, 2026-10-24:
I'm seeing what looks like the same root cause on an ESP32-C3 with a GC9A01, with a more specific symptom that may help the maintainers.

Environment:
- TFT_eSPI 2.5.43 (via PlatformIO)
- PlatformIO espressif32 7.0.1, Arduino-ESP32 core 3.20017.241212
- Board: esp32-c3-devkitm-1
- Display: GC9A01 240x240, 4-wire SPI

Symptom:
Instead of a blank screen, the device panics with a Store access fault (MTVAL 0x10) on the first SPI command. Decoded with esp32_exception_decoder:

  #0 TFT_eSPI::begin_tft_write() at TFT_eSPI.cpp:81
     (inlined by) TFT_eSPI::writecommand(unsigned char) at TFT_eSPI.cpp:982
  #1 TFT_eSPI::init(unsigned char) at TFT_eSPI.cpp:693

Cause:
In Processors/TFT_eSPI_ESP32_C3.h:

`  #define SPI_PORT SPI2_HOST`

SPI2_HOST is enum value 1 in the ESP-IDF headers. The C3 REG_SPI_BASE(i) macro only returns a base for i == 2:

 ` #define REG_SPI_BASE(i) (((i)==2) ? DR_REG_SPI2_BASE : 0)`

so SPI_USER_REG(SPI_PORT) evaluates to 0x10. The first SET_BUS_WRITE_MODE write (*_spi_user = ...) then faults.

Fix tested on hardware:

 ` #define SPI_PORT 2`

With this change the GC9A01 initialises and draws correctly on the C3.

This is consistent with the core-version comments above: the SPI port convention appears to have changed in Arduino-ESP32 2.0.15 and later. 

**_Note that I used Claude AI to understand and fix this issue, as well as draft this comment, as it was far beyond my skills; I hope it is useful not AI slop_**