Bodmer/TFT_eSPI #3456: TFT_eSPI is not working for ESP32 and ILI9341 DRIVER

Sinamajidi, 2024-08-21:
I have an ESP32 Devkit C board and a 2.4" TFT Shield with a red board color. The shield has pins on both sides: one side has parallel data and SD pins, and the other side has CS, RST, WR, RD, F_CS, and RS pins. I connected the TFT Display to the ESP32 and uncommented the section about it in User_Setup.h. However, I encountered a problem when I tried to build and upload the Smooth_Graphics_Demo example to the ESP32. The compiler terminated and returned the error that TOUCH_CS is not defined, even though the TFT shield doesn't have any pin with this name. To resolve this, I uncommented one of the definitions about this pin in User_Setup.h and set the TOUCH_CH to the CS of the TFT LCD, and the error disappeared. However, after uploading the program to my ESP32, the shield didn't display or sense anything, and the serial monitor didn't show any problems. I'm unsure of what to do next. I've provided all the necessary information. Please help me with this.

## Serial Monitor:
```

rst:0x1 (POWERON_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:2
load:0x3fff0030,len:1184
load:0x40078000,len:13232
load:0x40080400,len:3028
entry 0x400805e4
Booting...

```

## Main Program (Modified Smooth_Graphics_Demo Example):
```

// Sketch to demonstrate smooth (anti-aliased) graphics functions:
// Smooth graphics result in less pixel resolution jaggedness.

#include <TFT_eSPI.h> // Master copy here: https://github.com/Bodmer/TFT_eSPI

TFT_eSPI tft = TFT_eSPI();  // Invoke library, pins defined in User_Setup.h

TFT_eSprite spr = TFT_eSprite(&tft);

unsigned int rainbow(byte value);
void getCoord(int16_t x, int16_t y, float *xp1, float *yp1, float *xp2, float *yp2, int16_t r1, int16_t r2, float a);

// =========================================================================
// Setup
// =========================================================================
void setup() {
  Serial.begin(115200);
  Serial.println("Booting...");

  // Initialise the screen
  tft.init();

  // Ideally set orientation for good viewing angle range because
  // the anti-aliasing effectiveness varies with screen viewing angle
  tft.setRotation(0);

  tft.fillScreen(TFT_BLACK);

  // Small sprite for spot demo
  spr.createSprite(23, 23);
}

// =========================================================================
// Loop
// =========================================================================
void loop() {

  // drawSpot is for small anti-aliased circles, coordinates and radius are
  // floating point to allow sub-pixel positioning (large circles will
  // be slow to draw). Use fillSmoothCircle() for large circles.
  // In this case black is the background colour for the anti-aliasing
  float x = 10.5;
  float y = 10.5;
  float r = 8.6;
  tft.drawSpot(x, y, r, TFT_WHITE, TFT_BLACK);

  // Fill sprite with a colour
  spr.fillSprite(TFT_RED);
  // Draw spot in sprite, the background colour is omitted so function
  // reads background colour for aliasing. (To use this method with direct write
[... 497 lines cut ...]

// Optional reduced SPI frequency for reading TFT
#define SPI_READ_FREQUENCY  20000000

// The XPT2046 requires a lower SPI clock rate of 2.5MHz so we define that here:
#define SPI_TOUCH_FREQUENCY  2500000

// The ESP32 has 2 free SPI ports i.e. VSPI and HSPI, the VSPI is the default.
// If the VSPI port is in use and pins are not accessible (e.g. TTGO T-Beam)
// then uncomment the following line:
//#define USE_HSPI_PORT

// Comment out the following #define if "SPI Transactions" do not need to be
// supported. When commented out the code size will be smaller and sketches will
// run slightly faster, so leave it commented out unless you need it!

// Transaction support is needed to work with SD library but not needed with TFT_SdFat
// Transaction support is required if other SPI devices are connected.

// Transactions are automatically enabled by the library for an ESP32 (to use HAL mutex)
// so changing it here has no effect

// #define SUPPORT_TRANSACTIONS

```

Splarkszter, 2024-08-22:
I'm having having the same problems as you (except for the touch thing).
Setup file should be configured correctly, everything wired, etc. Blank white screen.
**ST7789** Red Tab.

The only way I have been able to make the display work is with **adafruit library** BUT **only on software SPI** mode, I have not been able to use Hardware SPI at all!

elgerg, 2024-08-22:
What does your serial monitor output?
Is it crash/rebooting?

Splarkszter, 2024-08-22:
I'm a complete idiot... I was using 10K resistors on the 3.3V pins now it works!

Reason of why they were there is because I tested with arduino first, and it has 5V for the pins...

I still don't understand why it worked using SoftwareSPI or whatever... maybe the lower clock was able to get trough the resistance?

Sinamajidi, 2024-08-23:
> I'm having having the same problems as you (except for the touch thing). Setup file should be configured correctly, everything wired, etc. Blank white screen. **ST7789** Red Tab.
>
> The only way I [...]

@Splarkszter 

My TFT display doesn't have MISO, MOSI pins for using the Adafruit Library or any other SPI libraries. Instead, it has RW, RD, CS,... pins which are compatible with this library. However, now, as I commented first, it doesn't do anything!

arduhe, 2024-09-01:
Same behaviour here - I am using an ESP32_dev with ILI9341

ets Jun  8 2016 00:22:57

rst:0x8 (TG1WDT_SYS_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:2
load:0x3fff0030,len:1184
load:0x40078000,len:13232
load:0x40080400,len:3028
entry 0x400805e4

Even after downgrading from 2.5.**43** to 2.5.**34** the problem persists