Bodmer/TFT_eSPI #3457: ESP32 S3 ILI9341 problems with rotation of display and rendering

MYCHMA, 2024-08-22:
To minimise effort to resolve issues the following should be provided as a minimum:
1. A description of the problem and the conditions that cause it to occur     
Problem occures with setRotation(3) and fillrect() maybe even more funcions.
the problem is only on platformio the arduino seams to be funcioning right.

3. IDE (e.g. Arduino or PlatformIO)
platform io
5. TFT_eSPI library version (try the latest, the problem may have been resolved!) from the Manage Libraries... menu
latest 2.5.43
7. Board package version (e.g. 2.0.3) available from the Boards Manager... menu
2.0.11 arduino platformio idk (maybe latest?)
9. Procesor, e.g RP2040, ESP32 S3 etc
esp32 s3
11. TFT driver (e.g. ILI9341), a link to the vendors product web page is useful too.
ili9341 2.4 ich generic red
13. Interface type (SPI or parallel)
SPI

```c++
Serial.begin(115200);
tft.begin();
tft.setRotation(3);
Serial.print(tft.height());
Serial.print(" a " );
Serial.println(tft.width());
tft.fillRect(0, 0, 320, 240, TFT_BLUE);
``` 
on arduino output is right:
240 a 320
on platformio is 
320 a 240

its  seam to be the problem is that the setRotation doesnt overide the rendering system.

I would love to show pictures but i dont have a camera at the moment lol(in 2024)
 but on arduino the rectangle is for a whole screen 
and for platforio is like 2/3 my guess its seams to be rotated 90 degrees 
 
Btw: I did run your library thru lines of code counter and like 500k of lines of code ??? wau 
THANK YOU FOR MAKING MY DREAMS A REALITY.

BTW2:Im makeing my own OS for microcontroller to make my computer , smartphone.
its not at github yet but it will be on makers faire in brno yey

and again Thank you :-)

bionicbone, 2024-09-18:
For info, I had this issue also and spent quite a bit of time messing, trying to find the cause.

My system is Visual Studio with Visual Micro installed (**not platformio**), arduino ESP32 CORE v2.0.17, TFT_eSPI v 2.5.43.

I beleive it could be caused by a failed claibration since initially the display rotation was set to 3 and displaying correctly, but touch sensing was as though in rotation 2. However, while investigating sometimes it was OK. I got to a stage where I could no longer replicate the issue so had to stop investigating. It almost felt like the touch senor on the display had become "bedded in" on the corners and now worked perfectly well. 

At the time I was looking at changing this code in calibrateTouch(); to be more specific about rotation(3), but I was not conviced this was correct, become this makes X & Y coords become transposed.

`    //touchCalibration_rotate = true;
    if(values[4] > values[5]) touchCalibration_rotate = true;`

MYCHMA, 2024-09-18:
thanks for replay but i dont use the touch screen at all.  Idk whats going on and why last version is so old now

arduhe, 2024-11-01:
Although I think TFT_eSPI is a great library - especially in supporting different TFTs - IMO it has a design flaw when using its touchscreen support. If you compare the code in Touch.cpp with the [XPT2046_Touchscreen library](https://github.com/PaulStoffregen/XPT2046_Touchscreen)  it's easy to see that TFT_eSPI doesn't take any care for the rotation. As a result the return coordinates only fit for rotation=1.

I've made a small modification to the calibration example to verify this

```
--- a/examples/Generic/Touch_calibrate/Touch_calibrate.ino
+++ b/examples/Generic/Touch_calibrate/Touch_calibrate.ino
@@ -35,6 +35,9 @@ void setup() {
 
   // Clear the screen
   tft.fillScreen(TFT_BLACK);
+  tft.setRotation(3);
+  tft.println();
+  tft.printf("rot=%d, w=%d, h=%d", tft.getRotation(), tft.width(), tft.height());
   tft.drawCentreString("Touch screen to test!",tft.width()/2, tft.height()/2, 2);
 }

```

and ran the sketch with all 4 rotations and this is what happened 

[image]
[image]
[image]
[image]

As you can see - only rotation=1 yields the correct coordinates for the touch, all others are wrong.

arduhe, 2024-11-01:
☺️
I am sorry - but I think I must apologize. The flaw was on my side - as I didn't see the crucial clue in the calibration example.

```
   // Set the rotation to the orientation you wish to use in your project before calibration
   // (the touch coordinates returned then correspond to that rotation only)
   tft.setRotation(1);
```

If both rotations - that while measuring the calibration data and that while running the rest of the sketch - match, everything works fine, i.e. if you're planning to use another rotation as 1 you must use that while calibrating, too.

I think, the only odd thing that remains - if you're planning to alter the rotation in your sketch dynamically, you have to manage the corresponding calibration data on your own and you must set them when changing rotation.