Bodmer/TFT_eSPI #3464: ESP32-C3 using TFT_eSPI@^2.5.43 version, what is the reason for repeated reboots?

43823845, 2024-08-27:
Guru Meditation Error: Core  0 panic'ed (Store access fault). Exception was unhandled.

Core  0 register dump:
MEPC    : 0x4205061a  RA      : 0x42050612  SP      : 0x3fca6c80  GP      : 0x3fc8c600  
TP      : 0x3fc89c68  T0      : 0x00000003  T1      : 0xffc3ffff  T2      : 0x00000020
S0/FP   : 0x3fca2b50  S1      : 0x00000000  A0      : 0x00000001  A1      : 0xffffffff
A2      : 0x00000000  A3      : 0xfdffffff  A4      : 0x08000000  A5      : 0x00000010  
A6      : 0xffffffff  A7      : 0x04c4b400  S2      : 0x00000002  S3      : 0x00000001  
S4      : 0x000000a0  S5      : 0x3fc8dd6c  S6      : 0x00000a00  S7      : 0x00000000
S8      : 0x00000000  S9      : 0x00000000  S10     : 0x00000000  S11     : 0x00000000  
T3      : 0x0196e6aa  T4      : 0x0000000f  T5      : 0x003c0000  T6      : 0x00000003
MSTATUS : 0x00001881  MTVEC   : 0x40380001  MCAUSE  : 0x00000007  MTVAL   : 0x00000010
MHARTID : 0x00000000

ZnMnCr, 2024-08-27:
When I used the ST7789_DRIVER and changed the pin to suit the air core esp32c3,I meet the similar problem just as yours.
[image]
[image]

[image]
`#include <Arduino.h>
#include "TFT_eSPI.h"
#include <SPI.h>
// 定义LED所接的引脚
const int ledPin = 13;
TFT_eSPI tft = TFT_eSPI(); // 初始化TFT_eSPI对象
void setup()
{
  // 初始化数字引脚13为输出模式
  pinMode(ledPin, OUTPUT);
  Serial.begin(115200);
   tft.init();
   tft.setRotation(0);
   tft.fillScreen(TFT_GREEN); // 初始化屏幕颜色大写
}

void loop()
{
  tft.fillScreen(TFT_GREEN); // 初始化屏幕颜色大写
  // 将引脚设置为高电平来点亮LED
  digitalWrite(ledPin, HIGH);
  // 等待1秒
  delay(1000);
  // 将引脚设置为低电平来熄灭LED
  digitalWrite(ledPin, LOW);
  // 再次等待1秒
  delay(1000);
}`

43823845, 2024-08-27:
> 当我使用 ST7789_DRIVER 并改变引脚以适应空芯 esp32c3 时，我遇到了与您类似的问题。 ![图像](https://private-user-images.githubusercontent.com/84188568/361796812-0d843d38-5548-49c6-9b34-66f6d2e7f7e4.png?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3MjQ3NjYyOTgsIm5iZiI6MTcyNDc2NTk5OCwicGF0aCI6Ii84NDE4ODU2OC8zNjE3OTY4MTItMGQ4NDNkMzgtNTU0OC00OWM2LTliMzQtNjZmNmQyZTdmN2U0LnBuZz9YLUFtei1BbGdvcml0aG09QVdTNC1ITUFDLVNIQTI1NiZYLUFtei1DcmVkZW50aWFsPUFLSUFWQ09EWUxTQTUzUFFLNFpBJTJGMjAyNDA4MjclMkZ1cy1lYXN0LTElMkZzMyUyRmF3czRfcmVxdWVzdCZYLUFtei1EYXRlPTIwMjQwODI3VDEzMzk1OFomWC1BbXotRXhwaXJlcz0zMDAmWC1BbXotU2lnbmF0dXJlPWYzN2YwMzMyMzRlOGVjOTc4ODZjYWQwYWQxMjllMTQ2ZjllZjUzMjc2YzkwMTg5MTkyZjU2NzMxZTY5ZjczMWQmWC1BbXotU2lnbmVkSGVhZGVycz1ob3N0JmFjdG9yX2lkPTAma2V5X2lkPTAmcmVwb19pZD0wIn0.XOcB4N2-73bhcXwP2nj1q9h4oWcRtB5QGFHayCbI-P8) ![图像](https://private-user-images.githubusercontent.com/84188568/361796946-87e588eb-26ac-4eb2-95e8-4e6bb96dc558.png?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3MjQ3NjYyOTgsIm5iZiI6MTcyNDc2NTk5OCwicGF0aCI6Ii84NDE4ODU2OC8zNjE3OTY5NDYtODdlNTg4ZWItMjZhYy00ZWIyLTk1ZTgtNGU2YmI5NmRjNTU4LnBuZz9YLUFtei1BbGdvcml0aG09QVdTNC1ITUFDLVNIQTI1NiZYLUFtei1DcmVkZW50aWFsPUFLSUFWQ09EWUxTQTUzUFFLNFpBJTJGMjAyNDA4MjclMkZ1cy1lYXN0LTElMkZzMyUyRmF3czRfcmVxdWVzdCZYLUFtei1EYXRlPTIwMjQwODI3VDEzMzk1OFomWC1BbXotRXhwaXJlcz0zMDAmWC1BbXotU2lnbmF0dXJlPWQ3Yzg3OTUyODk0ODJlOTgwNWZjYTgzMTgzNzY2ZGI3ZDUzYWEwYzAwZTIyYzIxMmIxMTViYjYwNDY0MjRlOGImWC1BbXotU2lnbmVkSGVhZGVycz1ob3N0JmFjdG9yX2lkPTAma2V5X2lkPTAmcmVwb19pZD0wIn0.ooJxo0IUju7pLi8B0K3Y_iK1T9bboZ7_RKYOxbql5xQ)
> 
> ![图像](https://private-user-images.githubusercontent.com/84188568/361796891-a9b15aa7-bc57-420f-aef3-796954f01c14.png?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3MjQ3NjYyOTgsIm5iZiI6MTcyNDc2NTk5OCwicGF0aCI6Ii84NDE4ODU2OC8zNjE3OTY4OTEtYTliMTVhYTctYmM1Ny00MjBmLWFlZjMtNzk2OTU0ZjAxYzE0LnBuZz9YLUFtei1BbGdvcml0aG09QVdTNC1ITUFDLVNIQTI1NiZYLUFtei1DcmVkZW50aWFsPUFLSUFWQ09EWUxTQTUzUFFLNFpBJTJGMjAyNDA4MjclMkZ1cy1lYXN0LTElMkZzMyUyRmF3czRfcmVxdWVzdCZYLUFtei1EYXRlPTIwMjQwODI3VDEzMzk1OFomWC1BbXotRXhwaXJlcz0zMDAmWC1BbXotU2lnbmF0dXJlPTE5ZjIwZmExYzA2ZThmNTEzNjFlOTQzMTY0YmQyNmJhYWQ0YzM1OTkyNDllMzU2N2RlN2NhYTNiM2I3ODMyOWQmWC1BbXotU2lnbmVkSGVhZGVycz1ob3N0JmFjdG9yX2lkPTAma2V5X2lkPTAmcmVwb19pZD0wIn0._F_YdLOBSxKl2MQhyHD8P_4tkt3J09UwmUt6NR6OINQ) `#include <Arduino.h> #include "TFT_eSPI.h" #include <SPI.h> // 定义 LED 所接的引脚 const int ledPin = 13; TFT_eSPI tft = TFT_eSPI(); // 初始化TFT_eSPI对象 void setup() { // 初始化数字引脚13为输出模式 pinMode(ledPin, OUTPUT); 串行.开始(115200); tft.init(); tft.setRotation(0); tft.fillScreen(TFT_GREEN); //初始化屏幕颜色大写 }
> 
> 无效循环（） { tft.fillScreen（TFT_GREEN）; //初始化屏幕颜色大写 //将引脚设置为高电平来点亮LED digitalWrite(ledPin, HIGH); // 等待1秒 delay(1000); // 将引脚设置为低电平来启动 LED digitalWrite(ledPin, LOW); // 再次等待1秒 delay(1000); }`

43823845, 2024-08-27:
> 当我使用 ST7789_DRIVER 并改变引脚以适应空芯 esp32c3 时，我遇到了与您类似的问题。 ![图像](https://private-user-images.githubusercontent.com/84188568/361796812-0d843d38-5548-49c6-9b34-66f6d2e7f7e4.png?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3MjQ3NjYyOTgsIm5iZiI6MTcyNDc2NTk5OCwicGF0aCI6Ii84NDE4ODU2OC8zNjE3OTY4MTItMGQ4NDNkMzgtNTU0OC00OWM2LTliMzQtNjZmNmQyZTdmN2U0LnBuZz9YLUFtei1BbGdvcml0aG09QVdTNC1ITUFDLVNIQTI1NiZYLUFtei1DcmVkZW50aWFsPUFLSUFWQ09EWUxTQTUzUFFLNFpBJTJGMjAyNDA4MjclMkZ1cy1lYXN0LTElMkZzMyUyRmF3czRfcmVxdWVzdCZYLUFtei1EYXRlPTIwMjQwODI3VDEzMzk1OFomWC1BbXotRXhwaXJlcz0zMDAmWC1BbXotU2lnbmF0dXJlPWYzN2YwMzMyMzRlOGVjOTc4ODZjYWQwYWQxMjllMTQ2ZjllZjUzMjc2YzkwMTg5MTkyZjU2NzMxZTY5ZjczMWQmWC1BbXotU2lnbmVkSGVhZGVycz1ob3N0JmFjdG9yX2lkPTAma2V5X2lkPTAmcmVwb19pZD0wIn0.XOcB4N2-73bhcXwP2nj1q9h4oWcRtB5QGFHayCbI-P8) ![图像](https://private-user-images.githubusercontent.com/84188568/361796946-87e588eb-26ac-4eb2-95e8-4e6bb96dc558.png?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3MjQ3NjYyOTgsIm5iZiI6MTcyNDc2NTk5OCwicGF0aCI6Ii84NDE4ODU2OC8zNjE3OTY5NDYtODdlNTg4ZWItMjZhYy00ZWIyLTk1ZTgtNGU2YmI5NmRjNTU4LnBuZz9YLUFtei1BbGdvcml0aG09QVdTNC1ITUFDLVNIQTI1NiZYLUFtei1DcmVkZW50aWFsPUFLSUFWQ09EWUxTQTUzUFFLNFpBJTJGMjAyNDA4MjclMkZ1cy1lYXN0LTElMkZzMyUyRmF3czRfcmVxdWVzdCZYLUFtei1EYXRlPTIwMjQwODI3VDEzMzk1OFomWC1BbXotRXhwaXJlcz0zMDAmWC1BbXotU2lnbmF0dXJlPWQ3Yzg3OTUyODk0ODJlOTgwNWZjYTgzMTgzNzY2ZGI3ZDUzYWEwYzAwZTIyYzIxMmIxMTViYjYwNDY0MjRlOGImWC1BbXotU2lnbmVkSGVhZGVycz1ob3N0JmFjdG9yX2lkPTAma2V5X2lkPTAmcmVwb19pZD0wIn0.ooJxo0IUju7pLi8B0K3Y_iK1T9bboZ7_RKYOxbql5xQ)
> 
> ![图像](https://private-user-images.githubusercontent.com/84188568/361796891-a9b15aa7-bc57-420f-aef3-796954f01c14.png?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3MjQ3NjYyOTgsIm5iZiI6MTcyNDc2NTk5OCwicGF0aCI6Ii84NDE4ODU2OC8zNjE3OTY4OTEtYTliMTVhYTctYmM1Ny00MjBmLWFlZjMtNzk2OTU0ZjAxYzE0LnBuZz9YLUFtei1BbGdvcml0aG09QVdTNC1ITUFDLVNIQTI1NiZYLUFtei1DcmVkZW50aWFsPUFLSUFWQ09EWUxTQTUzUFFLNFpBJTJGMjAyNDA4MjclMkZ1cy1lYXN0LTElMkZzMyUyRmF3czRfcmVxdWVzdCZYLUFtei1EYXRlPTIwMjQwODI3VDEzMzk1OFomWC1BbXotRXhwaXJlcz0zMDAmWC1BbXotU2lnbmF0dXJlPTE5ZjIwZmExYzA2ZThmNTEzNjFlOTQzMTY0YmQyNmJhYWQ0YzM1OTkyNDllMzU2N2RlN2NhYTNiM2I3ODMyOWQmWC1BbXotU2lnbmVkSGVhZGVycz1ob3N0JmFjdG9yX2lkPTAma2V5X2lkPTAmcmVwb19pZD0wIn0._F_YdLOBSxKl2MQhyHD8P_4tkt3J09UwmUt6NR6OINQ) `#include <Arduino.h> #include "TFT_eSPI.h" #include <SPI.h> // 定义 LED 所接的引脚 const int ledPin = 13; TFT_eSPI tft = TFT_eSPI(); // 初始化TFT_eSPI对象 void setup() { // 初始化数字引脚13为输出模式 pinMode(ledPin, OUTPUT); 串行.开始(115200); tft.init(); tft.setRotation(0); tft.fillScreen(TFT_GREEN); //初始化屏幕颜色大写 }
> 
> 循环无效（） { tft.fillScreen（TFT_GREEN）; //初始化屏幕颜色大写 //将引脚设置为高电平来点亮LED digitalWrite(ledPin, HIGH); // 等待1秒 delay(1000); // 将引脚设置为低电平来启动 LED digitalWrite(ledPin, LOW); // 再次等待1秒 delay(1000); }`

Yes I found this problem when using LVGL, at first I thought it was LVGL incompatibility, after troubleshooting the problem seems to be in the screen buffer setup, as long as you load lv_tft_espi_create(TFT_HOR_RES, TFT_VER_RES, draw_buf, sizeof(draw_buf)); this piece of code will then reboot repeatedly.

ZnMnCr, 2024-08-28:
I test the solution that you mentioned ,it works!

43823845, 2024-08-28:
> 我测试了您提到的解决方案，它有效！
Can you share the exact method, I tried #define USE_HSPI_PORT but still not working.

ZnMnCr, 2024-08-28:
According to the above mentioned, change the esp32 version of arduinoIDE and roll back to 2.0.14

43823845, 2024-08-28:
> According to the above mentioned, change the esp32 version of arduinoIDE and roll back to 2.0.14

Thank you for your answer, I am testing as you said

ZnMnCr, 2024-08-28:
I have another question, I can change the version of the board in ARDUINOIDE, what should I do in platformIO and where should I go to change the version?

mvoss96, 2024-08-30:
platform = espressif32@6.6.0

JacketJack113, 2024-09-05:
Verified another solution from #3429 , I was using esp32-s3 pico and followed the solution from @patsjemoe . In short, his solution was to replace `#define SPI_PORT FSPI` into `#define SPI_PORT 3` in TFT_eSPI_ESP32_S3.h in the Processors folder.

With that, I further tested it with SPI_PORT 0-3, so instead of writing `#define SPI_PORT 3`, I replaced it with `#define SPI_PORT N`, where N is a value ranged from 0-3, uploading the code with the latest version of the esp32 board package (3.0.4), TFT_eSPI Library (2.5.43) and arduino IDE 2.3.2. After uploading the code with the tweaked TFT_eSPI_ESP32_S3.h, I recorded the result and found out that there're 3 cases regarding the tweak.

Case 1: Replacing the line with `#define SPI_PORT 0` or `#define SPI_PORT 1`.
Tweaking the code with these 2 definition would **replicate the Guru Meditation Error** as shown by @43823845 .

Case 2: Replacing the line with `#define SPI_PORT 2`.
The only definition that works flawlessly in the testing, the screen does display properly, no Guru Meditation Error was produced.

Case 3: Replacing the line with `#define SPI_PORT 3`.
Guru Meditation Error was stopped being produced. However, the screen didn't display properly, i.e. blank screen.

The test wasn't perfect though, I'm not quite sure how the SPI_PORT was defined inside the board package, nor I tested the solution with the default HSPI pin. And for case 1, instead of the exact runtime error as @43823845 , mine's _Guru Meditation Error: Core 1 panic'ed (StoreProhibited)_, which I'm not quite sure if there's a difference in between those 2 error (as my code breaks exactly on the line where it tries to initiatize the ILI9341 display).

TL;DR, Replace `#define SPI_PORT FSPI` into `#define SPI_PORT 2` in TFT_eSPI_ESP32_S3.h in the Processors folder is another possible solution that solves the Guru Meditation Error boot loop issue.

Jackiin, 2024-11-05:
By rolling back to esp32 2.0.14 in board manager, the problem was solved perfectly. And if you cannot see Nologo ESP32C3 Super Mini anymore, you can just select ESP32C3 Dev Board from the menu.

KamalTolouiFard, 2025-04-17:
Hello i have a similar issue with ESP8266 and ST7796 display driver but the reboot happens when i call a drawing function like `fillScreen()` and etc.

i try downgrade my board package from 3.1.2 to 2.4.0 and report back.