Bodmer/TFT_eSPI #3466: Arduino V3 support?

hitecSmartHome, 2024-08-28:
**Is there a plan to support Arduino V3 and ESP IDF 5.x?**

There was a major SPI reformat and other things which breaks the library.
I suspect you know about the issue already.

lbernstone, 2024-08-28:
Can you please add more detail about what is broken?  It is working for me in v3.0.4

hitecSmartHome, 2024-08-29:
Will write details asap. I tought it was common. The lib crashing the esp and if we modify some spi defines it just not working.

hitecSmartHome, 2024-08-30:
Okay. The display works but I was using a JPEGDecoder library which stopped working. This library is also from Bodmer.

hitecSmartHome, 2024-08-30:
I also can't use tft.pushImage with byte arrays. The image is garbage but the correct dimensions

hitecSmartHome, 2024-08-30:
Sorry. my bad. It works allright