Bodmer/TFT_eSPI #3476: Issue with 2.5.43 and ST7789

ianm8, 2024-09-07:
Hi

I have previously used this library (version 2.5.34) with an ST7789 and an Pi Pico (RP2040) without issue. After updating to version 2.5.43 the display no longer works. That is, nothing is displayed.

Here is the user setup file:

#define ST7789_2_DRIVER    // Minimal configuration option, define additional parameters below for this display
#define CGRAM_OFFSET      // Library will add offsets required
#define TFT_WIDTH  135
#define TFT_HEIGHT 240
#define TFT_MISO D16
#define TFT_MOSI D19
#define TFT_SCLK D18
#define TFT_CS   D17
#define TFT_DC   D20
#define TFT_RST  D21
#define LOAD_GLCD   // Font 1. Original Adafruit 8 pixel font needs ~1820 bytes in FLASH
#define SPI_FREQUENCY  40000000
#define SPI_READ_FREQUENCY  20000000
#define SPI_TOUCH_FREQUENCY  2500000

Here is the test program:

/*

  Sketch to show how a Sprite is created, how to draw pixels
  and text within the Sprite and then push the Sprite onto
  the display screen.

  Example for library:
  https://github.com/Bodmer/TFT_eSPI

  A Sprite is notionally an invisible graphics screen that is
  kept in the processors RAM. Graphics can be drawn into the
  Sprite just as it can be drawn directly to the screen. Once
  the Sprite is completed it can be plotted onto the screen in
  any position. If there is sufficient RAM then the Sprite can
  be the same size as the screen and used as a frame buffer.

  A 16 bit Sprite occupies (2 * width * height) bytes in RAM.

  On a ESP8266 Sprite sizes up to 126 x 160 can be accommodated,
  this size requires 40kBytes of RAM for a 16 bit colour depth.
  
  When 8 bit colour depth sprites are created they occupy
  (width * height) bytes in RAM, so larger sprites can be
  created, or the RAM required is halved.

*/

// Set delay after plotting the sprite
#define DELAY 1000

// Width and height of sprite
#define WIDTH  240
#define HEIGHT 135

#include <SPI.h>
#include <TFT_eSPI.h>                 // Include the graphics library (this includes the sprite functions)

TFT_eSPI    tft = TFT_eSPI();         // Declare object "tft"
TFT_eSprite spr = TFT_eSprite(&tft);  // Declare Sprite object "spr" with pointer to "tft" object

static uint32_t frequency = 7105000;
#define _rev(x) (uint16_t)(((x)>>8)|((x)<<8))

void refresh(const uint32_t f)
{
  spr.fillSprite(TFT_BLACK);
  spr.drawLine(0, 0, WIDTH-1, HEIGHT-1, TFT_GREEN);
  spr.drawPixel(2,10,TFT_RED);
  spr.drawRect(50,10,10,10,TFT_RED);
  spr.drawRect(50,30,10,10,TFT_GREEN);
  spr.drawRect(50,50,10,10,TFT_BLUE);
  spr.drawRect(50,70,10,10,TFT_YELLOW);
  spr.drawRect(50,90,10,10,TFT_PURPLE);
  spr.drawRect(30,10,10,10,_rev(TFT_RED));
  spr.drawRect(30,30,10,10,_rev(TFT_GREEN));
  spr.drawRect(30,50,10,10,_rev(TFT_BLUE));
  spr.drawRect(30,70,10,10,_rev(TFT_YELLOW));
  spr.drawRect(30,90,10,10,_rev(TFT_PURPLE));

  //spr.drawString("Sprite",100,10,4);
  //spr.drawString("VK7IAN",100,30,3);
  spr.setTextColor(TFT_BLUE,TFT_BLACK);
  spr.drawString("VK7IAN",100,50,2);
  spr.setTextColor(TFT_GREEN,TFT_BLACK);
  spr.drawString("VK7IAN",100,70,1);
  spr.setTextSize(3);
  //spr.setTextColor(radio.locked()?TFT_RED:TFT_WHITE,TFT_BLACK);
  spr.setTextColor(TFT_WHITE,TFT_BLACK);
  spr.setCursor(80,0);
  spr.print(f);
  spr.pushSprite(0,0);
}

void setup()
{
  //Serial.begin(38400);
  //Serial.println();
  pinMode(LED_BUILTIN,OUTPUT);
  for (int i=0;i<5;i++)
  {
    digitalWrite(LED_BUILTIN,HIGH);
    delay(10);
    digitalWrite(LED_BUILTIN,LOW);
    delay(250);
  }

  // Initialise the TFT
  tft.init();
  tft.setRotation(1);
  tft.fillScreen(TFT_RED);

  // Optionally set colour depth to 8 or 16 bits, default is 16 if not specified
  // spr.setColorDepth(8);

  // Create a sprite of defined size
  spr.createSprite(WIDTH, HEIGHT);
  spr.fillSprite(TFT_RED);
  spr.pushSprite(0,0);
  delay(5000);
  for (int i=0;i<5;i++)
  {
    digitalWrite(LED_BUILTIN,HIGH);
    delay(10);
    digitalWrite(LED_BUILTIN,LOW);
    delay(250);
  }

}

void loop(void)
{
  static uint32_t next_update = 0;
  if (next_update<=millis())
  {
    next_update = millis()+50;
    refresh(frequency);
    frequency++;
  }
}

JimDrewGH, 2024-09-08:
What version of the Arduino core are you using?  This could be the "normal" issue of any core file version >v2.0.14 not being compatible.

ianm8, 2024-09-09:
Hi

I see now this issue has been reported before. I tried adding:

`#define RP2040_PIO_SPI`

but that did not work.

Cheers, Ian

pszecho, 2024-10-05:
Hi, 
I also have the same problem - I described it here: https://github.com/Bodmer/TFT_eSPI/discussions/3497 
Now I found this issues and I try to downgrade the library and now.... RP2040 + ST7789 work! 
My working setup:  arduino ide: 2.3.3 + tft_espi 2.5.34 (on the latest version I have black screen) 
Works on Setup 138 - I only change width from 240 to 135

ianm8, 2024-10-07:
Yes, reverting to 2.5.34 works ok. I had a look at the difference between 2.5.34 and 2.5.43 (using WinMerge) to see exactly what has changed. And not much, mostly comment updates. But there is a change to this line in TFT_eSPI_RP2040.h: 

2.5.34
`#define SET_BUS_READ_MODE  // spi_set_format(SPI_X,  8, (spi_cpol_t)0, (spi_cpha_t)0, SPI_MSB_FIRST)`

2.5.43
`#define SET_BUS_READ_MODE  spi_set_format(SPI_X,  8, (spi_cpol_t)0, (spi_cpha_t)0, SPI_MSB_FIRST)`

The spi_set_format function has been reinstated. So I updated to 2.5.42 again and commented out that function and now it's working.

Not sure why this is an issue really, it looks like it should be ok using that function. I don't know enough to investigate further. Hoping others can figure it out from here. :)

Cheers, Ian