Bodmer/TFT_eSPI #3479: SPI clock frequency switching

focussing, 2024-09-10:
Hello!

Our board consists of a 480x320 TFT / RP2040 combination.
The TFT SPI clock rate in my situation is 24MHz (via User_Setup.h) but I use an ADC running on max 1MHz so I use SPI.beginTransaction(SPISettings(1000000, MSBFIRST, SPI_MODE0)) every time I do an AD-conversion. 
That works fine but after the ADC cycle, I refresh the TFT window and see that the TFT SPI clock rate has become 8MHz.
How can I get that back to 24MHz?

I hope you can help.

Best regards,
Raymond

PJ5168, 2024-10-04:
There are many SPI or PIO available.
24Mhz is too low for 480x320, unless you use a RPI type TFT LCD.... oops...
If you have enough pins, suggest to use 125Mhz PIO SPI on TFT LCD, and use other SPI for 1Mhz ADC on a 250Mhz RP2040.

focussing, 2024-10-04:
You are completely right, however even if I use a single PIO SPI for the TFT I cannot make it faster than 24MHz whatever I try...

PJ5168, 2024-10-05:
Sir,
You may refer to issue #2432 for more info. The 125Mhz PIO SPI is working fine on 250Mhz RP2040. And it's fast!
BTW, the 250Mhz RP2040 may be to slow to process the 1Msps data from ADC and display on the 480x320 screen or even 320x240. For instance, hard to process and display the data from 90640 onto 320x240 TFT LCD. Used Pi Zero W or Pi Zero 2W for much better result.
2 cents,

focussing, 2024-10-14:
Just to be sure I am using Arduino so maybe that is different?

PJ5168, 2024-10-16:
The TFT_eSPI is working great on Arduino. The 125Mhz PIO SPI is working on 250Mhz RP2040 using Arduino.
Which RP2040 board package you used in Arduino?

focussing, 2024-10-18:
this is the RP2040 board package:

https://github.com/earlephilhower/arduino-pico