Bodmer/TFT_eSPI #3486: The T-Display code is broken

stevexyz, 2024-09-20:
I just bought the TTGO T-Display v1.1, recognized in Arduino as:

> BN: LilyGo T-Display
> VID: 1a86
> PID: 55d4
> SN: 589A009269

Installed arduino (on Linux Mint), copied this repository library TFT_eSPI to the Arduino directory and tried some video examples but none seems working. Below the output of the compilation:

```
WARNING: Category 'Sound' in library ESP_I2S is not valid. Setting to 'Uncategorized'
WARNING: Category 'Sensor' in library ESP_NOW is not valid. Setting to 'Uncategorized'
WARNING: Category 'Sound' in library ESP_SR is not valid. Setting to 'Uncategorized'
WARNING: Category '' in library ESP Insights is not valid. Setting to 'Uncategorized'
WARNING: Category '' in library ESP RainMaker is not valid. Setting to 'Uncategorized'
WARNING: Category '' in library TFLite Micro is not valid. Setting to 'Uncategorized'
WARNING: Category '' in library WiFiProv is not valid. Setting to 'Uncategorized'
In file included from /home/steve/Arduino/libraries/TFT_eSPI/TFT_eSPI.h:48,
                 from /home/steve/Arduino/libraries/TFT_eSPI/TFT_eSPI.cpp:17:
/home/steve/Arduino/libraries/TFT_eSPI/Processors/TFT_eSPI_ESP32.c: In function 'void dc_callback(spi_transaction_t*)':
/home/steve/Arduino/libraries/TFT_eSPI/Processors/TFT_eSPI_ESP32.h:115:22: error: 'GPIO' was not declared in this scope
  115 |         #define DC_D GPIO.out_w1ts = (1 << TFT_DC)//;GPIO.out_w1ts = (1 << TFT_DC)
      |                      ^~~~
/home/steve/Arduino/libraries/TFT_eSPI/Processors/TFT_eSPI_ESP32.c:652:27: note: in expansion of macro 'DC_D'
  652 |   if ((bool)spi_tx->user) DC_D;
      |                           ^~~~
/home/steve/Arduino/libraries/TFT_eSPI/Processors/TFT_eSPI_ESP32.h:114:22: error: 'GPIO' was not declared in this scope
  114 |         #define DC_C GPIO.out_w1tc = (1 << TFT_DC)//;GPIO.out_w1tc = (1 << TFT_DC)
      |                      ^~~~
/home/steve/Arduino/libraries/TFT_eSPI/Processors/TFT_eSPI_ESP32.c:653:8: note: in expansion of macro 'DC_C'
  653 |   else DC_C;
      |        ^~~~
/home/steve/Arduino/libraries/TFT_eSPI/TFT_eSPI.cpp: In member function 'void TFT_eSPI::begin_tft_write()':
/home/steve/Arduino/libraries/TFT_eSPI/Processors/TFT_eSPI_ESP32.h:159:22: error: 'GPIO' was not declared in this scope
  159 |         #define CS_L GPIO.out_w1tc = (1 << TFT_CS); GPIO.out_w1tc = (1 << TFT_CS)
      |                      ^~~~
/home/steve/Arduino/libraries/TFT_eSPI/TFT_eSPI.cpp:39:5: note: in expansion of macro 'CS_L'
   39 |     CS_L;
      |     ^~~~
/home/steve/Arduino/libraries/TFT_eSPI/TFT_eSPI.cpp: In member function 'void TFT_eSPI::end_tft_write()':
/home/steve/Arduino/libraries/TFT_eSPI/Processors/TFT_eSPI_ESP32.h:160:22: error: 'GPIO' was not declared in this scope
  160 |         #define CS_H GPIO.out_w1ts = (1 << TFT_CS)//;GPIO.out_w1ts = (1 << TFT_CS)
      |                      ^~~~
/home/steve/Arduino/libraries/TFT_eSPI/TFT_eSPI.cpp:56:7: note: in expansion of macro 'CS_H'
   56 |       CS_H;
      |       ^~~~
/home/steve/Arduino/libraries/TFT_eSPI/TFT_eSPI.cpp: In member function 'void TFT_eSPI::begin_tft_read()':
/home/steve/Arduino/libraries/TFT_eSPI/Processors/TFT_eSPI_ESP32.h:159:22: error: 'GPIO' was not declared in this scope
  159 |         #define CS_L GPIO.out_w1tc = (1 << TFT_CS); GPIO.out_w1tc = (1 << TFT_CS)
      |                      ^~~~
/home/steve/Arduino/libraries/TFT_eSPI/TFT_eSPI.cpp:77:5: note: in expansion of macro 'CS_L'
   77 |     CS_L;
      |     ^~~~
/home/steve/Arduino/libraries/TFT_eSPI/TFT_eSPI.cpp: In member function 'void TFT_eSPI::end_tft_read()':
/home/steve/Arduino/libraries/TFT_eSPI/Processors/TFT_eSPI_ESP32.h:160:22: error: 'GPIO' was not declared in this scope
  160 |         #define CS_H GPIO.out_w1ts = (1 << TFT_CS)//;GPIO.out_w1ts = (1 << TFT_CS)
      |                      ^~~~
/home/steve/Arduino/libraries/TFT_eSPI/TFT_eSPI.cpp:97:7: note: in expansion of macro 'CS_H'
   97 |       CS_H;
      |       ^~~~
/home/steve/Arduino/libraries/TFT_eSPI/TFT_eSPI.cpp: In member function 'void TFT_eSPI::writecommand(uint8_t)':
/home/steve/Arduino/libraries/TFT_eSPI/Processors/TFT_eSPI_ESP32.h:114:22: error: 'GPIO' was not declared in this scope
  114 |         #define DC_C GPIO.out_w1tc = (1 << TFT_DC)//;GPIO.out_w1tc = (1 << TFT_DC)
      |                      ^~~~
/home/steve/Arduino/libraries/TFT_eSPI/TFT_eSPI.cpp:548:3: note: in expansion of macro 'DC_C'
  548 |   DC_C;
      |   ^~~~
/home/steve/Arduino/libraries/TFT_eSPI/TFT_eSPI.cpp: In member function 'void TFT_eSPI::writedata(uint8_t)':
/home/steve/Arduino/libraries/TFT_eSPI/Processors/TFT_eSPI_ESP32.h:115:22: error: 'GPIO' was not declared in this scope
  115 |         #define DC_D GPIO.out_w1ts = (1 << TFT_DC)//;GPIO.out_w1ts = (1 << TFT_DC)
[... 112 lines cut ...]
Hash of data verified.
Compressed 311904 bytes to 171895...
Writing at 0x00010000... (9 %)
Writing at 0x0001a6d4... (18 %)
Writing at 0x00028c45... (27 %)
Writing at 0x0002e1d2... (36 %)
Writing at 0x00033878... (45 %)
Writing at 0x000391e0... (54 %)
Writing at 0x0003e49f... (63 %)
Writing at 0x000438c5... (72 %)
Writing at 0x00048e74... (81 %)
Writing at 0x00051a9d... (90 %)
Writing at 0x0005952a... (100 %)
Wrote 311904 bytes (171895 compressed) at 0x00010000 in 3.2 seconds (effective 773.2 kbit/s)...
Hash of data verified.

Leaving...
Hard resetting via RTS pin...
```

BUT the only board effect is to light up the display!

Did I do something wrong? In case, can you elaborate in the homepage instructions?

Thank you

praegustator, 2024-09-25:
I think I had/have the same issue as you; in the end though I was able to upload and partially see the rendered images (so far, analog clock and PNG). Mine is probably a chinese clone of [LilyGo](https://www.lilygo.cc/en-pl/products/lilygo%C2%AE-ttgo-t-display-1-14-inch-lcd-esp32-control-board) (since there is no label on it). It is though detected as LilyGo T-Display.

[image]

I can only advice to double-check these files:

.../Arduino/libraries/TFT_eSPI/**User_Setup_Select.h**

```c++
#include <User_Setups/Setup25_TTGO_T_Display.h>
```

.../Arduino/libraries/TFT_eSPI/**User_Setup.h**

```c++
#define USER_SETUP_INFO "User_Setup"

// ...

#define ILI9341_DRIVER 

// ...

#define TFT_MISO  PIN_D6
#define TFT_MOSI  PIN_D7
#define TFT_SCLK  PIN_D5 

#define TFT_CS    PIN_D8
#define TFT_DC    PIN_D3
#define TFT_RST   PIN_D4

// ...

```

These settings I used in Arduino IDE:

```
Board: LilyGo T-Display
JTAG Adapter: Disbled
PSRAM: Disbled
Partition Scheme: Default 4MB with spiffs (1.2MB APP/1.5MB SPIFFS)
CPU Frequency: 240MHz
Flash Mode: QIO
Flash Frequency: 80 MHz
Flash Size: 4MByte (32Mbit)
Upload Speed: 921600
Arduino Runs On: Core 1
Events Runs On: Core 1
Core Debug Level: None
Erase All Flash: Disabled
```

Note: script it validated and uploaded, and I got an image on the screen; however, those scripts from library examples were cropped/ panned/ zoomed/ showed only particularly. I might not be uploading correctly, in the end. I also tried [this](https://github.com/Xinyuan-LilyGO/TTGO-T-Display) fork (or is it?) with no luck.

Will be interested in others' experience on this topic.

stevexyz, 2024-09-26:
I checked and they are ok, but I just realized now that this issue should have been opened on the T-Display specific repository, so closing it here and open it there!