Bodmer/TFT_eSPI #3493: No output on the Cheap Yellow Display

efsane14010, 2024-09-28:
Only raise issues for problems with the library and/or provided examples. Post questions, comments and useful tips etc in the "Discussions" section.

To minimise effort to resolve issues the following should be provided as a minimum:
1. İ upload the code and it just doesnt work
2. Arduino ide
3. Latest version
4. Board package version (e.g. 2.0.3) available from the Boards Manager... menu(idk)
5. Esp32 Cheap Yellow Display. Im using the Esp32 Devkit
6. TFT driver (e.g. ILI9341), a link to the vendors product web page is useful too. (idk)

Plus further information as appropriate to the problem:
1. No need to screen shot since it just doesnt put any output also no errors

I used this code: (Im just trying to learn how to actually use this screen and not use the webflashers)
https://github.com/Bodmer/TFT_eSPI/blob/master/examples/320%20x%20240/TFT_Clock/TFT_Clock.ino

elgerg, 2024-09-28:
Is the device crashing/rebooting?

Could this be related?
https://github.com/Bodmer/TFT_eSPI/issues/3329

florinbaciuu, 2024-10-01:
Can u provide us your UserSetup.h file? 

I have an Cyd 2432S028R and i tested [link](https://github.com/Bodmer/TFT_eSPI/blob/master/examples/320%20x%20240/TFT_Clock/TFT_Clock.ino) and I don't find any problem with the library. 
[image]

I will leave you my **UserSetup.h** here , maybe you can figure out what the problem would be.
```
// #define ILI9341_DRIVER       // Generic driver for common displays
#define ILI9341_2_DRIVER // Alternative ILI9341 driver, see https://github.com/Bodmer/TFT_eSPI/issues/1172
// #define ILI9342_DRIVER

// #define TFT_SDA_READ      // This option is for ESP32 ONLY, tested with ST7789 and GC9A01 display only

//  #define TFT_RGB_ORDER TFT_RGB  // Colour order Red-Green-Blue
//  #define TFT_RGB_ORDER TFT_BGR  // Colour order Blue-Green-Red

#define TFT_WIDTH 320
#define TFT_HEIGHT 240

// #define TFT_INVERSION_ON
// #define TFT_INVERSION_OFF

#define TFT_BL 21             // LED back-light control pin
#define TFT_BACKLIGHT_ON HIGH // Level to turn ON back-light (HIGH or LOW)

#define TFT_MOSI 13 // In some display driver board, it might be written as "SDA" and so on.
#define TFT_MISO 12
#define TFT_SCLK 14
#define TFT_CS 15  // Chip select control pin
#define TFT_DC 2   // Data Command control pin
#define TFT_RST -1 // Reset pin (could connect to Arduino RESET pin)
#define TFT_BL 21  // LED back-light

// #define TOUCH_CS 33     // Chip select pin (T_CS) of touch screen

#define LOAD_GLCD  // Font 1. Original Adafruit 8 pixel font needs ~1820 bytes in FLASH
#define LOAD_FONT2 // Font 2. Small 16 pixel high font, needs ~3534 bytes in FLASH, 96 characters
#define LOAD_FONT4 // Font 4. Medium 26 pixel high font, needs ~5848 bytes in FLASH, 96 characters
#define LOAD_FONT6 // Font 6. Large 48 pixel font, needs ~2666 bytes in FLASH, only characters 1234567890:-.apm
#define LOAD_FONT7 // Font 7. 7 segment 48 pixel font, needs ~2438 bytes in FLASH, only characters 1234567890:-.
#define LOAD_FONT8 // Font 8. Large 75 pixel font needs ~3256 bytes in FLASH, only characters 1234567890:-.
#define LOAD_FONT8N // Font 8. Alternative to Font 8 above, slightly narrower, so 3 digits fit a 160 pixel TFT
#define LOAD_GFXFF // FreeFonts. Include access to the 48 Adafruit_GFX free fonts FF1 to FF48 and custom fonts
#define SMOOTH_FONT

#define SPI_FREQUENCY 55000000
//#define SPI_FREQUENCY 40000000
// #define SPI_FREQUENCY  65000000
//#define SPI_FREQUENCY  80000000
#define SPI_READ_FREQUENCY 20000000
#define SPI_TOUCH_FREQUENCY 2500000
#define USE_HSPI_PORT
#define SUPPORT_TRANSACTIONS
```
I will also attach **platformio.ini**

```
; PlatformIO Project Configuration File for Sunton_2432S028_ESP32
;===========================================================================;
;! Don't change
[platformio]
workspace_dir = ./
core_dir = ${PROJECT_DIR}/core
platforms_dir = ${platformio.core_dir}/platforms
packages_dir = ${platformio.core_dir}/pkg
##-------------------------
src_dir = ${PROJECT_DIR}/src
libdeps_dir = ${PROJECT_DIR}/libdeps
lib_dir = ${PROJECT_DIR}/lib
data_dir = ${PROJECT_DIR}/data
[... 192 lines cut ...]
  "connectivity": [
    "wifi",
    "bluetooth",
    "ethernet",
    "can"
  ],
  "debug": {
    "openocd_board": "esp-wroom-32.cfg"
  },
  "frameworks": [
    "arduino",
    "espidf"
  ],
  "name": "esp32-2432S028R",
  "upload": {
    "flash_size": "4MB",
    "maximum_ram_size": 327680,
    "maximum_size": 4194304,
    "require_upload_port": true,
    "speed": 460800
  },
  "url": "https://www.aliexpress.com/item/1005004502250619.html",
  "vendor": "Sunton"
}
```