Bodmer/TFT_eSPI #3510: White screen issue, can't seem to find solution

Cptnyar77, 2024-10-15:
Hi all,
I am having a white screen issue.
I first tried the Rainbow screen example - no avail.
Then the Read-Write diagnostic
can't seem to figure it out.

IDE: Arduino IDE
Library Vrsn: 2.5.43
Board Manager Vrsn:2.0.17
TFT: R61581
ESP32: DOIT Devkit V1
Using 8-bit parralel

Below is my changes to the User_setup.h Followed by the serial monitor output after code was uploaded

#define R61581_DRIVER

// Tell the library to use 8-bit parallel mode (otherwise SPI is assumed)
#define TFT_PARALLEL_8_BIT

// The ESP32 and TFT the pins used for testing are:
#define TFT_CS   33  // Chip select control pin (library pulls permanently low
#define TFT_DC   15  // Data Command control pin - must use a pin in the range 0-31
#define TFT_RST  32  // Reset pin, toggles on startup

#define TFT_WR    4  // Write strobe control pin - must use a pin in the range 0-31
#define TFT_RD    2  // Read strobe control pin

#define TFT_D0   12  // Must use pins in the range 0-31 for the data bus
#define TFT_D1   13  // so a single register write sets/clears all bits.
#define TFT_D2   26  // Pins can be randomly assigned, this does not affect
#define TFT_D3   25  // TFT screen update performance.
#define TFT_D4   19
#define TFT_D5   21
#define TFT_D6   27
#define TFT_D7   14

==========================================================================

 Pixel value written = 80
 Pixel value read    = CFD3
 ERROR                 ^^^^
 Pixel value written = 100
 Pixel value read    = CFD3
 ERROR                 ^^^^
 Pixel value written = 200
 Pixel value read    = CFD3
 ERROR                 ^^^^
 Pixel value written = 400
 Pixel value read    = CFD3
 ERROR                 ^^^^
 Pixel value written = 800
 Pixel value read    = CFD3

Basitadam, 2024-10-16:
Can you run "Read_User_Setup" and send the serial output?

Cptnyar77, 2024-10-16:
[code]
TFT_eSPI ver = 2.5.43
Processor    = ESP32
Frequency    = 240MHz
Transactions = Yes
Interface    = Parallel
Display driver = 6158
Display width  = 320
Display height = 480

MOSI    = GPIO 13
MISO    = GPIO 12
SCK     = GPIO 14
TFT_CS   = GPIO 33
TFT_DC   = GPIO 15
TFT_WR   = GPIO 4
TFT_RD   = GPIO 2

TFT_D0   = GPIO 12
TFT_D1   = GPIO 13
TFT_D2   = GPIO 26
TFT_D3   = GPIO 25
TFT_D4   = GPIO 19
TFT_D5   = GPIO 21
TFT_D6   = GPIO 27
TFT_D7   = GPIO 14

Font GLCD   loaded
Font 2      loaded
Font 4      loaded
Font 6      loaded
Font 7      loaded
Font 8      loaded
Smooth font enabled

Do I have to somehow disable the SPI stuff?

Basitadam, 2024-10-16:
You shouldn't need to but you can be sure just defining SPI pins as -1. Like #define TFT_SCLK -1

Cptnyar77, 2024-10-16:
ReadWrite test still returns all errors.
However now it is returning '0'
instead of 'CFD3'

==========================================
ERROR                 ^^^^
 Pixel value written = 4
 Pixel value read    = 0
 ERROR                 ^^^^
 Pixel value written = 8
 Pixel value read    = 0
 ERROR                 ^^^^
 Pixel value written = 10
 Pixel value read    = 0
 ERROR                 ^^^^
=============================================

Basitadam, 2024-10-16:
I can't debug further because my ESP32 Board version is 3.0.5. If it won't make any problem, I suggest you upgrade from 2.0.17 to it

Cptnyar77, 2024-10-16:
When I ugrade to 3.0.5 i get a compilation error. After searching around a bit, somebody suggested rolling it back.

Basitadam, 2024-10-17:
That is unfortunate. Can happen when switching form 2.x to 3.x.
is your setup on a breadboard? Removing the display first, connecting D0-D7 lines with pull-up or pull-down resistors might give you a clue. You can debug if your software reads back what you set with the resistors. 
Do not connect D0-D7 lines directly to Vcc or GND. They are set to output while sending commands

ealvarezd, 2025-05-06:
Greetings.
I have the same problem described in this post.
I've configured the files for a display that isn't defined in the libraries, and it works perfectly in many of the examples, but when I try to read a pixel, I get the same error as Cptnyar77.
I've connected a logic analyzer, and the problem is that the TFT_RD pin is always at 1. I've run "Read_User_Setup" and it's defined, but it never changes value.

[code]
TFT_eSPI ver = 2.5.43
Processor    = RP2040
Transactions = Yes
Interface    = Parallel
Display driver = 8310
Display width  = 240
Display height = 320

TFT_CS   = GPIO 4
TFT_DC   = GPIO 1
TFT_RST  = GPIO 0
TFT_WR   = GPIO 2
TFT_RD   = GPIO 3

TFT_D0   = GPIO 10
TFT_D1   = GPIO 11
TFT_D2   = GPIO 12
TFT_D3   = GPIO 13
TFT_D4   = GPIO 14
TFT_D5   = GPIO 15
TFT_D6   = GPIO 16
TFT_D7   = GPIO 17

Font GLCD   loaded
Font 2      loaded
Font 4      loaded
Font 6      loaded
Font 7      loaded
Font 8      loaded
Smooth font enabled

The pixel is drawn correctly but always reads the value 0

 Pixel value written = 40
 Pixel value read    = 0
 ERROR                 ^^^^
 Pixel value written = 80
 Pixel value read    = 0
 ERROR                 ^^^^
 Pixel value written = 100
 Pixel value read    = 0

Any ideas?