Bodmer/TFT_eSPI #3537: Whether the TFT_eSPI library supports rp2350

Carlos-TGW, 2024-11-05:
IDE：Arduino IDE
chip：RP2350
Configuration：
[image]

Using the screen：3.5 LCD
Screen driver：ILI9488
The problem：
When rp2040 uses TFT_eSPI library to drive ILI9488 screen, the use is normal. However, when rp2350 uses TFT_eSPI library to drive ILI9488 screen, the screen is abnormal. It compiles, but it doesn't display properly.

Here's the code:
#include <Arduino.h>
#include <TFT_eSPI.h>

#define  LCD_BL 18

TFT_eSPI tft = TFT_eSPI(); /* TFT instance screenWidth, screenHeight */

void setup() {
  // put your setup code here, to run once:
    Serial.begin( 115200 );
    
    tft.begin();          /* TFT init */
    tft.setRotation( 1); /* Landscape orientation, flipped */
    tft.invertDisplay(false);

}
uint16_t flag=0;
void loop() {
  // put your main code here, to run repeatedly:
  delay(4000);   
  tft.fillScreen(TFT_RED);
  delay(1000);
  tft.fillScreen(TFT_GREEN);
  delay(1000);
  tft.fillScreen(TFT_BLUE); 
  delay(1000);
  tft.fillScreen(TFT_WHITE);
  delay(1000);
  tft.fillScreen(TFT_BLACK);
  delay(1000);
  tft.fillScreen(TFT_DARKGREY); 
  delay(1000);

}

I want to ask: does TFT_eSPI library support rp2350 to drive ILI9488?

Carlos-TGW, 2024-11-05:
I used 4-wire SPI to link the ILI9488 screen

webzep, 2024-11-27:
@Carlos-TGW this worked for you with RP2350?

nstepanets, 2024-12-05:
I checked RP2350 + NV3030B display, everything works the same as on RP2040 (ARDUINO PICO VERSION 4.3.1)