Bodmer/TFT_eSPI #3544: 1.28inch_ESP32-2424S012 boot loop

chung1608, 2024-11-10:
LCD works fine when I use ESP32 library ver2.0.2 but when I update ESP32 library to 3.0.5, ESP32 falls into boot loop

17:40:08.708 -> rst:0x3 (RTC_SW_SYS_RST),boot:0xc (SPI_FAST_FLASH_BOOT)
17:40:08.708 -> Saved PC:0x40048b82
17:40:08.708 -> SPIWP:0xee
17:40:08.708 -> mode:DIO, clock div:1
17:40:08.708 -> load:0x3fcd5820,len:0x458
17:40:08.708 -> load:0x403cc710,len:0x814
17:40:08.753 -> load:0x403ce710,len:0x2880
17:40:08.753 -> entry 0x403cc710

JVRMTS, 2024-11-11:
I have the same problem, with version 3.0.1 of esp32 on arduino it works correctly, with the latest version it goes into a reset loop.

I use a 3.5" SPI TFT the user_setup file is configured like this:

#define TFT_MISO 19
#define TFT_MOSI 23
#define TFT_SCLK 18
#define TFT_CS   15  // Chip select control pin
#define TFT_DC    2  // Data Command control pin
#define TFT_RST   4  // Reset pin (could connect to RST pin)
//#define TFT_RST  -1  // Set TFT_RST to -1 if display RESET is connected to ESP32 board RST

// For ESP32 Dev board (only tested with GC9A01 display)
// The hardware SPI can be mapped to any pins

//#define TFT_MOSI 15 // In some display driver board, it might be written as "SDA" and so on.
//#define TFT_SCLK 14
//#define TFT_CS   5  // Chip select control pin
//#define TFT_DC   27  // Data Command control pin
//#define TFT_RST  33  // Reset pin (could connect to Arduino RESET pin)
//#define TFT_BL   22  // LED back-light

#define TOUCH_CS 21     // Chip select pin (T_CS) of touch screen

The serial monitor output is:

ets Jun  8 2016 00:22:57

rst:0x8 (TG1WDT_SYS_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:1
load:0x3fff0030,len:4832
load:0x40078000,len:16460
load:0x40080400,len:4
load:0x40080404,len:3504
entry 0x400805cc
ets Jun  8 2016 00:22:57

chung1608, 2024-11-11:
> Can you send your setup? Pin defines file
this is HW fix pin I can't change 
#define TFT_MISO -1
#define TFT_MOSI 7
#define TFT_SCLK 6
#define TFT_CS 10 
#define TFT_DC 2 
#define TFT_RST -1 // Reset pin (could connect to Arduino RESET pin)
#define TFT_BL 3

JVRMTS, 2024-11-11:
I have defined it in user_setup and it is still the same
[image]

Basitadam, 2024-11-12:
> I have defined it in user_setup and it is still the same ![2024-11-11](https://private-user-images.githubusercontent.com/74095359/385041227-3b1c6037-9a48-4d65-90e9-6ef582160a1a.png?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3MzE0MDI2MjcsIm5iZiI6MTczMTQwMjMyNywicGF0aCI6Ii83NDA5NTM1OS8zODUwNDEyMjctM2IxYzYwMzctOWE0OC00ZDY1LTkwZTktNmVmNTgyMTYwYTFhLnBuZz9YLUFtei1BbGdvcml0aG09QVdTNC1ITUFDLVNIQTI1NiZYLUFtei1DcmVkZW50aWFsPUFLSUFWQ09EWUxTQTUzUFFLNFpBJTJGMjAyNDExMTIlMkZ1cy1lYXN0LTElMkZzMyUyRmF3czRfcmVxdWVzdCZYLUFtei1EYXRlPTIwMjQxMTEyVDA5MDUyN1omWC1BbXotRXhwaXJlcz0zMDAmWC1BbXotU2lnbmF0dXJlPWIxYWE0NzQ2ODJlODA1ODU2OTliMzAyZGViMGJmNTFjN2NlYzdlYmYyZWU0ODc2YjliYTA5MWJiOTJkYWYyMGUmWC1BbXotU2lnbmVkSGVhZGVycz1ob3N0In0.7v8EFqnhDkuBuDftLge9CNGtA8nPzAkD_aZFj201nu0)

Sorry. Misread your MCU model. It works on ESP32-S3.
Are you sure the reset is because of TFT_eSPI? 
if you put
```
while(1);
```
right after   ```tft.init();```  line still reboots?

JacketJack113, 2024-11-13:
#3464

chung1608, 2024-11-13:
reboot right after tft.init();
[image]

Basitadam, 2024-11-13:
> reboot right after tft.init(); [...]

Now I see. You have ESP32-C3 board. Can you test with this "TFT_eSPI_ESP32_C3.h" file from this link:

[https://www.filemail.com/d/bsgocapqwmdpeyj](url) 

I had the same issue. Modified the .h file

BR,
Tim

chung1608, 2024-11-13:
another isssue, do not exit init_tft
[image]

Basitadam, 2024-11-13:
> another isssue, do not exit init_tft [...]

Strange. Here is my setup, tested again now, works OK:

Win10
Arduino 2.3.3
TFT_eSPI 2.5.43
ESP32 3.0.5

user.select.h:
```
// Setup for the ESP32 C3 with ILI9341 display
// Note SPI DMA with ESP32 C3 is not currently supported

#define USER_SETUP_ID 70
// See SetupX_Template.h for all options available

#define ILI9341_DRIVER
//#define ST7796_DRIVER
//#define ILI9488_DRIVER

// Adafruit qtpy default
//TFT_CS 6
//TFT_MOSI 7
//TFT_MISO 8
//TFT_SCLK 10

// Lolin C3 mini default
//TFT_CS 5
//TFT_MOSI 4
//TFT_MISO 3
//TFT_SCLK 2

//ESP32 C3 generic default
//TFT_CS 7
//TFT_MOSI 6
//TFT_MISO 5
//TFT_SCLK 4

#define TFT_CS   7

#define TFT_MOSI 6
#define TFT_MISO 5
#define TFT_SCLK 4

#define TFT_DC    10
#define TFT_RST   9

//#define TOUCH_CS  1 // Optional for touch screen

#define LOAD_GLCD
#define LOAD_FONT2
#define LOAD_FONT4
#define LOAD_FONT6
#define LOAD_FONT7
#define LOAD_FONT8
#define LOAD_GFXFF

#define SMOOTH_FONT

//#define SPI_FREQUENCY  27000000
#define SPI_FREQUENCY  40000000   // Maximum for ILI9341

#define SPI_READ_FREQUENCY  6000000 // 6 MHz is the maximum SPI read speed

#define SPI_TOUCH_FREQUENCY 2500000
```

My board:

[image]

Adruino board setup:

[image]

And output:
[image]

SPI pins are different but you said your setup was working before

Basitadam, 2024-11-13:
Works well on ESP32 3.0.7, too

chung1608, 2024-11-14:
no luck with me, I'm use this LCD 
[image]
[image]

Basitadam, 2024-11-14:
Can you send your user.select.h file. I want to test with it

chung1608, 2024-11-14:
> Can you send your user.select.h file. I want to test with it

https://www.filemail.com/d/rdbthmsgtkbybzs

Basitadam, 2024-11-14:
> > Can you send your user.select.h file. I want to test with it
> 
> https://www.filemail.com/d/rdbthmsgtkbybzs

Thank you.

https://github.com/user-attachments/assets/3c65affe-95f4-4ea1-b3b7-4d4603b3c607

OK. It stuck after tft.init like yours. Something with the User_Setup.h file.
I created a new one. I am sending it with the link below. There is a Arduino test sketch and the new setup file

https://www.filemail.com/d/eaavnnfigzjkjfg

Save the setup file (Setup70tk_ESP32_C3_GC9A01.h) to your "....\Arduino\libraries\TFT_eSPI\User_Setups" folder,
Add this line 
```
#include <User_Setups/Setup70tk_ESP32_C3_GC9A01.h>    // Setup file for ESP32 C3 with SPI GC9A01
```
to your  User_Setup_Select.h  file,
Comment the line #include <User_Setup.h>  in the  User_Setup_Select.h file.
Upload the Arduino sketch "TFT_eSPI_SimpleTest.ino" and enjoy

Hope this will solve your issue.

chung1608, 2024-11-14:
working, thank you ^^

Basitadam, 2024-11-14:
> working, thank you ^^

HappyFace();

SideshowAlex, 2024-11-24:
@Basitadam The content of the link has expired. Can you repost your header file inline or in a gist?

Basitadam, 2024-11-24:
> @Basitadam The content of the link has expired. Can you repost your header file inline or in a gist?

Sure:
```
// Setup for the ESP32 C3 with GC90A1 display
// 2024.11.14 by Timucin KANATLI

#define USER_SETUP_ID 70
#define GC9A01_DRIVER

#define TFT_CS   10
#define TFT_MOSI 7
#define TFT_MISO 5
#define TFT_SCLK 6

#define TFT_DC    2
#define TFT_RST   -1

#define TFT_BL   3             // LED back-light control pin
#define TFT_BACKLIGHT_ON HIGH  // Level to turn ON back-light (HIGH or LOW)

#define LOAD_GLCD
#define LOAD_FONT2
#define LOAD_FONT4
#define LOAD_FONT6
#define LOAD_FONT7
#define LOAD_FONT8
#define LOAD_GFXFF

#define SMOOTH_FONT
#define SPI_FREQUENCY  40000000   
#define SPI_READ_FREQUENCY  6000000 

```

kelenMr, 2025-03-16:
ELF file SHA256: 3773ba351e5cdcb3

Rebooting...
ESP-ROM:esp32c3-api1-20210207
Build:Feb  7 2021
rst:0x3 (RTC_SW_SYS_RST),boot:0xc (SPI_FAST_FLASH_BOOT)
Saved PC:0x40381ef2
SPIWP:0xee
mode:DIO, clock div:1
load:0x3fcd5810,len:0x438
load:0x403cc710,len:0x90c
load:0x403ce710,len:0x2624
entry 0x403cc710
Guru Meditation Error: Core  0 panic'ed (Store access fault). Exception was unhandled.

Core  0 register dump:
MEPC    : 0x42001d74  RA      : 0x42001d64  SP      : 0x3fc95ad0  GP      : 0x3fc8c000  
TP      : 0x3fc861d4  T0      : 0x00000000  T1      : 0xffc3ffff  T2      : 0x00000001  
S0/FP   : 0x60004000  S1      : 0x3fc8d590  A0      : 0x00000001  A1      : 0xffffffff  
A2      : 0x00000000  A3      : 0xfdffffff  A4      : 0x08000000  A5      : 0x00000010  
A6      : 0x00000001  A7      : 0x04c4b400  S2      : 0x00000001  S3      : 0x00000000  
S4      : 0x00000000  S5      : 0x00000000  S6      : 0x00000000  S7      : 0x00000000  
S8      : 0x00000000  S9      : 0x00000000  S10     : 0x00000000  S11     : 0x00000000  
T3      : 0x00000000  T4      : 0x0000000f  T5      : 0x003c0000  T6      : 0x00000003  
MSTATUS : 0x00001881  MTVEC   : 0x40380001  MCAUSE  : 0x00000007  MTVAL   : 0x00000010  
MHARTID : 0x00000000  

Stack memory:
3fc95ad0: 0x00000000 0x3fc8f000 0x01312d00 0x00000001 0x00000001 0x3fc8f000 0x3fc8d590 0x4200208e
3fc95af0: 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x3fc8f000 0x42003000 0x42000034
3fc95b10: 0x00000000 0x00000000 0x42003000 0x42006374 0x00000000 0x00000000 0x00000000 0x4038795e
3fc95b30: 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0xa5a5a5a5 0xa5a5a5a5 0xa5a5a5a5
3fc95b50: 0xa5a5a5a5 0xa5a5a5a5 0xa5a5a5a5 0xa5a5a5a5 0xbaad5678 0x00000160 0xabba1234 0x00000154
3fc95b70: 0x3fc95ab0 0x1cbfba2e 0x3fc8e32c 0x3fc8e32c 0x3fc95b70 0x3fc8e324 0x00000018 0xa746a9f2
3fc95b90: 0xf8458e99 0x3fc95b70 0x00000000 0x00000001 0x3fc93b60 0x706f6f6c 0x6b736154 0x21550e00
3fc95bb0: 0x00b23eee 0x00000000 0x3fc95b50 0x00000001 0x00000002 0x00000000 0x00000000 0x00000000
3fc95bd0: 0x3fc8fa04 0x3fc8fa6c 0x3fc8fad4 0x00000000 0x00000000 0x00000001 0x00000000 0x00000000
3fc95bf0: 0x00000000 0x42020994 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000
3fc95c10: 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000
3fc95c30: 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000
3fc95c50: 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000
3fc95c70: 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000
3fc95c90: 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000
3fc95cb0: 0x00000000 0x00000000 0x00000000 0x00000000 0x3e000000 0xbaad5678 0x00000060 0xabba1234
3fc95cd0: 0x00000054 0x00000000 0x3fc95cd4 0x3fc95b70 0x00000000 0x00000000 0x3fc95cec 0xffffffff
3fc95cf0: 0x3fc95cec 0x3fc95cec 0x00000000 0x3fc95d00 0xffffffff 0x3fc95d00 0x3fc95d00 0x00000000
3fc95d10: 0x00000001 0x00000000 0x9600ffff 0x00000000 0xb33fffff 0x00000000 0xbaad5678 0x00000060
3fc95d30: 0xabba1234 0x00000054 0x00000000 0x3fc95d38 0x00000000 0x00000000 0x00000000 0x3fc95d50
3fc95d50: 0xffffffff 0x3fc95d50 0x3fc95d50 0x00000000 0x3fc95d64 0xffffffff 0x3fc95d64 0x3fc95d64
3fc95d70: 0x00000001 0x00000001 0x00000000 0xfd00ffff 0x00000000 0xb33fffff 0x00000000 0xbaad5678
3fc95d90: 0x0000001c 0xabba1234 0x00000010 0x00000000 0x00000000 0x3fc8bd10 0x4201eec2 0xbaad5678
3fc95db0: 0x00046959 0x3fc8f014 0x3fc8f014 0xdbe7d144 0x1cfb2184 0x72bd6dc9 0x551e3d66 0x273f2b46
3fc95dd0: 0xaf05cf5c 0xa65255d5 0xdeb053f6 0xf56f5838 0x3e737ce4 0xc98cc7db 0x82dc888e 0x1898d463
3fc95df0: 0xfa07bab2 0x44a0a71b 0xa31d9345 0x84d35f44 0x87f1a017 0x49f0b3b1 0xfc3d684d 0x22f07aff
3fc95e10: 0x87a6b22b 0xa3d45eca 0xbf7f49bf 0x0756bd4c 0x7a40a5d7 0x9eac16d9 0x61e59179 0x14ccb79d
3fc95e30: 0xb798d0cd 0x0ec06500 0xcba6ba66 0x0eae7c52 0x1343df5b 0xc38f123b 0xf0598e19 0x5c31e043
3fc95e50: 0x6614f424 0xe173b326 0x967d5c49 0x6889531d 0x8a8f352b 0xfae1214e 0xbd6c2d0f 0x2044c365
3fc95e70: 0x58ebc933 0x1fb474c9 0xbb40cb3c 0x11f38de1 0xa3631d0f 0xa729f90d 0xbe717ad8 0xd88b2a71
3fc95e90: 0xc54b2fd9 0x8e569f29 0x32318ec7 0x6bf865e3 0xe0d4c378 0xf50d6a3c 0xb5b5aa6a 0xe752540d
3fc95eb0: 0x8f7ed3df 0xf9bd0be6 0xb25e80ba 0x4f78b30e 0xbf934be4 0xe84abac2 0x47902f7a 0x80ddbf59

ELF file SHA256: 3773ba351e5cdcb3

Rebooting...
ESP-ROM:esp32c3-api1-20210207帮我看下这个报错什么什么原因

Basitadam, 2025-03-16:
@kelenMr are you using the modified "TFT_eSPI_ESP32_C3.h" file?

kelenMr, 2025-03-16:
启用了现在报错这个p32c3-api1-20210207
Build:Feb  7 2021
rst:0x8 (TG1WDT_SYS_RST),boot:0xd (SPI_FAST_FLASH_BOOT)
Saved PC:0x40380082
SPIWP:0xee
mode:DIO, clock div:1
load:0x3fcd5810,len:0x438
load:0x403cc710,len:0x90c
load:0x403ce710,len:0x2624
entry 0x403cc710
ESP-ROM:esp32c3-api1-20210207
Build:Feb  7 2021
rst:0x8 (TG1WDT_SYS_RST),boot:0xd (SPI_FAST_FLASH_BOOT)
Saved PC:0x40380082
SPIWP:0xee
mode:DIO, clock div:1
load:0x3fcd5810,len:0x438
load:0x403cc710,len:0x90c
load:0x403ce710,len:0x2624
entry 0x403cc710
ESP-ROM:esp32c3-api1-20210207
Build:Feb  7 2021
rst:0x8 (TG1WDT_SYS_RST),boot:0xd (SPI_FAST_FLASH_BOOT)
Saved PC:0x40380082
SPIWP:0xee
mode:DIO, clock div:1
load:0x3fcd5810,len:0x438
load:0x403cc710,len:0x90c
load:0x403ce710,len:0x2624
entry 0x403cc710
ESP-ROM:esp32c3-api1-20210207
Build:Feb  7 2021
rst:0x8 (TG1WDT_SYS_RST),boot:0xd (SPI_FAST_FLASH_BOOT)
Saved PC:0x40380082
SPIWP:0xee
mode:DIO, clock div:1
load:0x3fcd5810,len:0x438
load:0x403cc710,len:0x90c
load:0x403ce710,len:0x2624
entry 0x403cc710
ESP-ROM:esp32c3-api1-20210207
Build:Feb  7 2021
rst:0x8 (TG1WDT_SYS_RST),boot:0xd (SPI_FAST_FLASH_BOOT)
Saved PC:0x40380082
SPIWP:0xee
mode:DIO, clock div:1
load:0x3fcd5810,len:0x438
load:0x403cc710,len:0x90c
load:0x403ce710,len:0x2624
entry 0x403cc710
ESP-ROM:esp32c3-api1-20210207
Build:Feb  7 2021
rst:0x8 (TG1WDT_SYS_RST),boot:0xd (SPI_FAST_FLASH_BOOT)
Saved PC:0x40380080
SPIWP:0xee
mode:DIO, clock div:1
load:0x3fcd5810,len:0x438
load:0x403cc710,len:0x90c
load:0x403ce710,len:0x2624
entry 0x403cc710
ESP-ROM:esp32c3-api1-20210207
Build:Feb  7 2021
rst:0x8 (TG1WDT_SYS_RST),boot:0xd (SPI_FAST_FLASH_BOOT)
Saved PC:0x40380082
SPIWP:0xee
mode:DIO, clock div:1
load:0x3fcd5810,len:0x438
load:0x403cc710,len:0x90c
load:0x403ce710,len:0x2624
entry 0x403cc710
ESP-ROM:esp32c3-api1-20210207
Build:Feb  7 2021
rst:0x8 (TG1WDT_SYS_RST),boot:0xd (SPI_FAST_FLASH_BOOT)
Saved PC:0x40380000
SPIWP:0xee
mode:DIO, clock div:1
load:0x3fcd5810,len:0x438
load:0x403cc710,len:0x90c
load:0x403ce710,len:0x2624
entry 0x403cc710

kelenMr, 2025-03-16:
应用了这个#include <User_Setups/Setup70tk_ESP32_C3_GC9A01.h>   
// Setup for the ESP32 C3 with GC90A1 display
// 2024.11.14 by Timucin KANATLI

#define USER_SETUP_ID 70
#define GC9A01_DRIVER

#define TFT_CS   10
#define TFT_MOSI 7
#define TFT_MISO 5
#define TFT_SCLK 6

#define TFT_DC    2
#define TFT_RST   -1

#define TFT_BL   3             // LED back-light control pin
#define TFT_BACKLIGHT_ON HIGH  // Level to turn ON back-light (HIGH or LOW)

#define LOAD_GLCD
#define LOAD_FONT2
#define LOAD_FONT4
#define LOAD_FONT6
#define LOAD_FONT7
#define LOAD_FONT8
#define LOAD_GFXFF

#define SMOOTH_FONT
#define SPI_FREQUENCY  20000000   
#define SPI_READ_FREQUENCY  6000000 使用的是这个

kelenMr, 2025-03-16:
我正常按照您的修改后遇到的就是
Rebooting...
ESP-ROM:esp32c3-api1-20210207
Build:Feb  7 2021
rst:0x3 (RTC_SW_SYS_RST),boot:0xd (SPI_FAST_FLASH_BOOT)
Saved PC:0x40381ef2
SPIWP:0xee
mode:DIO, clock div:1
load:0x3fcd5810,len:0x438
load:0x403cc710,len:0x90c
load:0x403ce710,len:0x2624
entry 0x403cc710
Guru Meditation Error: Core  0 panic'ed (Store access fault). Exception was unhandled.

Core  0 register dump:
MEPC    : 0x42001d74  RA      : 0x42001d64  SP      : 0x3fc95ad0  GP      : 0x3fc8c000  
TP      : 0x3fc861e4  T0      : 0xffffffff  T1      : 0xffc3ffff  T2      : 0x00000001  
S0/FP   : 0x60004000  S1      : 0x3fc8d590  A0      : 0x00000001  A1      : 0xffffffff  
A2      : 0x00000000  A3      : 0xfdffffff  A4      : 0x08000000  A5      : 0x00000010  
A6      : 0x00000000  A7      : 0x04c4b400  S2      : 0x00000001  S3      : 0x00000000  
S4      : 0x00000000  S5      : 0x00000000  S6      : 0x00000000  S7      : 0x00000000  
S8      : 0x00000000  S9      : 0x00000000  S10     : 0x00000000  S11     : 0x00000000  
T3      : 0x00000000  T4      : 0x0000000f  T5      : 0x003c0000  T6      : 0x00000003  
MSTATUS : 0x00001881  MTVEC   : 0x40380001  MCAUSE  : 0x00000007  MTVAL   : 0x00000010  
MHARTID : 0x00000000  

Stack memory:
3fc95ad0: 0x00000000 0x3fc8f000 0x02625a00 0x00000001 0x00000001 0x3fc8f000 0x3fc8d590 0x4200208e
3fc95af0: 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x3fc8f000 0x42003000 0x42000034
3fc95b10: 0x00000000 0x00000000 0x42003000 0x42006374 0x00000000 0x00000000 0x00000000 0x4038795e
3fc95b30: 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0xa5a5a5a5 0xa5a5a5a5 0xa5a5a5a5
3fc95b50: 0xa5a5a5a5 0xa5a5a5a5 0xa5a5a5a5 0xa5a5a5a5 0xbaad5678 0x00000160 0xabba1234 0x00000154
3fc95b70: 0x3fc95ab0 0x74b506d6 0x3fc8e32c 0x3fc8e32c 0x3fc95b70 0x3fc8e324 0x00000018 0xfc42e74c
3fc95b90: 0xd3ba9f0c 0x3fc95b70 0x00000000 0x00000001 0x3fc93b60 0x706f6f6c 0x6b736154 0x37d91f00
3fc95bb0: 0x00cc9753 0x00000000 0x3fc95b50 0x00000001 0x00000002 0x00000000 0x00000000 0x00000000
3fc95bd0: 0x3fc8fa04 0x3fc8fa6c 0x3fc8fad4 0x00000000 0x00000000 0x00000001 0x00000000 0x00000000
3fc95bf0: 0x00000000 0x42020994 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000
3fc95c10: 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000
3fc95c30: 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000
3fc95c50: 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000
3fc95c70: 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000
3fc95c90: 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000
3fc95cb0: 0x00000000 0x00000000 0x00000000 0x00000000 0x76000000 0xbaad5678 0x00000060 0xabba1234
3fc95cd0: 0x00000054 0x00000000 0x3fc95cd4 0x3fc95b70 0x00000000 0x00000000 0x3fc95cec 0xffffffff
3fc95cf0: 0x3fc95cec 0x3fc95cec 0x00000000 0x3fc95d00 0xffffffff 0x3fc95d00 0x3fc95d00 0x00000000
3fc95d10: 0x00000001 0x00000000 0x1b00ffff 0x00000000 0xb33fffff 0x00000000 0xbaad5678 0x00000060
3fc95d30: 0xabba1234 0x00000054 0x00000000 0x3fc95d38 0x00000000 0x00000000 0x00000000 0x3fc95d50
3fc95d50: 0xffffffff 0x3fc95d50 0x3fc95d50 0x00000000 0x3fc95d64 0xffffffff 0x3fc95d64 0x3fc95d64
3fc95d70: 0x00000001 0x00000001 0x00000000 0xf400ffff 0x00000000 0xb33fffff 0x00000000 0xbaad5678
3fc95d90: 0x0000001c 0xabba1234 0x00000010 0x00000000 0x00000000 0x3fc8bd10 0x4201eec2 0xbaad5678
3fc95db0: 0x00046959 0x3fc8f014 0x3fc8f014 0x75ee8613 0xc8f5b4a6 0x74787599 0x8cfcad19 0xbc173f70
3fc95dd0: 0x9949575a 0x7ac24fda 0x55bd6163 0xf9cdf293 0x27d31f6c 0x0b732339 0x63a3528e 0xe86eeac9
3fc95df0: 0x12849c6f 0x8e724d15 0x13457d14 0x1fc6803e 0x27c0cec6 0xf0f56135 0x0b216bff 0x1e9ac86d
3fc95e10: 0x8f517b2c 0xfdff26fe 0x1d6af512 0xe882975c 0x7ab09ba5 0x879645e6 0xa66423b5 0x46c68618
3fc95e30: 0xec065008 0x9a73a610 0xcaf7d52c 0x2c3df590 0x18e923a9 0x0598e19c 0xc31e083f 0xa16f4245
3fc95e50: 0x0b9d8535 0x33eae24f 0x444a98ec 0x54b9a99b 0xd7092a4c 0xeb61687f 0x0027eb2d 0xc75e4999
3fc95e70: 0xfda3a64a 0xda0619c0 0x1f38118d 0x2631d0f1 0x7a9f90c6 0xef17ad92 0x88b2af17 0x34b27d5d
3fc95e90: 0xe56df29c 0x2d18ec78 0xbf825e33 0x094c3446 0x5356a3c9 0x5ad9a4af 0x752540db 0xf7ed3dee
3fc95eb0: 0x1bd0be68 0xc5e809af 0x88cb316b 0xca697c09 0x09675897 0xf203ef5d 0xebb7db28 0xfff84ab这个错误我得版本号是[env:esp32c3]
platform = espressif32
board = esp32-c3-devkitm-1
framework = arduino
lib_deps = bodmer/TFT_eSPI@^2.5.43

kelenMr, 2025-03-16:
已经成功点亮但是向左边偏移了有解决方案嘛

holgiw, 2025-03-25:
any news here? i have same issue with ESP32 C3, GC9A01_DRIVER, IDE 2.3.4, ESP32 V 3.1.3,  reboot in a loop

Basitadam, 2025-03-25:
Yes. Use this (change the file type to .h and copy it to \TFT_eSPI\Processors folder)

[TFT_eSPI_ESP32_C3.txt](https://github.com/user-attachments/files/19454409/TFT_eSPI_ESP32_C3.txt)

holgiw, 2025-03-26:
thanks, but now the tft.init() doesnt come back

holgiw, 2025-03-26:
i have add #define USE_HSPI_PORT to the User_Setup.h and it works now