Bodmer/TFT_eSPI #3564: ST7789v2 based display not working with ST7789/ST7789_2 drivers (but work with ILI9341 driver!)

MagnusThome, 2024-11-23:
I have a problem getting this ST7789v2 display to work with either ST7789 or ST7789_2 drivers. 
https://www.digikey.se/en/products/detail/az-displays/ATM0200B3/18796376
https://www.azdisplays.com/PDF/ATM0200B3.pdf

bit bash example output:

```
 ===========================
Register 0x01: 0x00
Register 0x04: 0x858552
Register 0x09: 0x610000
Register 0x0A: 0x08
Register 0x0B: 0x00
Register 0x0C: 0x06
Register 0x0D: 0x00
Register 0x0E: 0x00
Register 0x0F: 0x00
Register 0x2E: 0x00
Register 0x3F: 0xFFFFFFFF
Register 0xDA: 0x85
Register 0xDB: 0x85
Register 0xDC: 0x52
===========================

Looks like driver chip is: Unknown                                 
```

#### 1: ILI9341_DRIVER  works, kinda
TFT_eSPI examples work with the ILI9341_DRIVER  with the user settings as enclosed further below but only with this atypical **landscape** setting line: 
```
TFT_eSPI tft = TFT_eSPI(320,240);       // Invoke custom library
```
(regardless TFT_WIDTH and TFT_HEIGHT in user settings set as landscape or portrait)  
  
Colors seem to be off though:
[image]

I got a bit confused with the Matrix example since it mixed horizontal and vertical behavior in odd ways depending on setRotation but maybe that example is built like that.
   
#### 2: ST7789_DRIVER(s), no go
Both the ST7789_DRIVER and ST7789_2_DRIVER just give me a black screen whatever other things in user settings I change
   
#### 3: Adafruit ST7789  driver and example works
The Adafruit ST7789 (v1.10.4) drivers and the example at this link works with the display
https://github.com/adafruit/Adafruit-ST7735-Library/tree/master/examples/graphicstest_st7789
(a bit odd but a few times I only got it running properly if I added a call to tft.setRotation(1) after the line with tft.init(240, 320); but I had problems repeating this issue)

 
#### I use:
Arduino IDE
TFT_eSPI v2.5.43
rp2040  v2.4.0

```
#define ILI9341_DRIVER
//#define ST7789_DRIVER   
//#define ST7789_2_DRIVER    

#define CGRAM_OFFSET      // Library will add offsets required
#define TFT_RGB_ORDER TFT_RGB  // Colour order Red-Green-Blue
//#define TFT_RGB_ORDER TFT_BGR  // Colour order Blue-Green-Red
#define TFT_INVERSION_ON
//#define TFT_INVERSION_OFF

#define TFT_SPI_PORT 0 // Set to 0 if SPI0 pins are used, or 1 if spi1 pins used

#define TFT_WIDTH  320
#define TFT_HEIGHT 240

#define TFT_MOSI 3
#define TFT_MISO -1 
#define TFT_SCLK 2
#define TFT_CS   1

#define TFT_DC   8 
#define TFT_RST  7 
```

MagnusThome, 2024-11-24:
A bit old thread, but does it have any bearing on this? I do have a separate reset hooked up though as mentioned in this thread.
https://github.com/Bodmer/TFT_eSPI/issues/322

MagnusThome, 2024-11-24:
Have now tried everything above with an ESP32 instead of the RP2040. Identical results.

nstepanets, 2024-12-05:
Looks like I've encountered the same problem. 

The display on the ST7789v3 controller doesn't show any information (I've tried ST7789_DRIVER and ST7789_2_DRIVER). 
The image appears when ILI9341_DRIVER is selected, but with artifacts.

It seems that the problem appeared relatively recently, as this display worked before.

nstepanets, 2024-12-05:
It seems that the problem with the black screen depends on this option:

```
// Some ST7789 boards do not work with Mode 0
#ifndef TFT_SPI_MODE
  #if defined(ST7789_DRIVER) || defined(ST7789_2_DRIVER)
    #define TFT_SPI_MODE SPI_MODE3
  #else
    #define TFT_SPI_MODE SPI_MODE0
  #endif
#endif
```

The display works only if SPI_MODE0 mode is selected.

MagnusThome, 2024-12-05:
Thank you thank you thank you!!! 

Works great with #define TFT_SPI_MODE SPI_MODE0 added :-)

MagnusThome, 2024-12-05:
Added:

#define TFT_SPI_MODE SPI_MODE0