Bodmer/TFT_eSPI #3568: TFT Feather S3 resets when init() is called

davepl, 2024-11-29:
Here's the entirety of the project, so it should be easy to replicate - but I can't imagine what it could be.

```
#include <Arduino.h>
#include <TFT_eSPI.h>

TFT_eSPI tft = TFT_eSPI();

void setup() 
{
  // put your setup code here, to run once:
  Serial.begin(115200);
  Serial.println("Hello World!");
  tft.init();
  tft.fillScreen(TFT_BLUE);
}

void loop() 
{
  Serial.println("Loop");
  delay(1000);
}

[env:adafruit_feather_esp32s3]
platform = espressif32
board = adafruit_feather_esp32s3
framework = arduino
upload_port = /dev/cu.usbmodemA142201
monitor_port = /dev/cu.usbmodemA142201
monitor_speed = 115200
lib_deps = bodmer/TFT_eSPI@^2.5.43
```

When I build and upload this, it resets in tft.init() on all of my TFT S3 Feather Adafruit boards.  Any ideas?

Basitadam, 2024-11-30:
Can you add this line to your setup file 
```
 #define USE_HSPI_PORT
```

or compiler this to compiler flags

```
 -D USE_HSPI_PORT
```
and test again

davepl, 2024-12-01:
That fixed it.  The pragmatist in me says "Holw cow, thanks!".  The programmer in me asks "WTF is that all bout then?"

Basitadam, 2024-12-01:
Tells the library which SPI port of S3 to use. Was not needed before but something changed in esp-idf library, now it is needed

sjorsvr, 2024-12-01:
When you have an SD card connected as well in your project, like mine, and use the SD library the code will crash. The S3 will not reboot anymore but when creating f.e. a new folder on the card the code will crash.