Bodmer/TFT_eSPI #3571: Add Heltec T190 spi  working!!!!

tonbor, 2024-12-04:
Well It took me a while but it is working :-)

// Support for 1.9" 320x172 Heltec T190 IPS TFT Display
#define USER_SETUP_ID 190

#define ST7789_DRIVER  // Full configuration option, define additional parameters below for this display
#define USE_HSPI_PORT // HSPI pins 10-13
#define TFT_RGB_ORDER TFT_BGR  // Colour order Blue-Green-Red

#define TFT_WIDTH 170  // ST7789 172 x 320
#define TFT_HEIGHT 320  // ST7789 240 x 320

#define TFT_BL           17    // LED back-light control pin
#define TFT_BACKLIGHT_ON HIGH  // Level to turn ON back-light (HIGH or LOW)

#define TFT_MOSI 48
#define TFT_SCLK 38
#define TFT_CS   39   // Chip select control pin
#define TFT_DC   47  // Data Command control pin
#define TFT_RST  40  // Reset pin (could connect to RST pin)

#define LOAD_GLCD   // Font 1. Original Adafruit 8 pixel font needs ~1820 bytes in FLASH
#define LOAD_FONT2  // Font 2. Small 16 pixel high font, needs ~3534 bytes in FLASH, 96 characters
#define LOAD_FONT4  // Font 4. Medium 26 pixel high font, needs ~5848 bytes in FLASH, 96 characters
#define LOAD_FONT6  // Font 6. Large 48 pixel font, needs ~2666 bytes in FLASH, only characters 1234567890:-.apm
#define LOAD_FONT7  // Font 7. 7 segment 48 pixel font, needs ~2438 bytes in FLASH, only characters 1234567890:-.
#define LOAD_FONT8  // Font 8. Large 75 pixel font needs ~3256 bytes in FLASH, only characters 1234567890:-.
//#define LOAD_FONT8N // Font 8. Alternative to Font 8 above, slightly narrower, so 3 digits fit a 160 pixel TFT
#define LOAD_GFXFF  // FreeFonts. Include access to the 48 Adafruit_GFX free fonts FF1 to FF48 and custom fonts

#define SMOOTH_FONT

#define SPI_FREQUENCY 27000000

jamesgbahr, 2024-12-25:
have you gotten the lora working on t190?

tonbor, 2024-12-25:
The TFT is working, the sx1262 too.

jamesgbahr, 2024-12-25:
do you have any test code, i cannot get lora to send a packet with t190

tonbor, 2024-12-25:
Sorry no, I only use the sx1262 to receive my weather sensors sending at 868mhz.

jamesgbahr, 2024-12-25:
even a reece test code would help not sure what i'm doing wrong trying to
get lora to init

tonbor, 2024-12-25:
[image]

tonbor, 2024-12-25:
[image]

closed 2024-12-25 by tonbor

tonbor, 2024-12-25:
> even a receive test code would help not sure what i'm doing wrong trying to get lora to init
> […](#)

Try https://github.com/jgromes/RadioLib/tree/master/examples/SX126x

jamesgbahr, 2024-12-25:
could you please share example, i have tried the examples but i believe one of my libraries are conflicting... if you could share platformio.ini would help immensely.

DonatPlayZ, 2025-12-17:
i would like to get a simple example Script with the TFT and SX1262 too.. I cant get them running i think its a simple initialisation error... 

A simple "Hello World" script would be enough