Bodmer/TFT_eSPI #3573: OpenFontRender not working - LilyGo T-Display 16Mb 1.14

edimarcr, 2024-12-05:
Board installed:    Esp32 by espressif systems library v3.0.7 (I have tested other versions without success.)

Graphics library:   TFT_eSPI by Bodmer v2.5.43

TTF font render support library:    OpenFontRender v1.2

I tried to compile the OpenFontRender library examples. They all give the following error:

Sketch uses 414541 bytes (31%) of program storage space. Maximum is 1310720 bytes.
Global variables use 21264 bytes (6%) of dynamic memory, leaving 306416 bytes for local variables. Maximum is 327680 bytes._

Basitadam, 2024-12-05:
I can't see the error

edimarcr, 2024-12-07:
> I can't see the error

This message indicates that the hardware does not have enough memory to run the program. If this limitation is occurring with sample programs, I understand that you will not be able to use OpenFontRender on this board until an update is made to one of the software components.

closed 2024-12-10 by edimarcr