Bodmer/TFT_eSPI #3580: White screen issue, esp32s3 and ili9341

JasonKam002, 2024-12-14:
I'm trying to get my ili9341 to display stuff with the tft library but it doesn't seem to work. I can ensure the pins are correctly configured, the user setup should also be correctly configured, the select user setup also selected the right user setup. I've also tried using the adafruit library to display stuff and it worked, so i know my ili9341 and esp32 are correctly connected too. I really can't find a solution to this, any help would be appreciated :)

I am using Arduino IDE and the tft library is at the latest version.

Basitadam, 2024-12-16:
Can you send your user setup contents

JasonKam002, 2024-12-16:
Here is my user setup for ili3941

[User_Setup_ili9341.zip](https://github.com/user-attachments/files/18157148/User_Setup_ili9341.zip)

Basitadam, 2024-12-17:
Can you add this line and try again
```
 #define USE_HSPI_PORT
```

JasonKam002, 2024-12-19:
[image]
like this? It didn't work

Basitadam, 2024-12-19:
Correct.
In your User_Setup_Select.h file, only uncommented line is "#include <User_Setup_ili9341.h>" ,right?

I don't see your backlight pin defined. Assuming it is hardwired ON?

Can you run this sketch and send the output: File>Examples>TFT_eSPI>Test and diagnostic>Read_User_Setup

JasonKam002, 2024-12-21:
Yes the backlight pin is hardwired on

The output was:
`[code]
TFT_eSPI ver = 2.5.43
Processor    = ESP32
Frequency    = 240MHz
Transactions = Yes
Interface    = SPI
Display driver = 9341
Display width  = 240
Display height = 320

MOSI    = GPIO 11
MISO    = GPIO 19
`
Btw im so confused on why that sketch worked, because all of the defines were for esp8266

JasonKam002, 2024-12-21:
Hey really good news, it worked!!! All the example sketch works now, though i have no clue why they work cuz i didn't change anything but adding the #define_HSPI_port since 3 days ago, but 3 days ago it still wasn't working. But its working now thank you so much!!!!!!!!!

Basitadam, 2024-12-21:
> Hey really good news, it worked!!! All the example sketch works now, though i have no clue why they work cuz i didn't change anything but adding the #define_HSPI_port since 3 days ago, but 3 days ago it still wasn't working. But its working now thank you so much!!!!!!!!!

Nice. 
Can you please close the issue. it is still open

closed 2024-12-21 by JasonKam002