Bodmer/TFT_eSPI #3588: Tumbler Numbers - Used to work now doesn't

DrJaymz, 2024-12-19:
I created a tumbler routine like the old tumblers you'd see on a Gas pump or car dashboard.
It works by working out where the digit we want is in the "window" then adding the digit before and after.  Its all drawn on a sprite which is then pushed to the display.

This used to look cool but now no longer works.  I think the problem is that the way in which text that is printed out of bounds is handled.  The previous and next digits may be partially or completely obscured.  This didn't used to matter.

I found the function setTextWrap and disabled it for x and y - doesn't make any difference.
For the following the sprite is 55x55 and I think digits are 0 to 99.  What it should do is show 10,20,30,40 passing by.

If you can think of a better way with clipping or scrolling rectangles I'm happy to change it.

```
void printTens(float digit, int x)
{
  digits.fillSprite(TFT_BLACK);
  digits.setTextDatum(ML_DATUM);
  digits.setFreeFont(&FreeSansBold12pt7b);
  digits.setTextWrap(false,false);

  digit += 0.5;

  float pixel_fraction = (float)digits.height() / 10;
  float vLocation;
  int figure = (int)digit;

  if (figure == 10)
    figure = 0;

  sprintf(strBuff, "%01d0", figure);
  vLocation = (10.0 * (digit - int(digit))) * pixel_fraction;
  Serial.printf("GVLOC: %0.1f %i\n",vLocation, digits.height());
  digits.setTextColor(TFT_GREEN);
  digits.drawString(strBuff, 0, vLocation, 1);

  figure++;
  if (figure == 10)
    figure = 0;

  sprintf(strBuff, "%01d0", figure);
  vLocation = ((10.0 * (digit - int(digit))) * pixel_fraction) - (float)digits.height();
  Serial.printf("YVLOC: %0.1f %i\n",vLocation, digits.height());
  digits.setTextColor(TFT_YELLOW);
  digits.drawString(strBuff, 0, vLocation, 1);

  figure = (int)digit;
  figure--;

  sprintf(strBuff, "%01d0", figure);
  vLocation = ((10.0 * (digit - int(digit))) * pixel_fraction) + (float)digits.height();
    Serial.printf("CVLOC: %0.1f %i\n",vLocation, digits.height());
  digits.setTextColor(TFT_CYAN);
  digits.drawString(strBuff, 0, vLocation, 1);

  digits.pushSprite(x, PANEL_MIDY);
}
```

DrJaymz, 2024-12-20:
[image]

I see these lines have appeared.  I commented out the checks on the Y coordinate but there must be more than that.
I would have thought partially obscured text was fairly commonly used. (but willing to conceed there is a better way).

jaley95630, 2025-02-08:
I'm not expert enough to analyze your code, but I did recently write some code that I think does what you want.  You'll have to change the font and FONT_HEIGHT.
I do hope it's not bad etiquette to post bits of code here.

Regards,

-Jeff

/*
  Sketch to show how to use pushSprite to show just a portion of a Sprite.
  By changing which portion of the Sprite is displayed, a scrolling number effect can be made.
  The Sprite "sdigits" contains a column of digits, 0 - 9.
*/
#include "GillSansNovaCond42.h" // font.
#define FONT_HEIGHT 32

#define IMG_WIDTH 320
#define IMG_HEIGHT 170
#include <TFT_eSPI.h>

TFT_eSPI tft = TFT_eSPI();
TFT_eSprite sdigits = TFT_eSprite(&tft); // Sprite object sdigits

//==========================================================================================
void setup() {
  tft.init();
  tft.setRotation(3);
  tft.fillScreen(TFT_BLUE);
  
  
  // Create a sprite for the scrolling numbers
  sdigits.setColorDepth(8);
  sdigits.createSprite(32, 11 * FONT_HEIGHT);
  sdigits.fillSprite(TFT_BLACK); // Fill sprite with black
  sdigits.loadFont(GillSansNovaCond42);
  sdigits.setTextColor(TFT_WHITE); // White text, no background
  sdigits.setTextDatum(TL_DATUM);  // Top Left coordinate datum
   
  long int row;
  for (row=0; row < 10; row++) {
    sdigits.drawNumber(row, 5, row * FONT_HEIGHT); // value, x, y, font#
  }
  sdigits.drawNumber(0, 5, 10 * FONT_HEIGHT); // add an extra 0 at the end.

}

//==========================================================================================
void loop() {
  sdigits.pushSprite(0, 0);
  sdigits.pushSprite(100, 70, 0, 0, 32, 32);

  for (int j = 0; j < 10; j++) {
    for (int i = 0; i < FONT_HEIGHT; i++) {
      sdigits.pushSprite(100, 70, 0, (j*FONT_HEIGHT)+i, 32, 32);
      delay (10);
    }
    delay (1000);
  }
   
} // Loop back and do it all again
//==========================================================================================