Bodmer/TFT_eSPI #3591: ILI9488 8bit Parallel and ESP32-S3 color issues

PeanutButcher, 2024-12-22:
Hi guys,

I feel a little bit confused about my situation. When I use a TFT ILI9488 SPI and the "TFT_flash_jpg" example of the TFT_eSPI library, all is working fine, the jpeg has the right colors. But I connected a parallel one, the colors do not look right at all, and there are little artifacts :

[image]
[image]
[image]

I have the same problem using a PNG decoder and a GIF Decoder.

The strangest thing is that when I try the TFT_graphicstest_one_lib example, the colors seem correct. The same was observed for all other examples of the library that does not use a decoder (jpg, png or gif) : 

[image]

I can conclude that when I create sprites or shapes using the the components of the library, I have no issues, but when using a decoder, I have some.

What is even more strange is that when I import a jpg that contains only red, green and blue, it looks correct (though I find the red a little bit washed out, but I think this is normal).

Original image :
[image]

Result on screen :
[image]

What I did/tried :
1) I uncommented "Setup70d_ILI9488_S3_Parallel.h"
2) I checked again and again my wiring
3) I tried the function "tft.setSwapBytes(true)"

If you guys have any tip to resolve that issue, I would be please to try it out !

Basitadam, 2024-12-27:
Can you try with adding this line tp your setup file:
```
#define TFT_RGB_ORDER TFT_BGR 
```
only for testing jpg, png or gif

PeanutButcher, 2024-12-27:
Hi, thanks for this proposal.

I have the exact same result as before, pixel perfect.

PeanutButcher, 2024-12-28:
I am the dumbest idiot ever

To all newbies like me, check the driver of your TFT ; for example a 3.5" TFT does not mean this is a 9488 driver...
Mine was actually 9486. I uncommented the driver 9486, and I used the Setup_70d and just commented the "#define ILI9488_DRIVER".

All is working fine now.

closed 2024-12-28 by PeanutButcher