Bodmer/TFT_eSPI #3601: TFT_Espi and SD lib not compatible

tonbor, 2024-12-30:
Example with SD not working with Esp32 S3 because TFT uses HSPI and SD lib not. So the SPI bus brakes when using both TFT and SD. Solution: https://forum.arduino.cc/t/esp32-initialize-hspi-as-default-spi/1155093/11

#define HSPI_SCK 12
#define HSPI_MISO 13
#define HSPI_MOSI 11
#define SD_CS 10
 hspi = new SPIClass(HSPI);
    hspi->begin(HSPI_SCK, HSPI_MISO, HSPI_MOSI, SD_CS);

    if(!SD.begin(SD_CS, *hspi)){
        Serial.println("Card Mount Failed");
        return;
    }
     tft.begin();

Took me 2 days to find this out!!!!!

tonbor, 2025-01-04:
> see #3464 : Replace #define SPI_PORT FSPI into #define SPI_PORT 2 in TFT_eSPI_ESP32_S3.h in the Processors folder in the tft_espi library. In windows you maybe also need to clear the folder c:\Users\username\AppData\Local\arduino\ : this clears the precompiled libraries then it works : I have a project with tft_espi library and sd library

in the user setup I have #define USE_HSPI_PORT so where must alter? Here ? 
// ESP32 specific SPI port selection
#ifdef USE_HSPI_PORT
  #ifdef CONFIG_IDF_TARGET_ESP32
    #define SPI_PORT HSPI  //HSPI is port 2 on ESP32
  #else
    
this a will never reach

     #elif CONFIG_IDF_TARGET_ESP32S3
    #define SPI_PORT FSPI

will I ?

sjorsvr, 2025-01-04:
// ESP32 specific SPI port selection
#ifdef USE_HSPI_PORT
  #ifdef CONFIG_IDF_TARGET_ESP32
    #define SPI_PORT HSPI  //HSPI is port 2 on ESP32
  #else
    #define SPI_PORT 3     //HSPI is port 3 on ESP32 S2
  #endif
#elif defined(USE_FSPI_PORT)
    #define SPI_PORT 2 //FSPI(ESP32 S2)
#else
  #ifdef CONFIG_IDF_TARGET_ESP32
    #define SPI_PORT VSPI
  #elif CONFIG_IDF_TARGET_ESP32S2
    #define SPI_PORT 2 //FSPI(ESP32 S2)
  #elif CONFIG_IDF_TARGET_ESP32S3
//    #define SPI_PORT FSPI // original line replaced by next line : <<<<<<<<<<<<<  replaced by
    #define SPI_PORT 2          // <<<<<<<<<<<<<<<<<<<<
  #endif
#endif

#ifdef RPI_DISPLAY_TYPE
  #define CMD_BITS (16-1)
#else
  #define CMD_BITS (8-1)
#endif

sjorsvr, 2025-01-04:
and this was the only change to get tft_espi working again with esp32 2.0.15 and higher

tonbor, 2025-01-04:
> and this was the only change to get tft_espi working again with esp32 2.0.15 and higher

I got TFT working with esp 32 3.0.6. But not TFT and SD together on the same SPI with only different CS pins

sjorsvr, 2025-01-04:
did you clear the c:\Users\username\AppData\Local\arduino\ folder ?

sjorsvr, 2025-01-04:
I use following pins :
D display S sd card
1  D cs
7 D dc
10 S cs
11 D sda and S mosi
12 D clk and S clk
13 S miso
 in arduino I use board esp32s3 Dev Module

and
this is in my user setup in the tft_espi library
....
#define TFT_CS   1
#define TFT_MOSI 11
#define TFT_MISO 13
#define TFT_SCLK 12

#define TFT_DC    7
//#define TFT_RST   10
#define TFT_RST   -1
....

tonbor, 2025-01-04:
got tft working for weeks, that is not the problem

tonbor, 2025-01-04:
I use following pins :
D display S sd card
18  D cs
7 D dc
10 S cs
11 D sda and S mosi
12 D clk and S clk
13 S miso
 in arduino I use board esp32s3 Dev Module

and
this is in my user setup in the tft_espi library
....
#define TFT_CS   18
#define TFT_MOSI 11
#define TFT_MISO 13
#define TFT_SCLK 12

#define TFT_DC    7
//#define TFT_RST   10
#define TFT_RST   15

sjorsvr, 2025-01-04:
did it work with esp32 2.0.14 ?

tonbor, 2025-01-04:
yes, as I said within the issue

sjorsvr, 2025-01-04:
look into my change above , And I did NOT include in my  user setup :
#define USE_HSPI_PORT 
so in the code above it will reach the replaced line

tonbor, 2025-01-05:
Thanks, this is working, :-) #define USE_HSPI_PORT makes spi working SPI_PORT 3 your change makes spi working at SPI_PORT 2, Suppose sd is working at SPI_PORT 2 too.

tonbor, 2025-01-05:
> Thanks, this is working, :-) #define USE_HSPI_PORT makes spi working SPI_PORT 3 your change makes spi working at SPI_PORT 2, Suppose sd is working at SPI_PORT 2 too.

had to change the ino setup too:
void setup() {
  Serial.begin(115200);

  // Set all chip selects high to avoid bus contention during initialisation of each peripheral
  //digitalWrite(22, HIGH); // Touch controller chip select (if used)
    pinMode(18, OUTPUT);
  pinMode(10, OUTPUT);
  digitalWrite(18, HIGH); // TFT screen chip select
  digitalWrite( 10, HIGH); // SD card chips select, must use GPIO 5 (ESP32 SS)
  if (!SD.begin(10)) {
    Serial.println("Card Mount Failed");
    return;
  }
  uint8_t cardType = SD.cardType();

  if (cardType == CARD_NONE) {
    Serial.println("No SD card attached");
    return;
  }
  Serial.print("SD Card Type: ");
  if (cardType == CARD_MMC) {
    Serial.println("MMC");
  } else if (cardType == CARD_SD) {
    Serial.println("SDSC");
  } else if (cardType == CARD_SDHC) {
    Serial.println("SDHC");
  } else {
    Serial.println("UNKNOWN");
  }
  uint64_t cardSize = SD.cardSize() / (1024 * 1024);
  Serial.printf("SD Card Size: %lluMB\n", cardSize);
  tft.begin();
  Serial.println("initialisation done.");
}

mhmdhz, 2025-01-30:
I have a similar problem, but I use an internet module w5500 and an LCD tft with ILI9488 driver 
which is as follows.
When I initialize the LCD
and then the internet module or vice versa
my LCD touch stops working and the touch no longer works,
and my code is as follows.

my user setup is : 

#define USER_SETUP_INFO "User_Setup"
// #define ILI9486_DRIVER
#define ILI9488_DRIVER     // WARNING: Do not connect ILI9488 display SDO to MISO if other devices share the SPI bus (TFT SDO does NOT tristate when CS is high)
#define SUPPORT_TRANSACTIONS

// --------------------------------------------------------------------- esp32 Wroom DA Module 
// #define TFT_SPI_PORT   HSPI
#define USE_HSPI_PORT

#define TFT_MISO    12  // SPI MISO (connect to display and touch if shared)
#define TFT_MOSI    13  // SPI MOSI (connect to display and touch if shared)
#define TFT_SCLK    14  // SPI Clock

// TFT display specific
#define TFT_CS      15  // Chip Select for TFT display (ILI9488)

// #define TFT_MISO    19  // SPI MISO (connect to display and touch if shared)
// #define TFT_MOSI    23  // SPI MOSI (connect to display and touch if shared)
// #define TFT_SCLK    18  // SPI Clock

// // TFT display specific
// #define TFT_CS      5  // Chip Select for TFT display (ILI9488)
#define TFT_DC      26   // Data/Command for TFT display
#define TFT_RST     25   // Reset for TFT display
#define TFT_BL      34   // Backlight control (if used)

// // Touch controller specific (e.g., XPT2046)
#define TOUCH_CS    33   // Chip Select for touch controller
// #define TOUCH_PEN   4   // PEN/IRQ pin for detecting touch events
#define TFT_WIDTH   320
#define TFT_HEIGHT  480

#define LOAD_GLCD   // Font 1. Original Adafruit 8 pixel font needs ~1820 bytes in FLASH
#define LOAD_FONT2  // Font 2. Small 16 pixel high font, needs ~3534 bytes in FLASH, 96 characters
#define LOAD_FONT4  // Font 4. Medium 26 pixel high font, needs ~5848 bytes in FLASH, 96 characters
#define LOAD_FONT6  // Font 6. Large 48 pixel font, needs ~2666 bytes in FLASH, only characters 1234567890:-.apm
#define LOAD_FONT7  // Font 7. 7 segment 48 pixel font, needs ~2438 bytes in FLASH, only characters 1234567890:-.
#define LOAD_FONT8  // Font 8. Large 75 pixel font needs ~3256 bytes in FLASH, only characters 1234567890:-.
//#define LOAD_FONT8N // Font 8. Alternative to Font 8 above, slightly narrower, so 3 digits fit a 160 pixel TFT
#define LOAD_GFXFF  // FreeFonts. Include access to the 48 Adafruit_GFX free fonts FF1 to FF48 and custom fonts
// Comment out the #define below to stop the SPIFFS filing system and smooth font code being loaded
// this will save ~20kbytes of FLASH
#define SMOOTH_FONT
#define SPI_FREQUENCY  40000000
// #define SPI_FREQUENCY  55000000 // STM32 SPI1 only (SPI2 maximum is 27MHz)
// #define SPI_FREQUENCY  80000000

// Optional reduced SPI frequency for reading TFT
#define SPI_READ_FREQUENCY  20000000

// The XPT2046 requires a lower SPI clock rate of 2.5MHz so we define that here:
 #define SPI_TOUCH_FREQUENCY  600000

and my code is this : 

#include <TFT_eSPI.h>
#include <lvgl.h>
#include "ui.h"
#include <Ethernet.h>
#include <EthernetUdp.h>
#include <SPI.h>
[... 89 lines cut ...]

#ifdef USE_HSPI_PORT
  #ifdef CONFIG_IDF_TARGET_ESP32
    #define SPI_PORT HSPI  //HSPI is port 2 on ESP32
  #else
    #define SPI_PORT 3     //HSPI is port 3 on ESP32 S2
  #endif
#elif defined(USE_FSPI_PORT)
    #define SPI_PORT 2 //FSPI(ESP32 S2)
#else
  #ifdef CONFIG_IDF_TARGET_ESP32
    #define SPI_PORT VSPI
  #elif CONFIG_IDF_TARGET_ESP32S2
    #define SPI_PORT 2 //FSPI(ESP32 S2)
  #elif CONFIG_IDF_TARGET_ESP32S3
//    #define SPI_PORT FSPI // original line replaced by next line : <<<<<<<<<<<<<  replaced by
    #define SPI_PORT 2          // <<<<<<<<<<<<<<<<<<<<
  #endif
#endif

#ifdef RPI_DISPLAY_TYPE
  #define CMD_BITS (16-1)
#else
  #define CMD_BITS (8-1)
#endif

tonbor, 2025-01-30:
So do you want to use HSPI and VSPI? What board you you use?

mhmdhz, 2025-01-31:
> So do you want to use HSPI and VSPI? What board do you use?

i am using the esp32 wroom Da module

tonbor, 2025-01-31:
SPI	MOSI	MISO	SCLK	CS
VSPI	GPIO 23	GPIO 19	GPIO 18	GPIO 5
HSPI	GPIO 13	GPIO 12	GPIO 14	GPIO 15