Bodmer/TFT_eSPI #3629: Latest build of tft_espi did a core panic at seedstudio xiao ESP32S3

fellpower, 2025-01-18:
Hey

 
i build an interface for a synthesizer : https://github.com/MacMannes/XVA1-User-Interface
after compiling all that codce with VSCode, i get a core panic on my seedstudio xiao esp32s3.

[image]

 
i backtraced the codes and get this:

[image]

this is my platform file:

[image]

I have to say, that im not a coder - so im lost. i can read, that its something with the tft_espi - but i cant fix that. whatever u need as info, i can bring it, if we can fix that problem.

this is my output if im compiling (for the board version i use)

[image]

i tried many versions of tft_espi - but nothing works.

i checked the line numbers of the backtrace errors - and its something with the tft_espi library - so it seems for me.

as i said, im not a coder - but i can help, with every information u need.

tonbor, 2025-01-20:
do you use the tft_st7789, what user_setup do you use? have you uncommend #define USE_HSPI_PORT for the esp32 S3

fellpower, 2025-01-21:
i tried wont work

if i change the espressif framework to 6.6.0 it works, but this is not that way to do ;)

fellpower, 2025-01-23:
i can compile, but i get an core panic, if i try to dra something. pixel or line or something....