Bodmer/TFT_eSPI #3636: ESP32C3 + ST7789 incompatible with ESP Arduino board package > 2.0.4

moeburn, 2025-01-25:
Only raise issues for problems with the library and/or provided examples. Post questions, comments and useful tips etc in the "Discussions" section.

To minimise effort to resolve issues the following should be provided as a minimum:
1. When compiling ESP32C3 + ST7789, it compiles successfully, but I get "Guru Meditation Error: Core 0 panic'ed (Store access fault). Exception was unhandled." and a bootloop.
2. Arduino IDE
3. Tried the latest TFT_eSPI library version 2.5.43, as well as 2.5.34 and 2.5.0
4. Tried board packages 3.1.1, 3.0.4, 2.0.17, and 2.0.4 (since it was mentioned on this library's readme).  Only 2.0.4 worked. 
5. ESP32C3
6. ST7789 
7. SPI

Plus further information as appropriate to the problem:
1. TFT to processor connections used - #define TFT_MOSI 6 //SDA
#define TFT_SCLK 4
#define TFT_DC    1  

#define TFT_RST  0  
2. A zip file containing your setup file [User_Setup.zip](https://github.com/user-attachments/files/18544536/User_Setup.zip)

3. A zip file  containing a simple and complete example sketch that demonstrates the problem but needs no special hardware sensors or libraries. [chrisvoc.zip](https://github.com/user-attachments/files/18544539/chrisvoc.zip)
4. "Guru Meditation Error: Core 0 panic'ed (Store access fault). Exception was unhandled."

datort, 2025-01-26:
I can second that. Just spent an hour trying to get a working esp8266 project running on my Waveshare ESP32 C3. The exception loop starts with the tft.init() call and results in the same Guru Meditation Error as described.

Basitadam, 2025-01-27:
Try with these. Extract and copy the files to "..\Arduino\libraries\TFT_eSPI\Processors" . Recompile and upload. (make sure to save a copy of your original files first, just in case)

[C3.zip](https://github.com/user-attachments/files/18563345/C3.zip)

moeburn, 2025-01-31:
> Try with these. Extract and copy the files to "..\Arduino\libraries\TFT_eSPI\Processors" . Recompile and upload. (make sure to save a copy of your original files first, just in case)
> 
> [C3.zip](https://github.com/user-attachments/files/18563345/C3.zip)

Well that got it working up to 2.0.17, thanks for that.  Now it won't even compile with 3+

Basitadam, 2025-02-02:
> Well that got it working up to 2.0.17, thanks for that. Now it won't even compile with 3+

What errors are you getting? Can you send a screenshot of the compiler output?