Bodmer/TFT_eSPI #3670: runtime errors ESP32C3 with the newest package: framework-arduinoespressif32 @ 3.20017.241212+sha.dcc1105b

homonto, 2025-02-23:
hi,

with this package: framework-arduinoespressif32 @ 3.20017.241212+sha.dcc1105b
it cannot initialise the screen:

```
06:45:32.950 > Guru Meditation Error: Core  0 panic'ed (Store access fault). Exception was unhandled.
06:45:32.950 > 
06:45:32.950 > Core  0 register dump:
06:45:32.950 > MEPC    : 0x4201d494  RA      : 0x4201d48c  SP      : 0x3fcaa080  GP      : 0x3fc92c00  
06:45:32.963 > TP      : 0x3fc62ee8  T0      : 0x00000003  T1      : 0xffc3ffff  T2      : 0x00000020  
06:45:32.967 > S0/FP   : 0x3fc96eb4  S1      : 0x00000000  A0      : 0x00000001  A1      : 0xffffffff  
06:45:32.978 > A2      : 0x00000003  A3      : 0x00000000  A4      : 0x08000000  A5      : 0x00000010  
06:45:32.984 > A6      : 0xffffffff  A7      : 0x04c4b400  S2      : 0x00000001  S3      : 0x00000000  
06:45:32.989 > S4      : 0x3fc92464  S5      : 0x3c173000  S6      : 0x3fc92441  S7      : 0x3fc97000  
06:45:33.000 > S8      : 0x3fc97000  S9      : 0x00000001  S10     : 0x3fc97000  S11     : 0x3fc97000  
06:45:33.006 > T3      : 0x02625a00  T4      : 0x0000000f  T5      : 0x003c0000  T6      : 0x00000003  
06:45:33.017 > MSTATUS : 0x00001881  MTVEC   : 0x40380001  MCAUSE  : 0x00000007  MTVAL   : 0x00000010  
06:45:33.022 > MHARTID : 0x00000000  
06:45:33.022 > 
06:45:33.022 > Stack memory:
06:45:33.028 > 3fcaa080: 0x00000003 0x00000000 0x03938700 0x00000301 0x3fc97000 0x3fc9d000 0x3fc96eb4 0x4201dc84
  #0  0x3fc96eb4 in ?? at src/main.cpp:31
  #1  0x4201dc84 in TFT_eSPI::init(unsigned char) at lib/TFT_eSPI/TFT_eSPI.cpp:693

```

installed:
```
PACKAGES: 
 - framework-arduinoespressif32 @ 3.20017.241212+sha.dcc1105b 
 - tool-esptoolpy @ 1.40501.0 (4.5.1) 
 - toolchain-riscv32-esp @ 8.4.0+2021r2-patch5
 
TFT_eSPI @ 2.5.43
```

works ok with: 
```
PACKAGES: 
 - framework-arduinoespressif32 @ 2.0.9 
 - tool-esptoolpy @ 1.40602.0 (4.6.2) 
 - toolchain-riscv32-esp @ 8.4.0+2021r2-patch5
 
TFT_eSPI @ 2.5.43
```

Basitadam, 2025-02-25:
Can you use this file and try again:

[TFT_eSPI_ESP32_C3.zip](https://github.com/user-attachments/files/18958958/TFT_eSPI_ESP32_C3.zip)

Copy it to
...\Arduino\libraries\TFT_eSPI\Processors
fol

homonto, 2025-02-26:
I can confirm that this SOLVED the problem. Can we have this correction in the main branch? ;-)

```
PACKAGES: 
 - framework-arduinoespressif32 @ 3.20017.241212+sha.dcc1105b 
 - tool-esptoolpy @ 1.40501.0 (4.5.1) 
 - tool-mkfatfs @ 2.0.1 
 - tool-mklittlefs @ 1.203.210628 (2.3) 
 - tool-mkspiffs @ 2.230.0 (2.30) 
 - toolchain-riscv32-esp @ 8.4.0+2021r2-patch5
...
|-- TFT_eSPI @ 2.5.43
```

Basitadam, 2025-02-26:
Can you run this simple test if it works (You may need to change Serial. to Serial0. depending on your board) :

```
/*
2024.10.22
-Simple fill screen test
*/

#include <TFT_eSPI.h>       // Include the graphics library

TFT_eSPI tft = TFT_eSPI();  // Create object "tft"

void setup(void) 
{
  Serial.begin(115200);
  delay(1000);
  Serial.println("Boot...");
  tft.init();
  tft.setRotation(1);
  tft.fillScreen(TFT_DARKGREY);
  tft.setTextFont(2);

  for(uint8_t ch=0; ch<3;ch++)
    {
     digitalWrite(TFT_BL,LOW);
     Serial.println("BL OFF");
     delay(150);
     digitalWrite(TFT_BL,HIGH);
     Serial.println("BL ON");
     delay(150);
    }
}

void loop()
{
  Serial.print("\n\n\n");
  Serial.println("Looping...");

  Serial.println("RED");
  tft.fillScreen(TFT_RED);
  tft.setCursor(TFT_HEIGHT/2,TFT_WIDTH/2);
  tft.print("RED");
  delay(2000); 

  Serial.println("GREEN");
  tft.fillScreen(TFT_GREEN);
  tft.setCursor(TFT_HEIGHT/2,TFT_WIDTH/2);
  tft.print("GREEN");
  delay(2000); 

  Serial.println("BLUE");
  tft.fillScreen(TFT_BLUE);
  tft.setCursor(TFT_HEIGHT/2,TFT_WIDTH/2);
  tft.print("BLUE");
  delay(2000); 
}
```

homonto, 2025-02-26:
Not sure what I am looking at: it starts and runs but the issue was with sprites:

```
06:43:10.393 > Boot...
06:43:10.985 > BL OFF
06:43:11.134 > BL ON
06:43:11.284 > BL OFF
06:43:11.436 > BL ON
06:43:11.584 > BL OFF
06:43:11.734 > BL ON
06:43:11.889 > 
06:43:11.890 > 
06:43:11.890 > 
06:43:11.890 > Looping...
06:43:11.890 > RED
06:43:13.913 > GREEN
06:43:15.937 > BLUE
06:43:17.966 > 
06:43:17.967 > 
06:43:17.967 > 
06:43:17.967 > Looping...
06:43:17.967 > RED
06:43:19.990 > GREEN
06:43:22.014 > BLUE
06:43:24.043 > 
```

Basitadam, 2025-02-26:
Wanted to be sure that display, so the SPI works

Basitadam, 2025-02-27:
Can you send your code? I can try to compile

ams-hh, 2025-02-27:
I can confirm the Guru-Chrash with an ESP32-S3-WROOM-1 Module.
Works until platform-espressif Package Version 6.6.0 (which comes with 
framework-arduinoespressif32 @ 3.20014.231204 (2.0.14))
From V6.7.0 on I see the Guru mediation error.

(with pioArduino Fork crashes as well)

But .... will this be fixed?  Last changes/fixes are quite long ago ...

Basitadam, 2025-02-27:
> I can confirm the Guru-Chrash with an ESP32-S3-WROOM-1 Module. Works until platform-espressif Package Version 6.6.0 (which comes with framework-arduinoespressif32 @ 3.20014.231204 (2.0.14)) From [...]

S3 fix is simple. Just add this to user setup .h file:
```
 #define USE_HSPI_PORT
```

homonto, 2025-02-27:
the function that is collapsing in new Arduino is this one:

```
void tft_message_mc_fixed_length(const char* message, uint8_t x, uint8_t y, uint16_t lenght, uint16_t font, uint16_t fgcolor, uint16_t bgcolor, bool full_width)
{    
    TFT_eSprite spr1 = TFT_eSprite(&tft);
    uint16_t l = spr1.textWidth(message,font);
    if (l > lenght) lenght = l;
    if (full_width) l = lenght;
    int h = spr1.fontHeight(font); 
    x = x-(l/2);
    int spr_w = l;
    int spr_h = h;
    spr1.setColorDepth(16);
    spr1.createSprite(spr_w, spr_h);
    spr1.fillSprite(bgcolor);

    spr1.setTextColor(fgcolor,bgcolor);
    spr1.setTextPadding(l);
    spr1.setTextDatum(MC_DATUM);
    spr1.drawString(message, l/2, h/2, font);

    spr1.pushSprite(x, y);
    spr1.deleteSprite();   
}
```

and is being called like this:

`tft_message_mc_fixed_length("any text", 60, 210, 118, 4, TFT_BLUE, TFT_WHITE, true); 
`

but as I said: works fine in 2.x

Basitadam, 2025-02-27:
Compiles fine. Strange:

[image]

homonto, 2025-02-27:
why wouldn't it? I said it panics from time to time - so run time error not compile error

Basitadam, 2025-02-27:
Missed it. Thought compile error

TheNextVision, 2025-03-27:
I have the same problem. Whenever I push a sprite the program crashes. Did you find a solution for the issue yet? @homonto

homonto, 2025-03-28:
no, I use old package 2.x and it works

zhcong, 2025-05-01:
same problem, but what's cause? see error happen at TFT_eSPI.cpp:693 in lib 2.5.43.
ask whitch version work out by the way? thk.