Bodmer/TFT_eSPI #3678: DC pin restriction for ESP32

redstonee, 2025-03-04:
Hello, I'm using ST7789 screen (whose DC pin is connected to IO42) with ESP32-S2 and ESP-IDF. When I'm configuring the pins, I notice that the DC pin of the screen is restricted up to IO31 in `Kconfig` while other pins seems normal according to the specific ESP32 model.  
However when I modify the restriction from 31 to 45 and set the DC pin to mine (which is 42) in sdkconfig, everything works nicely, so I wonder if there's a mistake in the `Kconfig` file.

sensenmann, 2025-03-05:
I have the same problem, with the new Waveshare ESP32-S3  2inch Display (https://www.waveshare.com/esp32-s3-lcd-2.htm).
It uses CS = 45
I can't get it to work.

redstonee, 2025-03-06:
Have u tried to edit `Kconfig` and see if it works?

sensenmann, 2025-03-06:
I'm quite new to ESP32 microcontroller development, so no - I'm not using menuconfig at the moment.
My project is running on platformio in CLion (without any makefiles, etc.)

So therefore in my platformio.ini I'm including my own config-tft.h file, where I have my Pin configuration.
This allows me to not have to change the User_Setup.h.

```
[env]
platform = espressif32
framework = arduino
platform_packages = framework-arduinoespressif32@https://github.com/espressif/arduino-esp32.git#2.0.14

...
build_flags =
    -include src/conf/hardware/config-tft.h
...

```

I'm pretty sure, that there are better or more modern ways to do it - but for mit it works really quiet well 😄

redstonee, 2025-03-06:
Yes Platform IO is much simpler to use, however it provides less project options compared with ESP-IDF which a little bit matters in my application😭