Bodmer/TFT_eSPI #3687: Blank screen with ESP32-S3 and 7789VI

TG2025, 2025-03-21:
Hey I'm failing to get any activity on my screen. I have tried with a few of the libraries examples like Arc_Fill or Colour_Test but the screen remains blank.
My Setup
-Board: ESP32-S3-MINI-1-N8
-Screen: [7789VI 2.4' SPI](https://newhavendisplay.com/2-4-inch-premium-spi-tft-without-touchscreen/) Connected with ribbon cable.
- Using Arduino ide with esp32 bsp version: 2.0.14 (with newer versions there is the [Guru Mediation Error on tft.begin()](https://github.com/Bodmer/TFT_eSPI/issues/3289) which I was able to resolve by switching to the older version, now it compiles and runs, but still with no response from the screen)
- Read_User_Setup output:
```
TFT_eSPI ver = 2.5.34 // also tried the newest
Processor    = ESP32
Frequency    = 240MHz
Transactions = Yes
Interface    = SPI
Display driver = 7789
Display width  = 240
Display height = 320

MOSI    = GPIO 11
MISO    = GPIO 13
SCK     = GPIO 12
TFT_CS   = GPIO 10
TFT_DC   = GPIO 14
TFT_RST  = GPIO 15

Font GLCD   loaded
Font 2      loaded
Font 4      loaded
Font 6      loaded
Font 7      loaded
Font 8      loaded
Smooth font enabled

Display SPI frequency = 27.00
[/code]
```
Circuit:

[image]
(pin numbers in the schematic are referring to the footprint not Gpio)

I have verified the screen works with another library, namely the Adafruit_ST7789.h +SPI.h, with witch I'm able to display things using the same pin definitions so the connections should be fine.

I have tried toggling additional setting in the User_Setup: 
```
//#define USE_HSPI_PORT
//#define TFT_SDA_READ
//different SPI_FREQUENCY speeds
```
With no result

I will appreciate any insight in to what else can be tried or what's the possible issue.

Basitadam, 2025-03-22:
Can you try using ```#define ST7789_2_DRIVER``` in your setup file

TG2025, 2025-03-26:
Thanks for the answer!
I have tried that, unfortunately the display also remains blank with `ST7789_2_DRIVER`
Curiously I see some activity on the screen when selecting `ST7735_DRIVER `

Basitadam, 2025-03-26:
I made another ST789 driver for myself. I don't recall why I needed it. I can share it but requires these files to be replaced:
- TFT_eSPI.cpp
- User_Setup_Select.h
These three to be added:
- ST7789_3_Init.h
- ST7789_3_Defines.h
- ST7789_3_Rotation.h

TG2025, 2025-03-29:
@Basitadam Would be very helpful if you could share it! I'll then try it out with my hardware.
Do you recall how your driver driver differs from the TFT_eSPI default ?

Basitadam, 2025-03-31:
Sorry. Not sure what is different. Maybe colors or basic init function. Here are the files (Please make sure you backup your files first). They were implemented to V2.5.43

https://www.filemail.com/d/jdynyxwzgljwbqf

tvixen, 2025-07-01:
Im running into same issue.
Could you share your files again.... they seem to have disappeared from the server.

> I made another ST789 driver for myself. I don't recall why I needed it. I can share it but requires these files to be replaced:
> 
> * TFT_eSPI.cpp
> * User_Setup_Select.h
>   These three to be added:
> * ST7789_3_Init.h
> * ST7789_3_Defines.h
> * ST7789_3_Rotation.h