Bodmer/TFT_eSPI #3697: Compilation error for ESP32-S3. [-Werror=misleading-indentation]

wailay, 2025-04-01:
I am using the ESP32-S3 and getting a compilation error on this header file
`TFT_eSPI/Processors/TFT_eSPI_ESP32_S3`

```cpp
components/TFT_eSPI/Processors/TFT_eSPI_ESP32_S3.h:559:25: error: this 'while' clause does not guard... [-Werror=misleading-indentation]      
  559 |                         while (*_spi_cmd & SPI_UPDATE);  \
```

Switching target back to esp32 and the compilation succeeds

Using vscode with ESP-IDF v5.4.0
TFT eSPI V2.5.43, GC9A01 driver with SPI

compiler: 
`xtensa-esp-elf\\esp-14.2.0_20241119\\xtensa-esp-elf\\bin\\xtensa-esp32-elf-gcc.exe",`

wailay, 2025-04-01:
I have this pull request that compile: https://github.com/Bodmer/TFT_eSPI/pull/3698

I suggest adding `clang-format` and keep the code properly formatted

libewa, 2025-04-23:
This problem is also present on `esp32c3`.

wailay, 2025-04-25:
It seems to only happen on Windows for me for some reason, on macos I am able to compile without any issues

libewa, 2025-04-26:
It depends on the compiler you're using. I get this error (on the ESP32-C3) also when compiling on macOS.