Bodmer/TFT_eSPI #3712: Connecting the SD1963 7-inch display to the ESP32-WROOM-32D - Problem

SP6U, 2025-04-19:
Hi everyone!

I'm currently learning how to program with PlatformIO in VSCode, and I’m working on a small project using an ESP32-WROOM-32D (38-pin) development board.
I’d like to connect a 7-inch TFT display with the SSD1963 controller, but I’m not sure where to start when it comes to hardware wiring and software setup.

Does anyone have a working example or ready-to-use solution (e.g. schematic, code, or project) for interfacing the SSD1963 (7") display with the ESP32?

I’d really appreciate some guidance on the following:

Which libraries are best for this display on ESP32?

Are there any drivers I need to install in PlatformIO?

How should I connect the display to the ESP32 (wiring)?

Any tips or example code to initialize the display and show basic graphics?

If someone could share a complete, ready-to-run ZIP file that works in VSCode with PlatformIO, I’d be incredibly grateful! That would really help me get started — I’ll take care of the rest on my own.

Thanks a lot in advance for your support and time!

SP6U, 2025-04-21:
@Bodmer
@v8-rt 
@jamesee 
  I've seen in other threads that you managed to successfully get this display working with the ESP32. Could I possibly ask for your help...?

tonbor, 2025-04-21:
16bit parallel is not supported

SP6U, 2025-04-21:
> 16bit parallel is not supported

Your answer really didn't change much, because I still don't know what's wrong and what's right. Is the code I wrote supposed to work? For now, all I care about is starting the display with touch...

tonbor, 2025-04-21:
> > 16bit parallel is not supported
> 
> Your answer really didn't change much, because I still don't know what's wrong and what's right. Is the code I wrote supposed to work? For now, all I care about is starting the display with touch...

can not find your code

tonbor, 2025-04-22:
in your setup there are only 8 bit parallel and not 16bit to send to the tft
// Szyna danych
#define TFT_D0   12  // DB0
#define TFT_D1   13  // DB1
#define TFT_D2   17  // DB2
#define TFT_D3   16  // DB3
#define TFT_D4    4  // DB4
#define TFT_D5    2  // DB5
#define TFT_D6   15  // DB6
#define TFT_D7   32  // DB7

SP6U, 2025-04-22:
> in your setup there are only 8 bit parallel and not 16bit to send to the tft
> // Szyna danych
> #define TFT_D0   12  // DB0
> #define TFT_D1   13  // DB1
> #define TFT_D2   17  // DB2
> #define TFT_D3   16 [...]

I'll make some corrections later

tonbor, 2025-04-22:
see https://github.com/lovyan03/LovyanGFX/discussions/345