Bodmer/TFT_eSPI #3717: ST7789 240 x 320 screen displays abnormally

sfcgta4794, 2025-04-26:
Hello, I am using some ESP32-S3 boards, each connects to 2 240 x 320 ST7789 screens. I follow the hardware and software design given by the following youtube example to produce a multiple display solution for playing some videos. 
https://www.youtube.com/watch?v=d49A0miFdqo

The ESP32 are driven by the programs which are under the following repository:
https://github.com/thelastoutpostworkshop/curved_screen

I could follow the video's instruction with its bundled codes to let the ESP boards to play the videos on the displays. However, the videos displayed on the screens looks strange and does not match with the original quality of the video. 

[image]

I wonder if my TFT-eSPI library version (2.5.43, which should be the latest) has issues in driving the 240 x 320 screens, or I have misconfigured the setup file so that it leads to such problem. I used "Setup24_ST7789.h" with the following settings. 

[image]

Thank you very much and looking forward to seeing any help or suggestion!

Basitadam, 2025-04-27:
Have you tried with ```setColorDepth(16);```

sfcgta4794, 2025-04-28:
> Have you tried with `setColorDepth(16);`

Would like to know where should I add setColorDepth into the code? Either the configuration file or my implementation ino file?

Basitadam, 2025-04-29:
if you are displaying images as sprites, you can use  ```imagename.setColorDepth(16);``` in your setup code, before ```imagename.createSprite(x,y);```

sfcgta4794, 2025-05-03:
> if you are displaying images as sprites, you can use `imagename.setColorDepth(16);` in your setup code, before `imagename.createSprite(x,y);`

I double-checked the program design with the example's author. He said the images are not displaying with sprites.

sfcgta4794, 2025-05-03:
And I wonder how can I check all ST7789 screen resolutions that are officially supported by the TFT-eSPI library? i would also like to pick some large resolution screens for further development.