Bodmer/TFT_eSPI #3719: Support for ESP32P4?

linghuye, 2025-04-29:
Use TFT_eSPI on ESP32 P4 dev board with a 7789 driver LCD screen, not working, the screen is black with backlight on, no crash.
Does anybody make TFT_eSPI work on ESP32 P4?
I tried TFT_eSPI_Generic.h and changed macro in TFT_eSPI_ESP32_S3.h to fit for ESP32P4, both not working, 
```
#define TFT_MOSI   29          
#define TFT_SCLK   30         
#define TFT_CS     28        
#define TFT_DC     31  
#define TFT_BL     2     
#define TFT_RST    52     

#define ST7789_DRIVER            
#define TFT_SDA_READ                // This flag work for esp32 C6
#define TFT_RGB_ORDER TFT_BGR   
#define TFT_WIDTH  240          
#define TFT_HEIGHT 320    
```

RedDoughnut, 2025-11-15:
I'm also having errors with tft_espi on the ESP32-P4

zekageri, 2025-11-18:
I'm also curious. I want to port my project to P4.