Bodmer/TFT_eSPI #3743: Title: ESP32-S3 (N8R8 OPI PSRAM) + ILI9341: TFT_eSPI crashes with StoreProhibited (EXCVADDR: 0x00000010) in tft.begin()

Stenliczz, 2025-05-24:
Hello Bodmer,

I'm encountering a persistent crash when trying to use TFT_eSPI with an ESP32-S3 on a Freenove ESP32-S3 WROOM N8R8 board. The crash is a Guru Meditation Error: Core 1 panic'ed (StoreProhibited). Exception was unhandled. EXCVADDR: 0x00000010 occurring during the tft.begin() call.

I've confirmed that PSRAM (8MB OPI) is active in the firmware when using the freenove_esp32_s3_wroom board definition with the latest PlatformIO espressif32 platform.

Crucially, the Adafruit_ILI9341 library works perfectly on the exact same hardware, pins, and board configuration (Freenove ESP32-S3 WROOM with PSRAM active). This strongly suggests the issue is specific to TFT_eSPI's interaction with the ESP32-S3 SPI/FSPI.

Additionally, TFT_eSPI (v2.5.30) does work without crashing if I use an older ESP32 Arduino Core (2.0.11 via PlatformIO espressif32 @ 6.4.0) with a generic esp32-s3-devkitc-1 board definition. However, with this older setup, PSRAM is reported as "No PSRAM" in the build and fails to initialize at runtime, making it unsuitable for my project.

This leads me to believe the crash might be related to TFT_eSPI's handling of ESP32-S3 with newer Arduino Core versions (>2.0.14, as some online discussions suggest).

Hardware Details:

Microcontroller: ESP32-S3 (on Freenove ESP32-S3 WROOM N8R8 board - 8MB Flash, 8MB OPI PSRAM)
Display: ILI9341
Pins Used:
MOSI: GPIO 11

[platformio.ini.txt](https://github.com/user-attachments/files/20422004/platformio.ini.txt)

[main.cpp.txt](https://github.com/user-attachments/files/20422005/main.cpp.txt)

[crash log.txt](https://github.com/user-attachments/files/20422006/crash.log.txt)

SCLK: GPIO 12
CS: GPIO 10
DC: GPIO 9
RST: -1 (Not connected, display has pull-up) 
Software Environment (PlatformIO) for Crashing TFT_eSPI setup:

platform = espressif32 (resolves to 6.10.0, using framework-arduinoespressif32 @ 3.20017.241212+sha.dcc1105b which is based on Arduino Core ~2.0.17)
board = freenove_esp32_s3_wroom (Build log confirms "8MB Flash / 8MB PSRAM")
TFT_eSPI versions tested that crash: 2.5.43, 2.5.30

Working Adafruit_ILI9341 Configuration (for comparison):

PlatformIO platform = espressif32 (latest, 6.10.0)
PlatformIO board = freenove_esp32_s3_wroom (Build log confirms "8MB Flash / 8MB PSRAM")
Pins: MOSI=11, SCLK=12, CS=10, DC=9, RST=-1
This setup works perfectly.

[env:adafruit_freenove_s3_test_working]
platform = espressif32
board = freenove_esp32_s3_wroom
framework = arduino
monitor_speed = 115200
lib_deps =
    adafruit/Adafruit ILI9341
    adafruit/Adafruit GFX Library

main.cpp
#include <Arduino.h>
#include <Adafruit_GFX.h>
#include <Adafruit_ILI9341.h>
#include <SPI.h>

#define TFT_CS    10
#define TFT_DC    9
#define TFT_RST   -1

Adafruit_ILI9341 tft = Adafruit_ILI9341(TFT_CS, TFT_DC, TFT_RST); // Uses default SPI pins (MOSI=11, SCLK=12 for S3 devkit variant)

void setup() {
  Serial.begin(115200);
  for (int i = 0; i < 10 && !Serial; ++i) { delay(100); }
  Serial.println("\n>>> Adafruit ILI9341 Test on Freenove ESP32-S3 WROOM <<<");
  if(psramFound()){
    Serial.printf("PSRAM Total: %u, Free: %u\n", ESP.getPsramSize(), ESP.getFreePsram());
  } else {
    Serial.println("PSRAM NOT FOUND!");
  }
  tft.begin();
  tft.setRotation(1);
  tft.fillScreen(ILI9341_MAGENTA);
  tft.setCursor(10, 50);
  tft.setTextColor(ILI9341_WHITE);
  tft.setTextSize(2);
  tft.println("Adafruit + Freenove S3 WORKS!");
  Serial.println("Adafruit display sketch finished setup. Display works.");
}
void loop() {}

fabse-hack, 2025-06-08:
same problem here :(

mischief, 2025-06-08:
i don't think we need "insight" from "claude 4" here, but rather some human understanding of the intent of upstream espressif's change of this API. a workaround in this repo for the broken upstream interface is ok, but what you have shared here is totally out of touch.

fabse-hack, 2025-06-08:
@mischief hmmmm okay, sorry..... i just beginn today with arduino ... esp32 s3, ili9341, lvgl and much problems.....

mike-lloyd03, 2025-07-08:
Same for me. The program no longer crashes after unregistering `REG_SPI_BASE` but I get a blank display no matter what I do. Does anyone have this library working with ESP32S3 right now?

sovran82, 2025-07-28:
Someone solved this issue. Have a look: https://www.youtube.com/watch?v=7FeBbqSW834&t=676s

tommywienert, 2025-07-30:
Hi there.
I watched the video and it didn't really explain how to fix the crash, nor the display staying white.
On some other issue in this repo, someone mentioned you should change 
https://github.com/Bodmer/TFT_eSPI/blob/master/Processors/TFT_eSPI_ESP32_S3.h#L83
to `#define SPI_PORT 3` for the ESP32-S3 to get rid of the panic. It fixed the panic, but also made the screen show nothing anymore in my case,
I fixed it by changing it to `#define SPI_PORT 2`

mike-lloyd03, 2025-08-06:
@niu541412 do you think you could open a PR so this can get fixed upstream?