Bodmer/TFT_eSPI #3746: Ruquest support ST7789V3

springeye, 2025-05-30:


nstepanets, 2025-06-05:
As far as I know, ST7789v3 is already supported by TFT_eSPI. Can you add a description of the problem?

springeye, 2025-06-06:
> As far as I know, ST7789v3 is already supported by TFT_eSPI. Can you add a description of the problem?

The pins connected to my screen are as follows(Using soft SPI):
```h
    #define TFT_SCLK PA7
    #define TFT_MOSI PA6
    #define TFT_RST  PA5
    #define TFT_DC   PA4
    #define TFT_CS   PA3
    #define TFT_BL  PA2
```

When I call init, the code gets stuck at line 691 of TFT_eSPI.cpp.
This line is stuck
```cpp
 writecommand(0x00); // Put SPI bus in known state for TFT with CS tied low
```
[image]

I saw ST7789 and ST7789_V2 in the TFT_Drivers directory, so I mistakenly thought it was incompatible with V3. Currently, I don't know where the problem lies with my code.